FreeCAD/FreeCAD-Bundle #2: FreeCAD should recognize when it's built as an .AppImage

luzpaz, 2017-07-24:
https://forum.freecadweb.org/viewtopic.php?f=10&t=15525&start=110#p167511
https://freecadweb.org/tracker/view.php?id=2987

luzpaz, 2017-07-31:
Workaround solution: https://github.com/AppImage/AppImageKit/issues/429#issuecomment-319132831

luzpaz, 2017-07-31:
@bblacey when you have a chance could you lend a hand with how to implement this for FC ?

luzpaz, 2018-02-09:
@realthunder I'd like to implement this but I'm having a challenging time figuring out where to do it for FC. Could you help ?

realthunder, 2018-02-09:
According to the workaround solution proposed by the AppImage guy, there is an APPIMAGE environment variable set by AppImage runtime. You can detect that and show it in the about page, for example check the environment variable at [here](https://github.com/FreeCAD/FreeCAD/blob/a20e13b8ae98e967b562d9c91756d6b3a55b9760/src/Gui/Splashscreen.cpp#L343), and return a different string, say 'Linux-AppImage'.
 
Alternatively, but a lot less convenient, you can have a custom built deb package. The AppImageKit does not build FC when it is building the AppImage. It just grabs a pre-built deb package from the freecad maintainer PPA. So you can ask the freecad maintainer to create a special deb package just for AppImage. I believe the PPA deb package uses Version.h header at [here](https://code.launchpad.net/~freecad-maintainers/+git/gitversioning), which contains various version, git hash, branch name strings. You can append any of this string with special keyword to identify AppImage.

probonopd, 2018-02-10:
At runtime, `$APPDIR` points to the mount point of the AppImage (the root of the AppDir) while `$APPIMAGE` points to the AppImage file. FreeCAD could use these at runtime to determine if it is running from an AppImage.

luzpaz, 2018-02-12:
Thanks everyone!
wmayer committed fix in 
https://github.com/FreeCAD/FreeCAD/commit/88e8c98d345c31f7c16711cb1702e10fba21f6c1

closed 2018-02-12 by luzpaz