FreeCAD/FreeCAD-Bundle #3: FileOpenSavePath points to /tmp

luzpaz, 2017-07-24:
https://forum.freecadweb.org/posting.php?mode=reply&f=10&t=15525#pr183554

luzpaz, 2017-08-15:
CC @probonopd

probonopd, 2017-08-15:
> I got an appimage and have run it without problems except for a nuisance. The path for opening files is only the current working dir of the appimage, which seems to always create a /tmp/.mount_FreeCAU5f21S when it runs and that is the "FileOpenSavePath"

An AppImage is a compressed filesystem that is mounted to a temporary mountpoint like `/tmp/.mount_FreeCAU5f21S`  when it is executed, and the application is then run from there. Since some applications hardcode paths to within `/usr`, we do a little trick: We patch `/usr` with a string of the same length, `././`, which means "this directory. To make this work, we `chdir()` to `usr/` inside the temporary mountpoint before executing the application. FreeCAD seems to present the user the current working directory (`$CWD`) when showing the file dialogs. This is why you see the temporary mountpoint location.

> Is there a way to change this when running an appimage?

Mutliple options, with the cleanest (but also hardest) option first:
1. Ensure that FreeCAD and its dependencies do not hardcode any paths to `/usr` anywhere (can also happen  during compilation), i.e., make the application fully "relocateable" (= run from any location) by using paths relative to the application only. Then we don't need the workaround described above and can remove the patch and the `chdir()`
2. When presenting the user with file dialogs, show `$OWD` (=original working directory) if this environment variable is set. It is always set inside an AppImage
3.  When presenting the user with file dialogs, show `$HOME`

triplus, 2019-08-31:
When migrating from PPA to Conda (FreeCAD 0.18+), as a basis for AppImage, things changed. File dialogs now default to current working directory. This is standard Linux behavior and therefore i am closing this issue report.

closed 2019-08-31 by triplus