FreeCAD/FreeCAD-Bundle #4: Segfault on Arch

randrej, 2017-11-10:
I can't seem to run the Appimages of 0.17 on my Arch install.
[strace](https://github.com/FreeCAD/FreeCAD-AppImage/files/1462706/strace.txt) of my problem is attached.  

The version is:
`FreeCAD-0.17.9747.glibc2.17-oce17.1-ng.x86_64.AppImage`
The other image doesn't work either.

Arch's current glibc is version 2.26-6, if that matters.

This is the output if that matters.

    FreeCAD 0.17, Libs: 0.17R9747 (Git)
    © Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2016
      #####                 ####  ###   ####
      #                    #      # #   #   #
      #     ##  #### ####  #     #   #  #   #
      ####  # # #  # #  #  #     #####  #   #
      #     #   #### ####  #    #     # #   #
      #     #   #    #     #    #     # #   #  ##  ##  ##
      #     #   #### ####   ### #     # ####   ##  ##  ##

    [1]    26406 segmentation fault (core dumped)  strace -o str ./FreeCAD-0.17.9747.glibc2.17-oce17.1-ng.x86_64.AppImage

probonopd, 2017-11-10:
Can you run with LD_DEBUG=libs ./FreeCAD*.AppImage, and see which libraries are loaded right before the crash?

randrej, 2017-11-11:
Output [attached](https://github.com/FreeCAD/FreeCAD-AppImage/files/1463660/libs.txt).

triplus, 2018-10-16:
I will close this issue report for now as i tested both available AppImages (PPA/Conda) on Manjaro Linux XFCE Edition (17.1.12). And both worked fine. Manjaro Linux is based on Arch and in addition i do remember some Arch users using the PPA based AppImage successfully in the past few months. Therefore whatever was causing this specific issue likely isn't relevant anymore.

closed 2018-10-16 by triplus