FreeCAD/FreeCAD-Bundle #6: Appimages should be signed

luzpaz, 2018-11-20:
When using the `AppimageUpdate` tool with the FC AppImages, one sees this anxiety-generating warning (see below). Beside the fact that one can still run the Appimage, the process doesn't inspire much confidence. What could we do about this? 

[image]

CC @sgrogan @bblacey @looooo

probonopd, 2018-11-20:
> What could we do about this?

 Use the `-s` switch of [appimagetool](https://github.com/AppImage/AppImageKit#appimagetool-usage) to embed a signature into the AppImage.

luzpaz, 2019-01-22:
https://gitter.im/FreeCAD/FreeCAD?at=5c4714dccb47ec3000836fc5

looooo, 2019-06-26:
volunteers?

luzpaz, 2019-06-26:
@looooo we'd need to have a gpg private and public key in order to do this.

looooo, 2019-06-26:
Not sure how to do this via ci. If anyone has experience with signing please send a PR or give some instructions.

triplus, 2019-08-27:
The problem is not being able to sign the packages. The main problem is including the signing step in Travis build procedure. I plan to take a look at some of the remaining FreeCAD 0.18 related AppImage issues in near future. Will ask on forum if anybody having the needed permission on Travis is interested in making it work. I can help figuring out the steps needed after.

triplus, 2019-08-27:
I prefer discussing such in-depth things on forum:

https://forum.freecadweb.org/viewtopic.php?f=10&t=34981&p=329620#p329620

Thanks.

probonopd, 2019-08-28:
@probonopd care to show us an example?

Check out `appimagetool -s`. It will sign with the key from your default GPG keyring.

probonopd, 2019-08-28:
Sure, we are doing it for our own AppImages. It's a bit tricky: We encrypted the private key in an archive secured with a long password which is stored as a secure variable in Travis CI.

https://github.com/AppImage/AppImageKit/blob/6baa36d0b1e0481aa36a36bb54140b81959b3874/travis/travis-build.sh#L24-L26

triplus, 2020-03-20:
First thanks for all the information. Concerning the FreeCAD 0.19 release i don't plan to implement this for now. The idea of sharing the private key, although encrypted, is what ticks me off.

P.S. Will leave the report opened for possible future re-evaluation. If somebody else would like to go ahead with this. I am OK with that.

probonopd, 2020-03-21:
> The idea of sharing the private key, although encrypted, is what ticks me off.

Can you elaborate? You are not "sharing" the private key if you use it to sign an AppImage, you are just _using_ it to make the signature. If you don't want Travis CI to sign the AppImages, you could

1. Have Travis CI build an unsigned one
2. Sign it after the fact on your local developer machine

If this is what you need, please let us know, so that we can discuss to implement this workflow in our tooling.

triplus, 2020-03-22:
Yes, i was talking about possible automation on Travis. As for other option, doing the signing locally, at best we likely would end up signing the "stable" AppImage. We had a few discussions about that on the forum in the past. Somehow i feel that currently this is just not feasible. For checking downloaded package integrity we provide a checksum. Beyond that, like downloading the package and re-upload the signed package, or sharing the encrypted private key. I don't plan to do that for now.

luzpaz, 2020-10-28:
IMHO, since we now have a technically feasible option to distribute appimages over bittorent (#49), we should resume discussions about signing them.

probonopd, 2020-10-29:
fwiw, it is possible to create a new key that is _just_ used for AppImage signing purposes.

luzpaz, 2020-10-29:
@probonopd do you mind elaborating?

probonopd, 2020-10-29:
If I read the above correctly, FreeCAD AppImages are currently not signed because the AppImage team is hesitant to upload, even encrypted, the FreeCAD private key somewhere. Which I can understand. Hence I suggested a completely new key to be made for the only purpose of signing AppImages.

luzpaz, 2021-02-25:
We're now generating through GIthub actions. Here is how to store secret variables in Actions
https://docs.github.com/en/actions/reference/encrypted-secrets  
I'll volunteer to create a gpg key  just for appimages and share it with @looooo

luzpaz, 2021-02-25:
> Sure, we are doing it for our own AppImages. It's a bit tricky: We encrypted the private key in an archive secured with a long password which is stored as a secure variable in Travis CI. [...]

this doesn't look encouraging:  
> #### NOTE ####
> # Signing is currently broken, as the secret to decrypt the key is not available at the moment.
> # Even worse, putting key material in a third-party execution environment is not a good idea
> # to begin with.
> # The signing doesn't add anything security wise anyway, so it doesn't really matter if it works
> # at the moment or not.
> #### END NOTE ####

ref: tps://github.com/AppImage/AppImageKit/blob/a3c69f66ea6404315bd02724cde92494a42ab3f0/ci/build.sh#L34-L40

adrianinsaval, 2021-08-11:
any progress on this?

luzpaz, 2021-08-12:
Feel free to jump in

probonopd, 2021-08-16:
Contributions to https://github.com/AppImage/AppImageKit to improve the situation would be welcome.

adrianinsaval, 2024-08-07:
this is now implemented for the x86_64 appimages with this key id https://keys.openpgp.org/vks/v1/by-fingerprint/F0D17301E5C1EB2A3C1DE96E94DF3243CACA7A94

but for some reason it's still failing for the aarch64 build. The error message is very vague so I'm not sure what's going on there:
```
[sign] signing requested
[sign] found gpgme version 1.23.2
[sign] found gcrypt version 1.10.3
[sign] calculated digest: 2862d31f63f49acd45e853a6f5fb6687a0f52ad7aebcf770b97b2564732aaf86
[sign] using engine OpenPGP (found in /usr/bin/gpg), version 2.2.27, gpgme requires at least version 1.4.0
[sign] passphrase available from environment, setting pinentry mode to loopback
[sign] gpgme_get_key(gpgme_ctx, key_id, &gpgme_key, true): call failed: End of file
Signing failed, aborting
```