FreeCAD/FreeCAD-Bundle #7: Generating Appimages (when and in-sync with the PPA and other OSs)

luzpaz, 2018-11-20:
@looooo writes in this [thread](https://forum.freecadweb.org/viewtopic.php?f=10&t=15197&start=220#p269403):
> FreeCAD builds are done every sunday. 

@sgrogan writes in a [thread](https://forum.freecadweb.org/viewtopic.php?f=4&t=32204#p268741):
> The magic is all done in the upload script https://raw.githubusercontent.com/probonopd/uploadtool/master/upload.sh
What it does is make sure that the new builds succeeded then deletes and creates a new one with the same name (preserving the url) then uploads the new assets.
The problem persists how to coordinate the OSX, Win, Conda/Appimage, PPA/Appimage coming from 4 different places at 4 different times?

So 2 things:
1. Can we generate daily builds of the dev version ?
2. Can we trigger all builds to build (OSX, Win, Conda/Appimage, PPA/Appimage) all together ?

looooo, 2018-11-21:
I am pretty happy with the one build a week strategy. For maximal stable builds I also do not do any caching. So build time is approx. 1:45. So I don't think to make the builds more often. Also have a look at download statistics: https://anaconda.org/freecad/freecad/files

luzpaz, 2018-11-21:
@looooo 
Isn't what we really want are the download stats from github because that is where people are downloading builds from...right?

luzpaz, 2018-11-21:
`curl -s 'https://api.github.com/repos/FreeCAD/FreeCAD/releases/tags/0.18_pre' | grep -A3 download_count`
```
      "download_count": 1131,
      "created_at": "2018-10-15T12:58:59Z",
      "updated_at": "2018-10-15T12:59:28Z",
      "browser_download_url": "https://github.com/FreeCAD/FreeCAD/releases/download/0.18_pre/FreeCAD_0.18-14976.e79a198-OSX-x86_64-Qt5.dmg"
```
```
      "download_count": 37,
      "created_at": "2018-11-20T15:55:30Z",
      "updated_at": "2018-11-20T15:55:38Z",
      "browser_download_url": "https://github.com/FreeCAD/FreeCAD/releases/download/0.18_pre/FreeCAD_0.18.15221.glibc2.17-x86_64.AppImage"
```
```
      "download_count": 5,
      "created_at": "2018-11-20T15:55:38Z",
      "updated_at": "2018-11-20T15:55:39Z",
      "browser_download_url": "https://github.com/FreeCAD/FreeCAD/releases/download/0.18_pre/FreeCAD_0.18.15221.glibc2.17-x86_64.AppImage.zsync"
```
```
      "download_count": 26,
      "created_at": "2018-11-20T15:55:40Z",
      "updated_at": "2018-11-20T15:55:49Z",
      "browser_download_url": "https://github.com/FreeCAD/FreeCAD/releases/download/0.18_pre/FreeCAD_0.18.15221_Conda_Py3Qt5_glibc2.12-x86_64.AppImage"
```
```
      "download_count": 5,
      "created_at": "2018-11-20T15:55:50Z",
      "updated_at": "2018-11-20T15:55:50Z",
      "browser_download_url": "https://github.com/FreeCAD/FreeCAD/releases/download/0.18_pre/FreeCAD_0.18.15221_Conda_Py3Qt5_glibc2.12-x86_64.AppImage.zsync"
```
```
      "download_count": 48,
      "created_at": "2018-11-20T22:45:28Z",
      "updated_at": "2018-11-20T22:51:23Z",
      "browser_download_url": "https://github.com/FreeCAD/FreeCAD/releases/download/0.18_pre/FreeCAD_0.18.15248_Conda_Py3QT5-WinVS2015_x64.7z"
```
```
      "download_count": 125,
      "created_at": "2018-11-20T22:28:18Z",
      "updated_at": "2018-11-20T22:31:28Z",
      "browser_download_url": "https://github.com/FreeCAD/FreeCAD/releases/download/0.18_pre/FreeCAD_0.18.15248_x64_dev_win.7z"
```
```
      "download_count": 20,
      "created_at": "2018-11-20T22:28:18Z",
      "updated_at": "2018-11-20T22:34:11Z",
      "browser_download_url": "https://github.com/FreeCAD/FreeCAD/releases/download/0.18_pre/FreeCAD_0.18.15248_x86_dev_win.7z"
```

triplus, 2019-09-06:
We have migrated to using Conda as a basis for AppImages. The AppImage creation and deployment procedure is now fairly automated by Travis. On Conda side @looooo gets to decide what happens and when. There are currently no plans to pack Travis build artifacts as an AppImage or to be in sync with the PPA.

Hence closing this issue.

closed 2019-09-06 by triplus