FreeCAD/FreeCAD-Bundle #8: TravisCI webhook to trigger a remote script (in this case to update the wiki)

luzpaz, 2018-11-20:
How do we setup a post-build TravisCI webhook that triggers a remote script to update the wiki {{Template:Development-Version}} template (and any future templates) that can be used to keep the link to the most up to date development version synchronized?

The script is already created. The only missing portions of the workflow is listening for the webhook and the webhook script on TravisCI itself. 

CC @kkremitzki

sgrogan, 2018-11-20:
This is on the right direction.
Can we get secure credentials for the wiki into Travis?
If so, we should be able to push the correct version to the wiki.
If we can add something to the "after_success:" block, it would be accurate and probably the most resilient.

luzpaz, 2018-11-20:
> Can we get secure credentials for the wiki into Travis?

@sgrogan didn't occur to me that we could do this. Would that mean installing the pywikibot and running it directly from travis? Is that what you mean? 

That's another order of complexity that I'm not sure about.
Right now the way I'm thinking is:
travisci successful appimage build triggers an endpoint that triggers the script on a remote server that logs in to the wiki (under my account) and posts the newest revision number to https://www.freecadweb.org/wiki/Template:Development-Version

kkremitzki, 2018-11-20:
> that triggers the script on a remote server

We could use the Frankfurt server for this.

sgrogan, 2018-11-21:
@luzpaz , yes this is what I'm suggesting. No need for an additional webhook or another server. Update the wiki link at the same time the build is deployed. Maybe we need a Travis-CI wiki account, we already pass github credentials. This should also work for the OSX builds as well as the two different AppImage builds.

luzpaz, 2022-07-11:
Giving up on this feature. Closing