FreeCAD/FreeCAD-Bundle #14: module canberra-gtk-module missing and xcb unhappy

thirschbuechler, 2019-04-06:
Tried the Appimage just now (Ubuntu 18.04) and failed.

Then canberra-part got resolved by [installing this](https://askubuntu.com/questions/342202/failed-to-load-module-canberra-gtk-module-but-already-installed) but xcb remains unhappy and prevents launching, failing silently. Only running it via terminal yielded the following output (before resolving canberra):
```
$ ./FreeCAD*
/bin/bash: /tmp/.mount_FreeCAA4khuw/lib/x86_64-linux-gnu/libtinfo.so.5: no version information available (required by /bin/bash)
FreeCAD 0.17, Libs: 0.17R13541 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2018
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Gtk-Message: 09:10:14.901: Failed to load module "canberra-gtk-module"
failed to create drawable
[xcb] Unknown sequence number while processing queue
[xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
[xcb] Aborting, sorry about that.
freecad: ../../src/xcb_io.c:259: poll_for_event: Assertion `!xcb_xlib_threads_sequence_lost' failed.
Aborted (core dumped)
```

thirschbuechler, 2019-04-06:
using [the FreeCAD forum xcb workaround](https://forum.freecadweb.org/viewtopic.php?f=3&t=30573) it worked, but not out of the box

thirschbuechler, 2019-04-06:
or more precisely, the advertised 0.17 FreeCAD AppImage has this problem, the "hidden" 0.19_pre in the releases-section doesn't: https://github.com/FreeCAD/FreeCAD/releases

thirschbuechler, 2019-04-06:
suggested: https://github.com/FreeCAD/FreeCAD-AppImage/compare/master...thirschbuechler:patch-1

luzpaz, 2019-04-06:
@thirschbuechler since this is solved on 0.19 I propose we close this ticket unless there is something left ?

normandc, 2019-04-07:
I want to point out that the missing canberra-gtk-module and xcb error are two totally separate and different issues. The first actually does not prevent FreeCAD from running. The freecad-stable and freecad-daily PPA packages have been getting this error for a long time and they have never failed to launch nonetheless. In fact, I'm puzzled by this dependency, since FreeCAD is Qt-based.

thirschbuechler, 2019-04-07:
@luzpaz yes i guess so, but I'd strongly suggest updating the FreeCAD website and the readme here (e.g. via my pull request). I also redirected  a forum entry here of someone being disappointed of the AppImage not working
@normandc well, seems someone left some old dependencies in, then^^

normandc, 2019-04-07:
@thirschbuechler but that's the thing, nowhere in the FreeCAD compilation is that dependency specifically required. It simply should not be.

probonopd, 2019-04-09:
>  I'm puzzled by this dependency, since FreeCAD is Qt-based.

Maybe a Gtk theme for Qt is being loaded?

triplus, 2019-09-06:
I am closing this issue report, due to FreeCAD 0.19 AppImage not having such issues. FreeCAD 0.18 AppImage uses older Qt version, and that is likely from where the (theming related) message comes from. We don't plan to upgrade the Qt version for FreeCAD 0,18 AppImage. As for the crashing issue. That is therefore a GPU driver related issue and it is not related to the AppImage.

closed 2019-09-06 by triplus