FreeCAD/FreeCAD-Bundle #19: Appimage 0.18 Crash on Arch

roboteck, 2019-05-01:
Downloaded the lastest appimage on the website and also used the Appimage updatetool to update the image.

when i try to create a new document it crashes.

attached crash report
[CrashReport.txt](https://github.com/FreeCAD/FreeCAD-AppImage/files/3134308/CrashReport.txt)

Harvie, 2019-08-11:
I have same problem:

```
./FreeCAD_0.18-16131-Linux-Conda_Py3Qt5_glibc2.12-x86_64.AppImage
/tmp/.mount_FreeCAhrs3mT
FreeCAD 0.18, Libs: 0.18R16131 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2019
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

During initialization the error 'ascii' codec can't decode byte 0xc3 in position 78: ordinal not in range(128) occurred in /home/harvie/.FreeCAD/Mod/SheetMetal/InitGui.py
Please look into the log file for further information
Program received signal SIGSEGV, Segmentation fault.
/tmp/.mount_FreeCAhrs3mT/AppRun: line 12: 21001 Segmentation fault      (core dumped) ${HERE}/usr/bin/FreeCAD "$@"

```

triplus, 2019-08-25:
Can any of you confirm if FreeCAD 0.18/0.19 works on Arch?

Thanks.

Harvie, 2019-08-26:
FreeCAD 0.18, Libs: 0.18R16131 (Git) seems to work for me. Only problem i've encountered is that path workbench does not load.

triplus, 2019-08-27:
Thanks for providing the feedback. Could you in addition please test the FreeCAD 0.19 AppImage too:

https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre

Harvie, 2019-08-27:
@triplus Yes! just checked and 0.19pre seems to be working for me!

triplus, 2019-08-27:
Great!

P.S. Now at this point i am not yet sure if we will be updating the FreeCAD 0.18 AppImage or not.

https://forum.freecadweb.org/viewtopic.php?f=10&t=34981&p=329668#p329668

triplus, 2019-08-31:
After some discussion on the forum it was decided, we likely won't be upgrading FreeCAD 0.18 AppImage dependencies (on Conda). Due to underlying technical difficulties. As it was confirmed FreeCAD 0.19 AppImage currently doesn't crash on Arch Linux, using FreeCAD 0.19 AppImage is considered as a solution.

Closing this issue report.

luzpaz, 2019-09-04:
@sgrogan would you be so kind to please notate "the v0.18.3 Release page that Arch and it's dervatives will experience a segfault when running this appimage but that it's been fixed in 0.19" or something to that effect?
Thanks!

probonopd, 2019-09-04:
Do you know what the fix consisted of? What was changed to solve the "underlying technical difficulties"? Asking out of curiosity because other applications may also get bug reports from Arch users.

triplus, 2019-09-04:
The main problems with FreeCAD 0.18 AppImage are related to the fact, we use Qt 5.6.2. FreeCAD 0.19 AppImage uses Qt 5.12. Upgrading FreeCAD 0.18 AppImage, to use Qt 5.12, that is a challenging task and it likely won't happen.

probonopd, 2019-09-04:
Ok, so we can say to other application's authors experience similar issues, "if this error occurs, try switching to Qt 5.12" then. Thanks, that is a helpful hint.

triplus, 2019-09-04:
OK, fair enough. I will reopen this issue report. Tomorrow i will try a bit harder, to find and explain the underlying problem.

probonopd, 2019-09-04:
Actually I was already (really!) thankful for the explanation that "switching to Qt 5.12" did the trick here. I was not joking. No need to investigate more at this point.

closed 2019-09-05 by triplus

triplus, 2019-09-05:
I am waiting for some further confirmation on forum, and maybe from people participating in this issue report. What i can already say is the use of Conda package libstdcxx-ng, that will result in crashes on Fedora 30, if version 7.2.0 (libstdc++.so.6.0.25) is used. Upgrading to version 9.1.0 (libstdc++.so.6.0.26) resolves it.

P.S. As for the Qt 5 debate. In general the experience with a version such as Qt 5.6 will not be optimal. From theming issues to missing menus and beyond. I guess at minimum Qt 5.9 should be used.

commit a30f6156 (2019-09-05): Upgrade libstdcxx

closes #19