FreeCAD/FreeCAD-Bundle #23: linux_dev qt5.9

looooo, 2019-06-23:
@triplus 
FreeCAD works with qt5.9 in a conda-environment but it is not launching if it is bundled as an appimage. In the logs there is something written about xkb.

wwmayer, 2019-06-23:
It may lack of the Qt plugin qt5/plugins/platforms/libqxcb.so

looooo, 2019-06-23:
I guess my hint was a bit missleading, but thanks anyway. I had another look at the log file and at the end there is something about qtwebengineprocess not found. So I set this env-variable:

`export QTWEBENGINEPROCESS_PATH=$HERE/usr/libexec/QtWebEngineProcess`

which lead to another crash:

```
Log: Installed Qt WebEngine locales directory not found at location /home/k/projects/FreeCAD-AppImage/conda/linux_dev/AppDir/usr/translations/qtwebengine_locales. Trying application directory...
Log: Qt WebEngine locales directory not found at location /tmp/.mount_FreeCA3R8XeX/usr/bin/qtwebengine_locales. Trying fallback directory... Translations MAY NOT not be correct.
```

```
[0623/184244.190446:ERROR:icu_util.cc(178)] Invalid file descriptor to ICU data received.
[0623/184244.190475:FATAL:content_main_runner.cc(691)] Check failed: base::i18n::InitializeICU(). 
#0 0x7fe60e6540be <unknown>
#1 0x7fe60e6666c2 <unknown>
#2 0x7fe60e63d068 <unknown>
#3 0x7fe60e01050e <unknown>
#4 0x7fe60e010718 <unknown>
#5 0x7fe60e007681 <unknown>
#6 0x7fe60e007732 QtWebEngineCore::WebContentsAdapter::WebContentsAdapter()
#7 0x7fe617958376 <unknown>
#8 0x7fe6179560c4 <unknown>
#9 0x7fe617956252 QWebEnginePage::QWebEnginePage()
#10 0x7fe61795bd8c QWebEngineView::page()
#11 0x7fe62571110a WebGui::BrowserView::BrowserView()
#12 0x7fe625709c27 WebGui::Module::openBrowserWindow()
#13 0x7fe62570895a Py::ExtensionModule<>::invoke_method_varargs()
```

qtwebengine_locales is in $HERE/usr/translations/qtwebengine_locals . Somehow we have to set it with a environment variable. Maybe something similar has to be done for osx.

probonopd, 2019-06-23:
It would be good if we could find a way without needing environment variables; https://doc.qt.io/archives/qt-5.11/qtwebengine-deploying.html says:

> At runtime, Qt WebEngine looks for the QtWebEngineProcess executable in the directory that QLibraryInfo::location(QLibraryInfo::LibraryExecutablesPath) returns. For Qt installations, this is QTDIR/libexec (Linux) or QTDIR\bin (Windows). The path can be changed by defining a qt.conf file, for example. Alternatively, an executable path can be set as a value of the QTWEBENGINEPROCESS_PATH environment variable.

So `qt.conf` seems to be the way to go; it is also what [linuxdeployqt](https://github.com/probonopd/linuxdeployqt) uses.

looooo, 2019-06-24:
qt.conf is currently not in the bundle. It looks like this for a conda-environment:

```
[Paths]
Prefix = /home/k/conda/envs/freecad-qt5.9
Binaries = /home/k/conda/envs/freecad-qt5.9/bin
Libraries = /home/k/conda/envs/freecad-qt5.9/lib
Headers = /home/k/conda/envs/freecad-qt5.9/include/qt
```

For the appimage we should replace the absolute paths with relatives. Or is it possible to use environment variables like ${PREFIX} here?
Also I am not sure how this should solve the reported issue as qt from conda seems to use a custom location for qwebengine_locals ($PREFIX/translations/qtwebengine_locals).

wwmayer, 2019-06-24:
> I guess my hint was a bit missleading, but thanks anyway.

What does this mean now? Does FreeCAD not even start or does it crash immediately after start?
FYI, in order to make the StartPage working with QtWebEngine a few things are needed:
* Qt must be able to start QtWebEngineProcess
* Qt must find the resource files of QtWebEngine (these are some .pak files and a .dat file)
* Qt should be able to find some locale files but AFAIR also works without them

Fore more information see also: https://forum.freecadweb.org/viewtopic.php?f=4&t=35943#p303953
Although this is about the Libpack on Windows I guess the QtWebEngine part also applies to Linux

looooo, 2019-06-24:
> What does this mean now? Does FreeCAD not even start or does it crash immediately after start?

Yes it was still crashing but I think xcb is not the problem. Anyway I added the qt.conf and locally this seems to work now. I added the same qt.conf also to the macOSX bundle. Not sure if this will work.

looooo, 2019-06-24:
Adding qt.conf still doesn't solve the crash. I guess it's best to leave this (appimages) open to someone with qt-experience (@triplus). I have enough frustration with keeping the conda-packages updated.

probonopd, 2019-06-24:
This is what linuxdeployqt uses

```
[Paths]
Prefix = ../
Plugins = plugins
Imports = qml
Qml2Imports = qml
```

`Prefix = ../` makes everything else relative to the directory above the `bin/` directory in which the main executable and the `qt.conf` file are both located.

looooo, 2019-06-24:
I tried with relative paths already:

```
[Paths]
Prefix = ..
Binaries = ../bin
Libraries = ../lib
Headers = ../include/qt

```

doesn't solve it either.

triplus, 2019-06-24:
> In the logs there is something written about xkb.

Just to make sure it's not something about xcb?

P.S. I still didn't have the time to read about progress done in the past month (forum). Will try to do that in next 48h. As for the problematic Qt 5.9 based AppImage. Could you upload it somewhere temporary? Thanks.

probonopd, 2019-06-24:
```
[Paths]
Prefix = ..
Binaries = ../bin
Libraries = ../lib
Headers = ../include/qt
```

If I understand it correctly, everything else in `qt.conf` is relative to Prefix, which would mean that you have specified binaries to be in `../../bin`, and libraries in `../../lib`, which is most likely not what you wanted to say.

looooo, 2019-06-25:
> Just to make sure it's not something about xcb?

Yes it was not about xcb. There was a note about xcb in the launcher image before the crash. That's lead to my assumption.

Anyway with all the help it seems like the qt.conf is working now and I it was possible to remove some env-variables.

> P.S. I still didn't have the time to read about progress done in the past month (forum). Will try to do that in next 48h.

Actually I did't think there happened too much about packaging (conda). Weekly conda-builds are now at qt5.9 py3.7 boost1.70 vtk8.2 which is the head of available conda-packages. Appimage and osx-bundle are now also using these deps. Theme-problems are solved by removing gtk2-package (solution provided by @realthunder, who also provided a fix for qt-webengine crash on linux. This are the changes from my side. I hope this helps to enjoying the rest of the 48h on more enjoyable things ; )

closed 2019-07-05 by looooo