FreeCAD/FreeCAD-Bundle #25: FreeCAD documentation no included in AppImage

luzpaz, 2019-07-03:
See bugtracker ticket No. 3960 
https://www.freecadweb.org/tracker/view.php?id=3960

looooo, 2019-07-05:
I guess we do not include docs in pre bundles. 0.18 should have it included.

luzpaz, 2019-08-22:
[image]

I see this with:  

OS: Manjaro Linux (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.17798 (Git) AppImage
Build type: Release
Branch: master
Hash: 7e60631239109c632a8f6cca83f6e7e5502e43a1
Python version: 3.7.3
Qt version: 5.9.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

triplus, 2019-08-22:
Yes, this issue wasn't resolved yet. I plan to take a closer look in the following days.

luzpaz, 2019-08-22:
Instead of bundling the help files in to the appimage and making it even more bloated... what if we offered it via the addon manager ?

triplus, 2019-08-22:
We had some discussions in the past on the forum and it was decided for stable releases to have documentation included. If you want to discuss this further best to do that on forum. As for the mentioned issue i plan to take a look regardless.

probonopd, 2019-08-23:
If FreeCAD uses AppImageUpdate for binary delta updates, then bundling the help files in to the AppImage even for continuous builds might be more feasible because it would only be an increase to the initial download size, while leaving subsequent updates pretty much unaffected. Maybe it could even be integrated into the FreeCAD GUI using e..g., https://github.com/antony-jr/updatedeployqt? cc @antony-jr

triplus, 2019-08-23:
We have enabled delta updates a while back.

triplus, 2019-09-04:
I took a look and the embedded documentation issue is twofold. I resolved the first issue, by override hard-coded paths with qt.conf. And now pressing on F1 should open the Qt Assistant. 

https://github.com/FreeCAD/FreeCAD-AppImage/commit/7c84d1f437e08dd12f0931865207e6229c80bd89

The second problem is not as straightforward to resolve. AppImage works in a way, it mounts the AppImage under /tmp/random_location. The random_location is read-only and changes after each run. Qt Assistant will on the other hand take freecad.qhc and cache it in .local/share/freecad/freecad/freecad.qhc location. The main problem is that in this cached file, the absolute path to FreeCAD help files will get set on the first Qt Assistant run. The embedded documentation will work for the first time, but starting FreeCAD for the second time, Qt Assistant won't find the help files anymore.

Reference:
https://forum.freecadweb.org/viewtopic.php?f=10&t=34981&p=331804#p331804

probonopd, 2019-09-04:
Possibly this is worth reporting to Qt upstream; I think it's not reasonable for a Qt application to assume it is always executed from the same location.

triplus, 2019-09-05:
Done:
https://bugreports.qt.io/browse/QTBUG-12050

vrubiolo, 2019-11-27:
Thanks for reporting the bug to upstream Qt. Do note however that the upstream ticket appears to have been closed before that so I am not sure if your comment will be read/addressed ...

triplus, 2020-02-16:
Basically the problem for us is the fact Qt Assistant is caching qhc files. Without manually deleting such files, when help files relocate, help won't work. The above PR, where i tired to manually unregister and register help files, that currently doesn't work, Qt Assistant fails and doesn't provide a reason for such outcome. I hence opened a new feature request, to add an option to Qt Assistant, for being able to opt-out of qhc caching in the first place: 

https://bugreports.qt.io/browse/QTBUG-82193

probonopd, 2020-02-17:
As a crude workaround for the meantime, you could have the `AppRun` script create a symlink to `$APPDIR`... did I mention it is crude?

triplus, 2020-02-17:
Thanks for the suggestion. For now i will leave this issue report open, for possible future reevaluation. The selected documentation support strategy regarding FreeCAD 0.18 and FreeCAD  0.19 AppImage is to embed the documentation and manage the cached .qhc file. : https://github.com/FreeCAD/FreeCAD-AppImage/commit/1e5a73293d27d7aa184f3dc085302dd7d8753ab8

https://forum.freecadweb.org/viewtopic.php?f=4&t=43015&p=369824#p369824

luzpaz, 2020-03-19:
New upstream ticket was opened https://bugreports.qt.io/browse/QTBUG-82193

luzpaz, 2020-12-04:
Can someone bump the upstream Qt bug ?

ok1vaw, 2021-01-26:
I have the same problem as above 0.19pre version. Although installing or uninstalling 0.18, nothing helps, same error as above.

RetiredWeseal, 2022-03-14:
I'm sorry, I am new at this but it looks like this issues is over a year old.  is there a fix for it? Here is what I get when I try to use help:Unable to launch Qt Assistant (/private/var/folders/jy/mbq5mqf57lg4ldcw_n3l6n1h0000gn/T/AppTranslocation/E8211E4B-995E-4255-B907-0754E8E33361/d/FreeCAD.app/Contents/Resources/bin/assistant

looooo, 2022-03-14:
we do not include the qt assistent in the weekly builds.

luzpaz, 2022-07-11:
Help docs are now in a 3rd party addon. See Addon Manager.  
Here is the specific addon repository: https://github.com/FreeCAD/FreeCAD-Help