FreeCAD/FreeCAD-Bundle #26: Include solve space and asm3

looooo, 2019-08-26:
What do you think about including asm3 and solvespace in 0.19 pre? As solvespace is a compiled library, I would prefer to build it with conda tools to force compatibility.

@realthunder
@triplus

realthunder, 2019-08-27:
Will it be simpler to install it from pip? I have already made it work using scikit-build under Linux. Just haven't got time to build all the wheels and upload it yet.

looooo, 2019-08-27:
Not sure if it is simpler. There is also an ongoing process to create wheels from conda-forge packages. So it might be possible to upload wheels from conda-recipes to pypi at some point. Not sure about scikit-build. Never tried that tool.

triplus, 2019-08-27:
If i understand your proposal correctly, you will add this packages to Conda and are asking if after they should be added to the AppImage too? I don't see any problem with that if that is desired by people. In addition if Conda will provide this packages for all platforms, that would be an additional plus. As in general i feel that AppImage should behave the same as the binaries on all supported platforms.

P.S. I personally much rather discuss such things on forum.

looooo, 2019-08-27:
> If i understand your proposal correctly, you will add this packages to Conda and are asking if after they should be added to the AppImage too?

Yes exactly.

> In addition if Conda will provide this packages for all platforms, that would be an additional plus.

Yes I think this will happen sooner or later. I am not sure how to package asm3 yet as it is not really a namespace package, but for sure this should be possible.

Another idea is to package only solvespace (which I already did for osx and linux) and make asm3 available via the addon-manager.

> P.S. I personally much rather discuss such things on forum.

:+1:

realthunder, 2019-08-28:
> Yes I think this will happen sooner or later. I am not sure how to package asm3 yet as it is not really a namespace package, but for sure this should be possible.

asm3 must be installed under `Ext` directory. If I remembered correctly, it is actually suggested by you? And that's why I mention pip install.

looooo, 2019-08-28:
> asm3 must be installed under Ext directory. If I remembered correctly, it is actually suggested by you? And that's why I mention pip install.

Maybe my instructions on this we're not very clear. Have a look here to see how the structure of a namespace package should look: https://github.com/FreeCAD/Workbench-Starterkit#minimal-structure-of-a-namespace-package-adding-a-workbench-to-freecad

If you use this structure the wb will be installable via add-on manager or pip.

Some examples:
https://github.com/looooo/freecad.gears
https://github.com/booya-at/OpenGlider

looooo, 2019-08-28:
> I briefly looked at addon manager source code. It seems that it can only install module into Mod directory?

Absolutely correct, but if you use the nested structure (as I suggested) for asm3 it will install into `Mod/FreeCAD_assembly3/freecad/asm3` and this will work as first `Mod/Freecad_assembly3` is added to sys.path (old way of module initialization) and afterwards is treated as a namespace package (extension to the `freecad`-namespace) by python and initialized (new initialization). This works by accident and never was planned this way.

>  It is always installed in Ext/freecad/asm3.

`Ext/freecad` shouldn't be used. This directory should be used for internal modules which want to use the new initialization or if we ever want to integrate an external workbench into the core FreeCAD library. I know the name is a bit misleading here.

realthunder, 2019-08-28:
> If you use this structure the wb will be installable via add-on manager or pip.

Yes, asm3 follows this structure. It is always installed in `Ext/freecad/asm3`.  I briefly looked at addon manager [source code](url=https://github.com/FreeCAD/FreeCAD/blob/60a54207acaee4dd560844f08d0fc4802493ce78/src/Mod/AddonManager/addonmanager_workers.py#L187). It seems that it can only install module into `Mod` directory? Are you sure it can install into `Ext/freecad`? Any settings required for that?

realthunder, 2019-08-28:
I see. I'll give it a try soon.

triplus, 2020-02-08:
I guess this was done for FreeCAD 0.19 AppImage, hence closing.

closed 2020-02-08 by triplus