FreeCAD/FreeCAD-Bundle #27: Consider making the AppImage self-updatable from the menu

probonopd, 2019-09-08:
Now that FreeCAD is promoting https://www.freecadweb.org/wiki/Appimage#Auto_Updating, why not make the AppImage self-updatable from the menu? Possibly using https://github.com/antony-jr/AppImageUpdaterBridge

cc @antony-jr

luzpaz, 2020-03-19:
Discussion thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=44324

antony-jr, 2020-03-19:
Look into https://github.com/TheAssassin/python-appimageupdate but it comes with a lot of issues because the library uses **cffi**, See https://github.com/Pext/Pext/issues/95

I once used Qt plugins with PyQt with no problem, But it might take some time to make documentation on that.

luzpaz, 2020-03-20:
@antony-jr Nice! Is it compatible with pyside2. I don't think FreeCAD is using PyQt5, I may be wrong.

antony-jr, 2020-03-20:
> @antony-jr Nice! Is it compatible with pyside2? I don't think FreeCAD is using PyQt5, I may be wrong.

It should if it has **QPluginLoader** implemented.

triplus, 2020-03-20:
Thanks for the detailed description. We discuss such things occasionally on the forum (for all platform binaries). Realistically likely we won't be implementing such integrated updating support in foreseeable future. The strategy of using an external updating application works good. Once you embed an updater tools you have to start taking care of potential update related issues. And overall keep everything in good shape. That is not desirable on our side ATM.

@sgrogan 

Could you add a note mentioning:

https://github.com/antony-jr/AppImageUpdaterBridge

Directly to releases page?

In short i feel that people using AppImages can handle a drag and drop operation just fine, for updating. And that already is an elegant and easy solution without introducing any additional maintenance burden and potential issues.

triplus, 2020-03-20:
The report can stay opened for possible future re-evaluation. If somebody else would like to go ahead with this. I am OK with that.

luzpaz, 2020-03-20:
hmm... I don't think this is such a heavy burden. If it breaks then we comment it out. I say we try it and see. This is an elegant solution in itself. I vote to test it out

triplus, 2020-03-20:
Sure, go ahead. I don't object others trying it out.

triplus, 2020-03-20:
If at some point we decide to move forward, what we can do is to first pack the library on Conda and for the AppImage to pick it up.

P.S. There is a plethora of Python developers using Linux on the FreeCAD forum and after they will gain access to such library from the FreeCAD Python console ...

looooo, 2020-03-21:
I tried with PySide2 but this wasn't successful:

```
>>> from PySide.QtCore import QPluginLoader
>>> path = os.path.join(os.environ["CONDA_PREFIX"], "lib", "libAppImageUpdaterBridge.so")
>>> from PySide.QtCore import QPluginLoader
>>> loader = QPluginLoader()
>>> loader.setFileName(path)
>>> loader.load()
False
>>> loader.errorString()
"Failed to extract plugin meta data from '~/miniconda3/envs/freecad/lib/libAppImageUpdaterBridge.so'"
```

any ideas?

antony-jr, 2020-03-21:
@looooo Did you build the plugin with Qt version lesser than or equal to the Qt version which loads the plugin?

luzpaz, 2020-03-21:
Nice, we also have a [conda recipe](https://forum.freecadweb.org/viewtopic.php?f=8&t=44324&p=379151#p379151) for the appimage-updater-bridge thanks to loooo

antony-jr, 2020-03-21:
@looooo I guess you built the library with the stable version **v1.1.6**, if that is true then the problem is that the plugin feature is in alpha, it only exists in the ```plugin``` branch. You have to use that branch to get a valid Qt Plugin.

antony-jr, 2020-03-28:
Please see https://forum.freecadweb.org/viewtopic.php?f=8&t=44324&start=10#p381756

looooo, 2020-03-29:
thanks a lot for your work. The package is updated and works:
```
>>> from PySide.QtCore import QPluginLoader
>>> path = os.path.join(os.environ["CONDA_PREFIX"], "lib", "libAppImageUpdaterBridge.so")
>>> from PySide.QtCore import QPluginLoader
>>> loader = QPluginLoader()
>>> loader.setFileName(path)
>>> loader.load()
True
```

probonopd, 2020-05-23:
Wow, great progress. What do people think, can this be used?

antony-jr, 2020-05-23:
> Wow, great progress. What do people think, can this be used?

The library can be used in Python with ease but we are still testing it in FreeCAD console which seems to have some issues. 

See https://antony-jr.github.io/AppImageUpdaterBridge/docs/PyQt5PluginExample.html

antony-jr, 2020-05-23:
The issue seems to be specific with FreeCAD's console, Please see https://forum.freecadweb.org/viewtopic.php?f=8&t=44324&p=401553#p401553

antony-jr, 2020-05-24:
@looooo I guess we should start using the GUI directly, considering https://forum.freecadweb.org/viewtopic.php?f=8&t=44324&p=401553#p401553

I'm almost done with the changes.

antony-jr, 2020-09-22:
@looooo @luzpaz Please see https://forum.freecadweb.org/viewtopic.php?f=8&t=44324&p=432932#p432932 for the error on updating when AppImage Launcher is installed.

antony-jr, 2020-09-30:
@probonopd Starting from FC 0.19.21514 (Development Build) there now exists an AppImage section found via **Edit → Preferences → AppImage.** 

[image]

Special thanks to @looooo

probonopd, 2020-10-01:
Congratulations and thank you to everyone involved.

luzpaz, 2020-10-22:
Thanks to all invovled

closed 2020-10-22 by luzpaz