FreeCAD/FreeCAD-Bundle #28: Running as root without --no-sandbox is not supported.

revolverve, 2019-10-30:
In distro :bionicpup64 8.0
these appimage  did load Ok
-FreeCAD_0.19-17068-Linux-Conda_Py3Qt5_glibc2.12-x86_64
-FreeCAD_0.18-16131-Linux-Conda_Py3Qt5_glibc2.12-x86_64
but 
these 
-FreeCAD_0.19-18614-Linux-Conda_Py3Qt5_glibc2.12-x86_64
-FreeCAD_0.19-18522-Linux-Conda_Py3Qt5_glibc2.12-x86_64
wont run : 
ERROR:zygote_host_impl_linux.cc(89)] Running as root without --no-sandbox is not supported. See https://crbug.com/63818
https://appimage.org/ on home page does state appimages.
¨..Can also run in a sandbox like Firejail...¨
Search fc forum found nothing about it and never had this error with that distro before.
Thanks
____

probonopd, 2019-10-30:
What has changed between those builds? Has FreeCAD started using QtWebEngineProcess? Are you running FreeCAD as root?

revolverve, 2019-10-30:
probonopd 

> Are you running FreeCAD as root?

in bionicpup64 or other puppy linux variants,yes.

> Has FreeCAD started using QtWebEngineProcess?

Consider a ¨midway-end-user¨ here ..minimum coding and no Qt at all.i  hesitate to open a new issue before asking devs what changed..

but 
-looking in non loading 0.19 appimage i did saw qtwebenginepluging
-did not look or ask if old ¨working¨ 0.19 does
-now looking at different github other appimage project i do find a lot of info about  sandbox related problems,usually more web oriented ones as Electron v5.

probonopd, 2019-10-30:
PuppyLinux is not one of the distributions we are generally targeting with AppImage, so unforeseen things may happen. Mainstream distributions like Ubuntu, Debian, Fedora, openSUSE are more likely to work with AppImages.

revolverve, 2019-10-30:
probonopd 
No big surprise ...But most appimage i've tested  in recent puppy linux ubuntu based variants worked  ,almost, out-of-the-box..
Did tried to mess around with what i found as a potential solution from other projects.
Editing Apprun...too easy... get a unrecognized invalid command--no-sandbox  
and create a .desktop the same..
It look like i'm back to school...
if i got to dig in python or Qt environment variable or not...:)
But what changes since -FreeCAD_0.19-17068-Linux-Conda_Py3Qt5_ which loaded ok??
Should i close the issue ,be wise and come back if appimage 0.19 release is problematic?
Fc is already so big ,would be sad to slow its dev pace in the run ...

sgrogan, 2019-10-30:
> What has changed between those builds? Has FreeCAD started using QtWebEngineProcess?

Yes

The way it works is that @looo provides weekly builds on Conda, @triplus has set up a Cron job on Travie-CI that packages these builds as an .AppImage.
Most of the magic happens here: https://github.com/FreeCAD/FreeCAD-AppImage/blob/master/conda/linux_dev/linux_dev.sh

revolverve, 2019-10-30:
probonopd
puppy linux users of a least one variant since 0.15 been able to use  ,first in their own ¨squashfs full dependencies¨ packing,now offer its appimage ,because it works ,and thank you all for it ,you now scratch your heads  ¨pack-squashing¨ it ...
  if potential fixes applicable  i can try test some,if it fall in the range of my limited knowledge .
Latest stable fc appimage do run here... wont fee left alone anyhow if dev branch dont!!
off topic a little;finally got the AppImageUpdater to work!!bring some more release to test!!
Thanks all.

sgrogan, 2019-10-30:
`Latest stable fc appimage do run here... wont fee left alone anyhow if dev branch dont!!'

stable uses an older conda env with QTWebKit
dev uses QTWebEngine

revolverve, 2019-10-31:
> dev uses QTWebEngine

ok,and ,
-so it will be for future stable release? i guess?
Bump in https://doc.qt.io/qt-5/qtwebengine-platform-notes.html:
¨To explicitly disable sandboxing, the QTWEBENGINE_DISABLE_SANDBOX environment variable can be set to 1 or alternatively the --no-sandbox command line argument can be passed to the user application executable.¨
Their user application executable solution get;
unrecognised option '--no-sandbox'

revolverve, 2019-10-31:
Got it to load using this modified google-chrome script on extracted appimage;
#!/bin/sh
   APP=/mnt/sda5/FreeCAD_0-revised-on-2019-10-30T12-12-25.19-18614-Linux-Conda_Py3Qt5_glibc2.12-x86_64.AppImage.extracted/AppRun
#script to run $APP as spot...
ARGS=""
[ "$1" ] && while [ "$1" ]; do ARGS="$ARGS \"$1\""; shift; done
if [ $(id -u) -eq 0 ]; then
 [ $XAUTHORITY ] && cp $XAUTHORITY /root/spot/.Xauthority 2>/dev/null
 touch /root/spot/.Xauthority
 #following line is mostly there to catch any root:root files that may have got copied in...
 find /root/spot \( -not -user spot -or -not -group spot \) -exec chown -h spot:spot {} \; &
 export XAUTHORITY=/root/spot/.Xauthority  
 export XDG_CONFIG_HOME=/root/spot/.config
 export XDG_CACHE_HOME=/root/spot/.cache
 export XDG_DATA_HOME=/root/spot/.local/share
 exec su spot -s /bin/sh -c "\"$APP\" $ARGS"
else #precaution
 exec "$APP" "$ARGS"
fi

Next step will be to get it to load without extracting it,get the ¨fuse¨cannot mount appimage...
is it close enough to closed the issue?
if it really was a valuable one...

probonopd, 2019-11-01:
> stable uses an older conda env with QTWebKit
dev uses QTWebEngine

File a bug with QTWebEngine then. It seems to have worse system requirements than its predecessor.

revolverve, 2019-11-02:
probonopd 
in the name of the few ¨root distro¨ i can try that.
Doubtful i have Qt's knowledge to help them a lot... 
Could be better rewriting  song...:)
You need coding
Baby I'm not fooling
I'm gonna send ya
Back to schooling
A-way down inside
A-honey you need it
I'm gonna give you my code
I'm gonna give you my code
.....sorry ,seriously ,i can look at it or could try to find a Qt schooled dev to help . 
So, do you keep this issue ¨alive¨ and post link to its Qt side?
Thanks you all ,your work is very appreciated.

revolverve, 2019-11-02:
Hmmm...slow start ...
Bug Reports
QtWebEngine component in bugreports.qt.io
https://bugreports.qt.io/browse/QTBUG/component/20522
Bump..
is QTWEBENGINE  version specific ??
Wow, qt bug reports is like funfairs mirrors houses...

revolverve, 2019-11-06:
https://bugreports.qt.io/browse/QTBUG-79710 
If it goes somewhere.

revolverve, 2019-11-07:
be it that;
-  the Qtwebengine/Qtwebkit choice in fc appimage build for future release be irreversible or not   and there are workarounds  for root glitch it cause here
- i wont dig more qtwebengine ,add ¨ the fears* of having to discuss insecure data$-root feeling¨ in a field where i must admitting that i will stay overwhelm in a near future   by my  limited capacities to help...* of my propensity to,again, ¨put my feet in my mouth¨;)
- its not the right place for a wiki ...
-the workload you guys already got  with freecad and its related must be enough.

maybe you should close this issue?
  and thank you all for your work and patience

triplus, 2019-11-07:
Hi  @revolverve 

As explained above, we have migrated to using QtWebEngine in FreeCAD 0.19, before we used QWebKit. The migration happened due to Qt deprecating QWebKit. That is where the difference comes from, comparing FreeCAD 0.18 and FreeCAD 0.19 AppImage. In addition you are likely using Assembly 3 AppImage in your tests above?

Anyway, by default we for sure won't be disabling QtWebEngine sandboxing. I feel that trying to detect, if the AppImage is run in root environment or not, and to disable sandboxing after, that likely isn't a good choice. But i understand, you would still like to run FreeCAD on Puppy Linux. Have you already tested the mentioned environment variable? Something like:

```
export QTWEBENGINE_DISABLE_SANDBOX=1
./FreeCAD.AppImage
```

P.S. If that will work it should be considered as a valid solution, for special use cases, like yours.

probonopd, 2019-11-07:
Why wouldn't it be a good idea to disable the sandbox when root detected? Preventing stuff from being run as root is counter to what "root" actually is, isn't it?

revolverve, 2019-11-08:
Hi triplus ,

> Something like:
export QTWEBENGINE_DISABLE_SANDBOX=1
./FreeCAD.AppImage

That is pretty straightforward!!!
Laughable me,add the line to AppRun is the only thing i did not tried!!?!!²!!

 It for sure be consider a valid solution for puppy linux users you can live without sandboxing.

 Thank you all for your work and patience.

triplus, 2019-11-12:
@probonopd 

I didn't investigate technical details in depth, but the way i understand it is Chromium developers decided sandboxing isn't supported, when running a Chromium based application as root. This therefore isn't related to AppImage. Other applications, such as Chrome browser, should behave the same. In the end therefore this is an end user (informed) decision, to disable sandboxing and use an application or not to do that.

@revolverve 

Good to hear that a workaround is working. FreeCAD currently isn't utilizing QtWebEngine heavily. Therefore disabling sandboxing isn't as problematic, security vise, as doing that for an application like a web browser. There is Web workbench in FreeCAD but i guess you don't plan to use it as a general purpose web browser.

closed 2019-11-12 by triplus

revolverve, 2019-11-13:
@triplus 
Yep,thats what i end up with ,no sandboxing= use the fc web engine to pay online bills at your own risk...;)