FreeCAD/FreeCAD-Bundle #30: adding options to launch other freecad-related appilications

looooo, 2019-12-11:
What do you think about adding the possibility to add other applications (like python, freecadcmd) with the appimage? I guess it's easy to add some switches to the initialization script.

probonopd, 2019-12-18:
As for `freecadcmd`I agree that it should go into the AppImage. It could be made possible to be invoked with something like

```
if [ ! -z $APPIMAGE ] ; then
  BINARY_NAME=$(basename "$ARGV0")
else
  BINARY_NAME=$(basename "$0")
fi
if [ ! -z "$1" ] && [ -e "$HERE/bin/$1" ] ; then
  MAIN="$HERE/bin/$1" ; shift
elif [ ! -z "$1" ] && [ -e "$HERE/usr/bin/$1" ] ; then
  MAIN="$HERE/usr/bin/$1" ; shift
elif [ -e "$HERE/bin/$BINARY_NAME" ] ; then
  MAIN="$HERE/bin/$BINARY_NAME"
elif [ -e "$HERE/usr/bin/$BINARY_NAME" ] ; then
  MAIN="$HERE/usr/bin/$BINARY_NAME"
else
  MAIN="$HERE/usr/bin/freecad"
fi
```

https://gitlab.com/inkscape/inkscape/blob/31dae0dc04d6bef0fc93706711ea918ce5187c7c/packaging/appimage/AppRun#L8-29

As for Python, it depends:
Bundling it means everything is working as tested out of the box, but any Python modules installed by the user in their local Python installation will not show up in FreeCAD and since the AppImage is read-only, they cannot add Python modules to the bundled instance (without unpacking the AppImage).

triplus, 2020-02-08:
Thanks for the info. I added such support: https://github.com/FreeCAD/FreeCAD-AppImage/commit/e328a1ee60d59f111b2a270d0c2b49556ce3a62e

Some examples:

```
./FreeCAD.AppImage
./FreeCAD.AppImage freecadcmd
./FreeCAD.AppImage python --version
./FreeCAD.AppImage python
./FreeCAD.AppImage python ./test.py
./FreeCAD.AppImage ccx -version
./FreeCAD.AppImage ccx -i test
```

Closing issue.

closed 2020-02-08 by triplus