FreeCAD/FreeCAD-Bundle #31: Bundle's should include license

looooo, 2020-01-02:
I guess the bundles needs to have a license included. Most likely the license should be gpl as gpl-libraries are definitely included (eg.: gmsh)

triplus, 2020-01-28:
Hi,

AppImage is in my opinion an aggregate work. Therefore it doesn't need to be GPL licensed by itself.

triplus, 2020-01-28:
Forum discussion:

https://forum.freecadweb.org/viewtopic.php?f=8&t=42853

looooo, 2020-02-08:
Talking to guys from conda-forge the overall response seems to be:
The bundle is not a meta-package and therefor doesn't need to be licensed itself. But we should add a list of all included packages with the following specifications:

- package-name
- version
- license
- development-url
- source-url
- url to build-scripts