FreeCAD/FreeCAD-Bundle #35: Bundle-name should include version-strings of occt/python/qt

looooo, 2020-02-19:
There is a need for different versions [1].

suggestion for filename: `FreeCAD_0.19-19667-Linux-Conda_occt7.4_py3.8_qt5.12_glibc2.12-x86_64.AppImage`

[1] https://forum.freecadweb.org/viewtopic.php?p=370235#p370235

triplus, 2020-02-19:
Feel free to do it but in my opinion this is just getting too complex. Heaps of Python code to maintain, AppImages with different versions of dependencies included, AppImage names long as a novel. There is just no justification for producing a new official AppImage because of a single bug somebody has. That is not a packaging issue.

looooo, 2020-02-19:
Revert?

triplus, 2020-02-19:
In my opinion there should be only one "official AppImage". If it needs to have an OCCT 7.3 included, due to a wide spread and general bug, that is not a problem, but if the issue is not a general one, people experiencing it will just need to cope with it and find a different solution. In addition i didn't check but there is an update information embedded in the AppImage name, that would need to be taken into consideration too. As all builds are now Py3/Qt5 we could consider removing that bit of information from the AppImage name. It was important when there was still an PY2/Qt4  option, but that isn't the case anymore.

looooo, 2020-02-19:
Ok I revert and remove the py/qt stuff. I thought this will allow to offer different appimages (py37+occt7.3, py38 + occt7.4) but as a consequence of not understanding the logic behind the upload, I guess this will not work.

In addition: do we need the glibc additional information. We do not provide an appimage for a different glibc version...

triplus, 2020-02-19:
The main problem is when things stop working. The .travis.yml file itself is already a bit complex, there is some Python code already, in addition i suspect we might be starting to hit a "network reliability size limit," as in the last two days, after i added documentation, daily AppImage and stable DMG didn't deploy successfully. It is too early to say, and best to wait a couple of days, something else on Travis or GitHub side could be involved. I see that GitHub is introducing some changes, therefore likely that could be affecting reliability. In addition if somebody reports an issue against one version of AppImage, or another ... this just becomes way too complex. Currently this logic is used for update information:

`FreeCAD*glibc2.12-x86_64.AppImage.zsync `

The rationale to include the glibc information was, this is the component that determines on how old Linux distributions AppImage will work. Older the glibc version, older Linux distributions are expected to work. And if glibc version increases, for update tools to not automatically update. There was a PPA AppImage available at that time, using a different glibc version. Update information therefore had to differ a bit. As there is only one AppImage now, glibc information could get omitted.

looooo, 2020-02-19:
> `FreeCAD*glibc2.12-x86_64.AppImage.zsync`

Damn, I renamed this to be consistent with the osx bundle. So I guess updating the appimage is curretly broken:

`FreeCAD_0.19-19667-Linux-glibc2.12-x86_64-conda.AppImage`

triplus, 2020-02-19:
Yes, this last change won't do: https://github.com/FreeCAD/FreeCAD-AppImage/commit/962f87f21145b8a1ad651cdda9c3e1ee16f4dd87

As currently an AppImage updater tool, based on the update information embedded in existing AppImages out there, is looking for this:

`FreeCAD*glibc2.12-x86_64.AppImage.zsync `

We would therefore need to change the update information to search for this:

`FreeCAD*glibc2.12-x86_64-conda.AppImage.zsync`

If we will change it in the future, best to do that on a release day. As changing it now will only make AppImage updater tools to complain about not finding the updates anymore. The removal of Py3/Qt5 information looks good to me.

triplus, 2020-02-19:
Note that i went ahead and reverted the file name to using current AppImage update info:

https://github.com/FreeCAD/FreeCAD-AppImage/commit/68636bfcf95163e9764860b09a713d59464e3419

looooo, 2020-02-19:
Thanks and sry for the mess. Best to talk about renaming the appimage after the 0.19 release.

closed 2020-02-19 by looooo