FreeCAD/FreeCAD-Bundle #40: Make AppImage Updater dialog user friendly

luzpaz, 2020-10-16:
Add the option to hide the backend log of progress (verifying zsync, identifying new blocks to download etc..) and keep the GUI simple.

antony-jr, 2020-10-17:
@luzpaz Actually there is a inbuilt GUI that is not used. It's the same one used in the unofficial updater. It shows you release notes too and a simple progress.

luzpaz, 2020-10-17:
@antony-jr how does one enable it ?

antony-jr, 2020-10-17:
Oops, Actually it's only usable in plugin in v2.x.

antony-jr, 2020-10-26:
@luzpaz The new version released, You can see the docs [here](https://antony-jr.github.io/QAppImageUpdate). But before you start using the new version **I think we might need to change the conda recipe** because the project was renamed. But Github takes care of all redirects but still some build specific stuff breaks.

Here is how you do it,

```
# v2.x code

# Get constants from plugin itself.
Action_CheckForUpdate = updater.getConstant("Action::CheckForUpdate")
Action_UpdateWGUI = updater.getConstant("Action::UpdateWithGUI") # Start updater with gui

def handleFinished(info, action):
    if action == Action_Update:
       print("New version Details:: {}".format(info))
       app.quit()
    elif action == Action_CheckForUpdate:
       if info["UpdateAvailable"].toBool():
          # updater.setGuiFlag(updater.getConstant("GuiFlag::Default"))
          # You can customize the GUI to some extent with flags. See docs.
          updater.start(Action_UpdateWGUI)
       else:
          print("You already have the new version")

# Connect
updater.finished.connect(handleFinished)

# Start the update process 
updater.start(Action_CheckForUpdate) 
```

If you want BitTorrent Support then FreeCAD need to use https://github.com/antony-jr/MakeAppImageTorrent to make torrent file at build time and upload it to the releases along with the .zsync file. Now you need to add ```-DDECENTRALIZED_UPDATE_ENABLED=ON``` flag to cmake to enable building QAppImageUpdate with torrent support. **Also you need to build libtorrent at build time. ping me if you want help.**

Then you just need to replace ```Action::UpdateWithGUI``` with ```Action::UpdateWithGUIAndTorrent```.

luzpaz, 2020-10-27:
@looooo can we try this?

antony-jr, 2021-03-11:
@luzpaz The PR is ready, Once the review is done, we can get user friendly GUI for updating https://github.com/looooo/freecad.appimage_updater/pull/5

antony-jr, 2021-04-20:
@luzpaz We can close this.

closed 2021-04-21 by luzpaz