FreeCAD/FreeCAD-Bundle #48: Preferences error

mwganson, 2020-10-22:
When you go to Edit menu -> Preferences and press OK you get this error:

File "/tmp/.mount_FreeCApBjojb/usr/lib/python3.8/site-packages/freecad/appimage/init_gui.py", line 22, in saveSettings
preference_tabel.SetBool("appimage_auto_udpate", bool(self.form.always_check_updates.isChecked()))
<class 'AttributeError'>: ''PySide2.QtWidgets.QWidget' object has no attribute 'always_check_updates'

This is using 0.19 Revision 22756.  I can't copy/paste the full info because it's in a virtual machine.

mwganson, 2020-10-22:
Reference to forum topic:

https://forum.freecadweb.org/viewtopic.php?f=3&t=51322

antony-jr, 2021-04-20:
@looooo Please close this. This is resolved. @mwganson Please reopen if this exist even now(i.e test the latest version) which is not likely though.

closed 2021-04-20 by looooo