FreeCAD/FreeCAD-Bundle #49: Distribute FreeCAD AppImages via Bittorrent (experimental feature)

luzpaz, 2020-10-28:
@probonopd tweeted https://twitter.com/probonopd/status/1320054292275933184  
@antony-jr opened https://github.com/AppImage/zsync2/issues/24#issuecomment-395096715  
Here are the instructions on how to use:https://github.com/antony-jr/MakeAppImageTorrent  

[fosstorrents](https://fosstorrents.com/softwares/cad/freecad) would be a great way to distribute them.

antony-jr, 2020-10-28:
You should also consider the privacy concerns by privacy advocates and AppImage core devs. See https://github.com/AppImage/AppImageKit/issues/175

aprice2704, 2020-12-19:
I second this proposal. I'm currently trying to download the appimage and finding it very unreliable -- it typically stalls out after 200MB or so. Neither chrome or wget seems to get it done. Is github throttling the account perhaps? In any event, distributing via BT with signatures posted on the dl page would be an excellent solution.

antony-jr, 2020-12-19:
> I second this proposal. I'm currently trying to download the appimage and finding it very unreliable -- it typically stalls out after 200MB or so. Neither chrome or wget seems to get it done. Is github throttling the account perhaps? In any event, distributing via BT with signatures posted on the dl page would be an excellent solution.
> 
> I typically get linux iso images this way and it works very well :)

@aprice2704 Yes github kind of limits the download speed and some times blocks you if you request a lot of things without interval. But you can get the AppImage via Bittorrent by FossTorrents -> https://fosstorrents.com/softwares/cad/freecad/ now. 

@luzpaz I suppose you proposed to use Bittorrent for updates too in Linux AppImages since FossTorrents already distributes FreeCad now.

luzpaz, 2020-12-19:
> I suppose you proposed to use Bittorrent for updates too in Linux AppImages since FossTorrents already distributes FreeCad now.  

@antony-jr does it work ?

aprice2704, 2020-12-20:
I did finally get the download from amazon, it took about 30 wgets (thankfully using the -c flag, something worth mentioning on the d/l page I think) and about 4hrs.

Thanks for the link @antony-jr , https://fosstorrents.com/softwares/cad/freecad/ is a very nice page that should imo be linked from the main Freecad d/l. Thanks for trying @luzpaz :), but the link you posted doesn't work :P I didn't find it through googling either.

I should mention that I gave up in frustration on Freecad some months ago because, in the 18.4 appimage I downloaded, the add-on manager did not work; therefore I could not do sheetmetal (my main desire) nor use my 3dconnexion mouse. I think the 'automatic constraints' button in sketcher did not work initially for me -- but claimed to be switched on. These three things made, what I now believe to be an amazing piece of software, seem like utter garbage :(

If it hadn't been for changing my laptop to linux, making Fusion 360 basically unusable on it, I doubt I'd have come back to try again. 0.19 is great!

Hoping someone long-in-the-project will pay attention to such 'first impression' issues, it is easy to forget them when you're used to the thing. Freecad now seems awesome.

aprice2704, 2020-12-20:
Wierd https://fosstorrents.com/softwares/cad/freecad/ is the *only* page on fosstorrents I can get to show, all others show "This site can’t be reached The connection was reset." <shrug>

luzpaz, 2020-12-20:
@aprice2704 of course, thanks for the feedback! I can't reproduce the issue you have with fosstorrents, site works fine for me.

antony-jr, 2020-12-20:
> > I suppose you proposed to use Bittorrent for updates too in Linux AppImages since FossTorrents already distributes FreeCad now.
> 
> @antony-jr does it work ?

Yes. The library is capable of doing it but the python library which is packaged in the AppImage uses v1.1.9 which does not support it. If I have some time to spare I plan to update the library to use v2.x and help change the GUI.

Also I was planning on making the changes to FreeCAD after I have done porting the example Application to v2.x (AppImage Updater) -> https://github.com/antony-jr/AppImageUpdater/tree/mk2 (Kind of trying to make it a viable alternative)

luzpaz, 2021-03-01:
@antony-jr hi! Hows the progress going on this? Thanks for all you work and the tip about the ENV variable issue (now fixed in 1ab7d8d) !

antony-jr, 2021-03-01:
> @antony-jr hi! Hows the progress going on this? Thanks for all you work and the tip about the ENV variable issue (now fixed in [1ab7d8d](https://github.com/FreeCAD/FreeCAD-AppImage/commit/1ab7d8daa271de2d7874fbf1a2e0a4f911604cf2)) !

Thanks for showing interest in this(I can see people wanting this since github started to rate limit very strictly now). I'm currently working on this. Hopefully soon enough you can use this feature at least in my updater very soon. Here is how it looks now and you can enable BT as opt-in,

[image]

**PS:** *I can't show you everything as of now, but I will definitely ping you when it's ready.*

antony-jr, 2021-04-02:
@luzpaz Now everything in the library is ready and the updater is released https://twitter.com/antonyjr0/status/1377885816110772229

**Now to enable the decentralized update for FreeCAD, we just need to upload the .torrent generated from MakeAppImageTorrent or from Foss Torrents but as you know this requires a lot of discussion from other core FreeCAD members. But the torrent file has to be generated on each build so MakeAppImageTorrent is the only choice I guess since Foss Torrents does not have any automation** 

Then we just need to add a check box in the preferences to allow to use BitTorrent protocol (This is to obey GDPR) and also warn users not to use it if they are in a metered connection. Also a seed button to seed the FreeCAD AppImage they have locally to other users (The library has routines for seeding and other stuff too).

luzpaz, 2021-04-02:
This is awesome news!  
CC 
@kkremitzki 
@yorikvanhavre

luzpaz, 2021-04-07:
@looooo lets test this shall we ?

antony-jr, 2021-04-07:
> @looooo lets test this shall we ?

Well @luzpaz for starters we can just distribute torrents in a separate repo for FreeCAD to enable torrent download/update for testing. It would be less pain for the maintainers and flexible for us too. You can use the AppImage Updater to use the torrent update feature.

luzpaz, 2021-04-20:
We've renamed the repo we're serving FreeCAD dev builds from FreeCAD-AppImage to FreeCAD-Bundle
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds