FreeCAD/FreeCAD-Bundle #50: Can't run FreeCAD AppImage, GUI broken, Updating doesn't work

XenGi, 2021-01-09:
I'm tryting to run the FreeCAD AppImage and it starts but the GUI is broken. I tried using the AppImageUpdate-x86_64.AppImage to maybe update it but that tool doesn't even start:
```
$ ./AppImageUpdate-x86_64.AppImage 
AppImageUpdate version 1-alpha (commit 1f52c72), build 21 built on 2020-12-13 07:19:11 UTC

(AppRun.wrapped:179591): Gtk-WARNING **: 01:09:14.976: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../gtk/gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Adwaita/16x16/status/image-missing.png: Fatal error reading PNG image file: bad parameters to zlib (gdk-pixbuf-error-quark, 0)
Bail out! Gtk:ERROR:../gtk/gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Adwaita/16x16/status/image-missing.png: Fatal error reading PNG image file: bad parameters to zlib (gdk-pixbuf-error-quark, 0)
fish: “./AppImageUpdate-x86_64.AppImage” terminated by signal SIGABRT (Abort)
```

When I try the command line version this happens:

```
./appimageupdatetool-x86_64.AppImage ~/.local/bin/FreeCAD_0.18-16146-rev1-Linux-Conda_Py3Qt5_glibc2.12-x86_64.AppImage 
Checking for updates...
Fetching latest release information from GitHub API
Could not find any artifacts in release data. Please contact the author of the AppImage and tell them the files are missing on the releases page.
ZSync URL not available. See previous messages for details.
Update check failed, exiting!
```

When I start FreeCAD it looks like this:
[image]

antony-jr, 2021-01-24:
@XenGi Does this issue still exists. Can you please post what GNU+Linux distribution and Desktop environment you are using? Also what version of FreeCAD you are using (You can get the information typically from Help -> About)

luzpaz, 2021-02-19:
@XenGi please follow-up

XenGi, 2021-02-20:
I'm using `FreeCAD_0.19-23578-Linux-Conda_glibc2.12-x86_64.AppImage` atm and that seems to work. I guess it was some misconfiguration on my side.

closed 2021-02-20 by XenGi

Afterschnitzel, 2021-07-19:
I have the same issue when compile FreeCAD or install it via apt.
AppImage works.
I am using Ubuntu 21.04 and try to run FreeCAD 0.19