FreeCAD/FreeCAD-Bundle #51: upload broken

looooo, 2021-01-21:
The upload is broken and needs to be fixed. @bblacey we need your help. Otherwise maybe @triplus can help.

ok1vaw, 2021-01-24:
Still the same, it writes following error with chocolatey:
ERROR: The remote file either doesn't exist, is unauthorized, or is forbidden for url 'https://github.com/FreeCAD/FreeCAD/releases/download/0.19_pre/FreeCAD_0.19.23721-Win-Conda_vc14.x-x86_64.7z'. Exception calling "GetResponse" with "0" argument(s): "The remote server returned an error: (404) Not Found."
The upgrade of freecad was NOT successful.
Error while running 'C:\ProgramData\chocolatey\lib\freecad\tools\chocolateyInstall.ps1'.
 See log for details....

luzpaz, 2021-02-03:
Can confirm (Manjaro Linux) 
```
$ ./appimageupdater-x86_64.AppImage FreeCAD_0.19-23756-Linux-Conda_glibc2.12-x86_64.AppImage 
AppImageUpdater (commit 9d4a0c5 ), build 151 built on Tue Sep 29 21:51:19 UTC 2020
Copyright (C) 2020, Antony Jr.

 [ 05:44:37 ] (💡 Info )      : [Running on] FreeCAD_0.19-23756-Linux-Conda_glibc2.12-x86_64.AppImage
 [ 05:44:44 ] (🛑 Fatal )     : The remote content was not found at the server (HTTP error 404)

Thank you for using AppImageUpdater 💖, if you find this project useful then please
consider to 🌟 this project at https://github.com/antony-jr/AppImageUpdater
```

antony-jr, 2021-04-20:
We can close this as this is resolved.

closed 2021-04-20 by looooo

ok1vaw, 2021-04-20:
Updating with chocolatey of the last version two days ago did not work again with following error:
ERROR: The remote file either doesn't exist, is unauthorized, or is forbidden for url 'https://github.com/FreeCAD/FreeCAD/releases/download/0.19.1/FreeCAD-0.19.1.a88db11-WIN-x64-installer.exe'. Exception calling "GetResponse" with "0" argument(s): "The remote server returned an error: (404) Not Found."

looooo, 2021-04-20:
@ok1vaw please wait a bit. We still have to change a few things and will update the update information of downstream services once that's done.
https://forum.freecadweb.org/viewtopic.php?f=10&t=57440&start=20