FreeCAD/FreeCAD-Bundle #57: Can't run FreeCAD AppImage from command line

teamblubee, 2021-03-02:
I am having similar issue to [this one](https://github.com/FreeCAD/FreeCAD-AppImage/issues/50)

AppImage and FreeCAD
```shell
AppImageUpdate-x86_64.AppImage  FreeCAD_0.19-24267-Linux-Conda_glibc2.12-x86_64.AppImage

/AppImageUpdate-x86_64.AppImage ./FreeCAD_0.19-24267-Linux-Conda_glibc2.12-x86_64.AppImage 
AppImageUpdate version 1-alpha (commit 1f52c72), build 21 built on 2020-12-13 07:19:11 UTC
Fetching release information for tag "" from GitHub API.
GitHub API request failed!
Could not find any artifacts in release data. Please contact the author of the AppImage and tell them the files are missing on the releases page.
ZSync URL not available. See previous messages for details.

```

This is the info inside the details of the gui
```shell
Fetching release information for tag "" from GitHub API.
GitHub API request failed!
Could not find any artifacts in release data. Please contact the author of the AppImage and tell them the files are missing on the releases page.
ZSync URL not available. See previous messages for details.
```

uname -a
```shell
Linux xxx 5.8.0-7642-generic #47~1612288990~20.10~b8113e7-Ubuntu SMP Wed Feb 3 02:27:26 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

about my release
```shell
NAME="Pop!_OS"
VERSION="20.10"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.10"
VERSION_ID="20.10"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=groovy
UBUNTU_CODENAME=groovy
LOGO=distributor-logo-pop-os
```

luzpaz, 2021-03-02:
This should be fixed with this latest commit 1ab7d8daa271de2d7874fbf1a2e0a4f911604cf2  (right @looooo ?)  
The new appimages haven't been generated yet. Once they are, the iteration after them should work.

teamblubee, 2021-03-03:
Thank you @luzpaz can you close this issue after the new app images are available?

luzpaz, 2021-03-03:
@looooo can you add `closes #57` when you make the appropriate commit?

looooo, 2021-03-03:
I am not sure if my commit addresses this issue. We might see some disconnection of the updating tools as we move to 0.20. This will be solved in the future by naming the pre-builds weekly-builds (See the release page). Releases (eg. 0.19) will be branches which uploads to corresponding releases. So the appimage-updating-tools will also work for releases. At least this is the theory.

The appimage updating tools will be disconnected. So downloading a 0.20-prebuild will be necessary if you want to have this functionality restored. But I am not sure this is connected to this issue. We still have some issues from the transition to github action.

closed 2024-07-05 by adrianinsaval