FreeCAD/FreeCAD-Bundle #59: libqhull_p.so.7 missing from AppImage v0.19-24267

atererus, 2021-03-15:
fatal error in AppImage FreeCAD 0.19 Revision: 24267 (Git) on Fedora 33 Workstation 5.10.19-200.fc33.x86_64
to reproduce : ./FreeCAD_0.19-24267 --run-test 0
to solve : yum install libqhull_p
IMHO libqhull_p should be in the appimage
forum : https://forum.freecadweb.org/viewtopic.php?p=487888
tracker : https://tracker.freecad.io/view.php?id=4542

looooo, 2021-03-24:
never heared about this dependency. When was it introduced?

pedromorgan, 2021-04-30:
Getting same error when I select "Reverse Engineering"..

```libqhull_p.so.7: cannot open shared object file: No such file or directory```

AppImage 0.19 on linux.. 24276(GIT) from 11.march 2021

luzpaz, 2021-12-04:
I guess this is the stable 0.19.3 AppImage from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/0.19.3 right?  
Can you reproduce on the dev version ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2022-02-03:
Per the mantis ticket:  
```
1) Start FreeCAD from AppImage downloaded via Github
2) From the Workbench selector select "Reverse Engineering"
3) Observe the error / reverse engineering workbench not being loaded.
```

adrianinsaval, 2024-07-05:
is this still relevant?

atererus, 2024-07-12:
No it is about version 0.19

closed 2024-07-13 by adrianinsaval