FreeCAD/FreeCAD-Bundle #61: TODO

looooo, 2021-03-24:
just a little reminder to make the next release easier:

- update the win-builds to use the 0.19.1 naming scheme
- osx: https://forum.freecadweb.org/viewtopic.php?f=15&t=51345&start=230#p490638

luzpaz, 2021-03-24:
Note: any changes you make may affect downstream services like:
chocolatey, fosstorrents, our website and wiki

looooo, 2021-03-25:
naming switched anyway for the next weekly builds. But this way we can continue having the same name and release to store the bundles. So this will make it easier for other services to work.

antony-jr, 2021-04-01:
@looooo Also I see that you are using tag 0.19.1 (or the version) for AppImage update information for versioned releases but if you use ```latest``` tag then any AppImage updater will take the latest release from the releases and sync to it. For example anything marked with latest flag in releases will be synced. **But this is not a high priority since everything is working as expected for now**.

Please ping me in github if you need any help with AppImage packaging in Github actions.

looooo, 2021-04-01:
@antony-jr what does `latest` reffer to? The latest uploaded appimage? Or the highest tag? Is it guaranteed that 0.19.1 appimage will not update to weekly-builds?

antony-jr, 2021-04-01:
> @antony-jr what does `latest` reffer to? The latest uploaded appimage? Or the highest tag? Is it guaranteed that 0.19.1 appimage will not update to weekly-builds?

Yes. See the below image,

[image]

The updater will use the tag which has that **latest release flag** as seen in the image above. **So in this case 0.19.1** is the latest.

antony-jr, 2021-04-01:
> Is it guaranteed that 0.19.1 appimage will not update to weekly-builds?

No. If the weekly builds are release and it is the highest tag then it is the latest in theory so we just need to make sure that we don't have weekly builds release as the latest.

looooo, 2021-04-01:
@antony-jr thanks. Can you review my commit:
https://github.com/FreeCAD/FreeCAD-AppImage/commit/066d76657513fb579bd47a42df31e9203a280f9f

closed 2024-07-05 by adrianinsaval