FreeCAD/FreeCAD-Bundle #62: Windows build missing lxml python library

KeithSloan, 2021-04-20:
Sorry cannot check as I don't have Windows, but because of issues users are having with the GDML workbench it is obvious that it is no longer in the AppImage

Was in FreeCAD_0.19.19424 and FreeCAD_0.19.19409_x64_Conda_Py3QT5-WinVS2015.7z 

Believe there are parts of FreeCAD that use, but cannot remember which so it will not just be the workbench affected.

looooo, 2021-04-28:
We are working towards a dependency-solver included in the bundles. Maybe that is something interesting for you:

https://forum.freecadweb.org/viewtopic.php?f=42&t=58156

luzpaz, 2022-03-25:
@KeithSloan can GDML make lxml a dependency in the `package.xml` and this import via the Addon Manager?

KeithSloan, 2022-03-25:
Don't understand what would go in package.xml. The metadata.txt already has
````
      workbenches=Part,Mesh
     pylibs=lxml,gmsh

    But I know on MacOS it does not install gmsh python library .

  Also this would only fix for FC 0.20

luzpaz, 2022-03-25:
@KeithSloan have you tested GDML install via the Addon Manager in v0.20dev? Does the addon manager pick those 2 deps up?

> But I know on MacOS it does not install gmsh python library

For v0.20dev?

> Also this would only fix for FC 0.20

Yea that's a good point.

KeithSloan, 2022-03-26:
Just tried with latest weekly of FreeCAD 0.20 i.e.27936 and using addon manager to uninstall, restart, install, restart
Does not install python gmsh library on Mac OS

luzpaz, 2022-03-26:
@KeithSloan we're on r28442 so it's not accurate enough of a test. Can you update ?

KeithSloan, 2022-03-26:
27936 is the latest MacOS version in weekly builds

luzpaz, 2022-03-26:
Yea, it's waaay behind due to a conda build issue :disappointed:

KeithSloan, 2022-03-26:
Also see [https://forum.freecadweb.org/viewtopic.php?f=3&t=67040&p=583387#p583387](url)

luzpaz, 2022-03-26:
hah, that link was deceiving. try clicking it :laughing:

KeithSloan, 2022-04-07:
Well this issue has probably put a CERN researcher off using FreeCAD and the GDML workbench, thanks guys :-(

Message received
````
On Windows, when I try the export in GDML, the following error happens:
 
14:27:26 running with xml.etree.ElementTree
14:27:26 Traceback (most recent call last):
File "<string>", line 4, in <module>
File "C:\Users\paolo\AppData\Roaming\FreeCAD\Mod\GDML\freecad\gdml\exportGDML.py", line 1971, in export
exportGDMLworld(first, filepath, fileExt)
File "C:\Users\paolo\AppData\Roaming\FreeCAD\Mod\GDML\freecad\gdml\exportGDML.py", line 1773, in exportGDMLworld
exportGDML(first, filepath, fileExt)
File "C:\Users\paolo\AppData\Roaming\FreeCAD\Mod\GDML\freecad\gdml\exportGDML.py", line 1745, in exportGDML
exportGDMLstructure(dirPath, fileName)
File "C:\Users\paolo\AppData\Roaming\FreeCAD\Mod\GDML\freecad\gdml\exportGDML.py", line 1686, in exportGDMLstructure
exportElementAsXML(dirPath, fileName, False, 'define', define)
File "C:\Users\paolo\AppData\Roaming\FreeCAD\Mod\GDML\freecad\gdml\exportGDML.py", line 1677, in exportElementAsXML
gdml.append(ET.Entity(elemName))

<class 'AttributeError'>: module 'xml.etree.ElementTree' has no attribute 'Entity'
Hoping can be useful.
````

My Reply
=======
The GDML workbench is developed using the python lxml library it should only fall back to xml if lxml is not available.

FreeCAD also uses lxml and should/was being built with lxml installed, where did you get your install image for FreeCAD from?

I got the image from the FreeCAD website https://www.freecadweb.org/downloads.php

"Your FreeCAD information would be useful FreeCAD | About FreeCAD | Copy to clipboard."

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +148 (Git)
Build type: Release
Branch: Branch_0.19.4
Hash: 476ecf091941bead59b14e44afa6064d5a66afa3
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Italian/Italy (it_IT)

"If you go to the Python Window and enter 'import lxml' I assume you get an error message."

Yes exactly.

luzpaz, 2022-04-07:
macos weekly build has been updated btw

closed 2024-07-05 by adrianinsaval

KeithSloan, 2024-07-06:
Don't understand why this has been closed "Not Planned" very confused.

Are you saying FreeCAD no longer uses lxml?
That having lxml as a python library in the metadata.txt of a workbench is just ignored.
That this will not be supported on macos builds

adrianinsaval, 2024-07-06:
issue has been inactive since 2022... besides from what I gathered on the conversation freecad itself doesn't use it, it is an external addon that uses it. You cannot expect us to ship every library in existance that some addon may need. If you have setup your addon properly and the addon manager is not working to install the dependency please create an issue on the main freecad repo as this repo is only about packaging.