FreeCAD/FreeCAD-Bundle #63: Conda generated 0.19.1 and 0.19.2 macOS builds are broken

luzpaz, 2021-04-25:
> The .dmg of latest compatible versions are non mountable (no filesystem available), maybe because they are formatted as APFS.

Tweet: https://twitter.com/AxiFerr/status/1386109676048183296

Error:  
[image]

Tweet: https://twitter.com/AxiFerr/status/1386364348881412096/photo/1

looooo, 2021-04-26:
There was an issue with 0.19.2 bundles (wrong version of freecad was used for the bundle) This should be fixed now.

luzpaz, 2021-04-26:
> There was an issue with 0.19.2 bundles (wrong version of freecad was used for the bundle)

I think that's unrelated to this report

looooo, 2021-04-26:
> > There was an issue with 0.19.2 bundles (wrong version of freecad was used for the bundle)
> 
> I think that's unrelated to this report

No, because I had to trigger the ci and the package was uploaded a second time. If you have a slow connection there might be an issue with downloading if the bundle was uploaded a second time. But this is only speculating. I tried the 0.19.2 bundle for osx right now and didn't see any issue.

luzpaz, 2021-04-27:
What version of macOS are you testing on? The reporter is using 'El Capitan'

looooo, 2021-04-27:
> MacOSX
Binary stable builds (no compiling necessary). Download the .dmg below, you may need to expand the assets tab to see the available packages. The build will only work correctly with macosx version >= 10.12.

As far as I can tell el capitan is 10.11. So not supported anyway. But the error is different. I really have no clue, sorry.

luzpaz, 2021-04-27:
@vejmarie ping

luzpaz, 2021-04-27:
@looooo on the FreeCAD download page it says we support El Capitan

[image]

luzpaz, 2021-04-27:
@vejmarie would macOS10.11 be supported if we "reformat the .dmg to HFS from APFS" ? Or is it more complicated than that?

looooo, 2021-04-27:
The qt5.12 package from conda-forge does not support 10.11:
https://gitter.im/conda-forge/conda-forge.github.io?at=5bd044fc1c100a4f29d6f0c8
https://github.com/conda-forge/qt-feedstock/blob/master/recipe/conda_build_config.yaml

But I am not entirely sure.

@conda-forge/core is the qt-5.12 package supposed to run on osx 10.11?

luzpaz, 2021-04-27:
Ok, I adjusted the homepage to reflect the correct minimum version of macOS supported.

Thanks for the explanation. That's good enough for me. Thanks @looooo    
Closing ticket (No fix needed).

closed 2021-04-27 by luzpaz