FreeCAD/FreeCAD-Bundle #67: libGL error: MESA-LOADER: failed to open radeonsi

pseudex, 2021-05-16:
**OS:** Fedora34
**GPU:** AMD 5700XT, Ryzen 4800H
**Driver:** amdgpu
**Display server:** x11

**Appimage**: FreeCAD_weekly-builds-24847-Linux-Conda_glibc2.12-x86_64.AppImage

**Problem**
FreeCAD doesn't start. It shows loading up with the splash screen. Crashes then after a while. Started from the terminal I get the following errors:

```
FreeCAD 0.20, Libs: 0.20R24847 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

libGL error: MESA-LOADER: failed to open radeonsi: /tmp/.mount_FreeCAM3nGdH/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib64/dri/radeonsi_dri.so) (search paths /usr/lib64/dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open radeonsi: /tmp/.mount_FreeCAM3nGdH/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib64/dri/radeonsi_dri.so) (search paths /usr/lib64/dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_FreeCAM3nGdH/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib64/dri/swrast_dri.so) (search paths /usr/lib64/dri)
libGL error: failed to load driver: swrast
/tmp/.mount_FreeCAM3nGdH/AppRun: Zeile 41: 37606 Abgebrochen             (Speicherabzug geschrieben) ${MAIN} "$@"
```

I also tried the same on my Notebook which I recently install fresh with fedora34 and got the same error.

**Workaround**
I was only able to get around this issue if I preload the lib in the terminal and start the appimage from there.

`export LD_PRELOAD=/usr/lib64/libstdc++.so.6`

While searching for this error I found other Programs like Mathlab to have similar issues. Therefore this might not be a FreeCAD related one. But from that point on I don't know anything else to do, to get this resolved for good.

I checked the embedded lib with
`strings <pathtolib>/libstdc++so.6 | grep GLIBCXX`
for the version it is  asking for. It will show the version in the output. But somehow the mesa driver tries to load a different one.
The version .19 installed with dnf does work.

looooo, 2021-06-04:
can you test a conda-environment directly?
-  install conda
- create a conda env with: `conda create -n freecad freecad -c freecad/label/dev -c conda-forge`

Maybe this is now a general limitation for conda-packages.

looooo, 2021-06-04:
I guess conda-forge will update the libstdcxx-ng packages. As far as I can tell this should solve this issue.
https://anaconda.org/conda-forge/libstdcxx-ng/files

@conda-forge/core any plans to update `libstdcxx-ng`

luzpaz, 2021-06-04:
@looooo maybe asking on https://gitter.im/conda-forge/conda-forge.github.io ?

luzpaz, 2021-07-10:
Maintained in this feedstock: https://github.com/conda-forge/ctng-compilers-feedstock  
https://anaconda.org/conda-forge/libstdcxx-ng

adrianinsaval, 2021-08-11:
This is the same issue as #73 so maybe it's ok to close it too however I would like to point out that the issue is still present in the stable release appimage, is there a chance of rebuilding the stable appimage?

luzpaz, 2021-08-12:
@adrianinsaval thanks for the reminder!
@looooo can we spin up a new stable appimage ?

looooo, 2021-08-12:
@luzpaz is there a 0.19.3 planned?

luzpaz, 2021-08-12:
Not anytime soon, AFAICT. I think we should regenerate the 'stable' appimage regardless

adrianinsaval, 2021-10-16:
there's probably going to be a 0.19.3 soon, I guess that one will have the fix

adrianinsaval, 2022-03-14:
well crap, this issue was reported again in debian testing...
https://forum.freecadweb.org/viewtopic.php?f=4&t=67033
new version of libstdc++ it seems
```
$ ls /usr/lib/x86_64-linux-gnu | grep libstdc++
libstdc++.so.6
libstdc++.so.6.0.30
```
Appimage has libstdc++.so.29
Do we really need to ship that library in the appimage?

rouckas, 2022-03-30:
I  can confirm that this issue appears in Ubuntu 22.4 (devel. version) and Freecad 0.19.3 and weekly-builds-28193. Running FreeCad as 
`LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./FreeCAD_weekly-builds-28193-Linux-Conda_glibc2.12-x86_64.AppImage`
can be used as a workaround

adrianinsaval, 2022-05-09:
possible solution mentioned by an user in the closed duplicate of this: https://github.com/FreeCAD/FreeCAD-Bundle/issues/73#issuecomment-1116408624

carsten-bruckhaus, 2022-05-26:
On two of my systems, FreeCAD does not launch anymore, either. Both have an AMD iGPU, both run Manjaro KDE. As @leoheck reported in #102, the `LD_PRELOAD=/usr/lib/libstdc++.so.6 ./FreeCAD_0.19.3-Linux-Conda_glibc2.12-x86_64_3ab8658a607146e30c97e40c83cf56e6.AppImage` workaround does not work for me, either (same error with or without LD_PRELOAD). The alternative and much more complicated workaround @leoheck  mentioned (extracting the Appimage, removing libstdc++.so.6, and repacking again) works for me, too.

```
LD_PRELOAD=/usr/lib/libstdc++.so.6 ./FreeCAD_0.19.3-Linux-Conda_glibc2.12-x86_64_3ab8658a607146e30c97e40c83cf56e6.AppImage
FreeCAD 0.19, Libs: 0.19R24366 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

libGL error: MESA-LOADER: failed to open radeonsi: /tmp/.mount_FreeCAHp5Kz3/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/radeonsi_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open radeonsi: /tmp/.mount_FreeCAHp5Kz3/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/radeonsi_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_FreeCAHp5Kz3/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
/tmp/.mount_FreeCAHp5Kz3/AppRun: line 41:  5070 Aborted                 (core dumped) ${MAIN} "$@"
```

eugenialoli, 2024-08-01:
Someone please fix this bug, it makes the appimage not usable on either Ubuntu or Mint. It just doesn't load, it fails to find the iris_drv.so lib.

adrianinsaval, 2024-10-27:
now that the newer appimages don't set LD_LIBRARY_PATH env variable anymore hopefully this issue will not pop up again, is there anyone who is still experiencing problems like this with any old or current appimage?