FreeCAD/FreeCAD-Bundle #68: Error-like messages while startup

mdziczkowski, 2021-05-16:
While starting the FreeCAD, I recieve strange error-like output (as included in the file).

**NOTE:** *Patches are trunchated for security reasons*
[freecad_errors.txt](https://github.com/FreeCAD/FreeCAD-Bundle/files/6489626/freecad_errors.txt)

luzpaz, 2021-05-16:
No appropriate place to post. Please use the forum. Off the bat I can see though that you're running an obsolete addon (flamingo) which has been superceded by Dodo workbench (see https://wiki.freecadweb.org/Flamingo_Workbench).

You probably need to update matplotlib separately. 
`ImportError: Matplotlib requires pyparsing>=2.2.1; you have 2.1.5`

Please ask on the forum and x-post back to this ticket.

mdziczkowski, 2021-05-17:
@luzpaz 
> Not appropriate place to post. Please use the forum.

Usually, this is the palace to post issues with the program and not forum

> You probably need to update matplotlib separately.
ImportError: Matplotlib requires pyparsing>=2.2.1; you have 2.1.5

I had updated pyparsing to the newest avaliavle version (2.4.6-1), but even that it show the error about the old version (2.1.5) of it

luzpaz, 2021-05-17:
Just saying, the forum is the first place to post as many issues get resolved there. The community vets the issue and deems it appropriate to open a ticket for. Or else what happens is folks circumvent the whole vetting process and it puts more pressure on the devs/maintainers/triagers to deal with potentially known or solvable problems due to downstream packaging issues (unrelated to FreeCAD).

luzpaz, 2021-05-17:
Also please always post your full: "Help -> About FreeCAD -> `Copy to clipboard`" info (whether you're in the forum or opening a ticket like this one)

luzpaz, 2022-04-03:
@mdziczkowski is this still happening?

luzpaz, 2022-07-11:
@mdziczkowski ping

mdziczkowski, 2022-07-12:
Sorry for the delay, but got independant from me internet access issues ( the internet provider seem to don't give me the full brandwitch and some times the speed even drop even bellow 1 MB/s, with causes that simple pages open quite long). 

I didn't had possibility because of above to check up the newest version :-(