FreeCAD/FreeCAD-Bundle #69: Crash on start WIN 10

nightgryphon, 2021-05-17:
FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64.7z
crash on win 10 laptop. The blank man window appears than immediately two crash windows appears
```
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	freecad.exe
  Application Version:	0.20.0.0
  Application Timestamp:	60a10bb8
  Fault Module Name:	Qt5WebEngineCore_conda.dll
  Fault Module Version:	5.12.9.0
  Fault Module Timestamp:	602ec71d
  Exception Code:	80000003
  Exception Offset:	0000000001a52763
  OS Version:	6.3.9600.2.0.0.768.100
  Locale ID:	1049
  Additional Information 1:	7bfe
  Additional Information 2:	7bfebc5eed5a5bf96e703642661f6019
  Additional Information 3:	a053
  Additional Information 4:	a053ff4c3b8c3d6529e93dfab8af4501
```

```
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	QtWebEngineProcess.exe
  Application Version:	5.12.9.0
  Application Timestamp:	602ec733
  Fault Module Name:	Qt5WebEngineCore_conda.dll
  Fault Module Version:	5.12.9.0
  Fault Module Timestamp:	602ec71d
  Exception Code:	80000003
  Exception Offset:	0000000002f3a100
  OS Version:	6.3.9600.2.0.0.768.100
  Locale ID:	1049
  Additional Information 1:	9ee0
  Additional Information 2:	9ee032e7175cb96689d4044ea90a99e1
  Additional Information 3:	e6ae
  Additional Information 4:	e6aebc3355880e721355e1b89ece5a3b
```

The crash log show multiple "Access Violation" errors.

```
Log: Id: 6504: SymInit: Symbol-SearchPath: '.;C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin;C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin;C:\WINDOWS;C:\WINDOWS\system32;SRV*C:\websymbols*http://msdl.microsoft.com/download/symbols;', symOptions: 530, UserName: 'First'
Log: Id: 6504: OS-Version: 6.2.9200 () 0x300-0x1
Log: Id: 6504: C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\freecad.exe:freecad.exe (00007FF789770000), size: 163840 (result: 0), SymType: '-nosymbols-', PDB: 'C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\freecad.exe', fileVersion: 0.20.0.0
Log: Id: 6504: C:\WINDOWS\SYSTEM32\ntdll.dll:ntdll.dll (00007FF91CD50000), size: 1757184 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\ntdll.dll', fileVersion: 6.3.9600.19678
Log: Id: 6504: C:\WINDOWS\system32\KERNEL32.DLL:KERNEL32.DLL (00007FF91C570000), size: 1306624 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\system32\KERNEL32.DLL', fileVersion: 6.3.9600.19719
Log: Id: 6504: C:\WINDOWS\system32\KERNELBASE.dll:KERNELBASE.dll (00007FF91A1B0000), size: 1134592 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\system32\KERNELBASE.dll', fileVersion: 6.3.9600.19724
Log: Id: 6504: C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\FreeCADGui.dll:FreeCADGui.dll (00007FF8FB5E0000), size: 19902464 (result: 0), SymType: '-exported-', PDB: 'C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\FreeCADGui.dll'
Log: Id: 6504: C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\FreeCADApp.dll:FreeCADApp.dll (00007FF8FD6F0000), size: 7098368 (result: 0), SymType: '-exported-', PDB: 'C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\FreeCADApp.dll'
Log: Init: Showing main window
Log: Main window restored
Log: Id: 6504: C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\FreeCADBase.dll:FreeCADBase.dll (00007FF9074D0000), size: 2256896 (result: 0), SymType: '-exported-', PDB: 'C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\FreeCADBase.dll'
Err: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 14 was sent to QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget
Object tree:
	QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget is child of
	WebGui::WebView is child of
	WebGui::BrowserView is child of
	QMdiSubWindow is child of
	QWidget is child of
	QMdiArea is child of
	Gui::MainWindowErr: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 17 was sent to QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget
Object tree:
	QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget is child of
	WebGui::WebView is child of
	WebGui::BrowserView is child of
	QMdiSubWindow is child of
	QWidget is child of
	QMdiArea is child of
	Gui::MainWindowLog: Show main window
Log: Toolbars restored
Log: 3Dconnexion device not attached.
Log: Init: Entering event loop
Err: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 1 was sent to QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget
Object tree:
	QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget is child of
	WebGui::WebView is child of
	WebGui::BrowserView is child of
	QMdiSubWindow is child of
	QWidget is child of
	QMdiArea is child of
	Gui::MainWindowLog: Id: 6504: C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\MSVCP140.dll:MSVCP140.dll (00007FF9055E0000), size: 598016 (result: 0), SymType: '-exported-', PDB: 'C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\MSVCP140.dll', fileVersion: 14.28.29325.2
Log: Id: 6504: C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\VCRUNTIME140.dll:VCRUNTIME140.dll (00007FF908D60000), size: 102400 (result: 0), SymType: '-exported-', PDB: 'C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\VCRUNTIME140.dll', fileVersion: 14.28.29325.2
Log: Id: 6504: C:\WINDOWS\SYSTEM32\api-ms-win-crt-runtime-l1-1-0.dll:api-ms-win-crt-runtime-l1-1-0.dll (00007FF9031C0000), size: 16384 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\api-ms-win-crt-runtime-l1-1-0.dll', fileVersion: 10.0.14393.2247
Log: Id: 6504: C:\WINDOWS\SYSTEM32\api-ms-win-crt-environment-l1-1-0.dll:api-ms-win-crt-environment-l1-1-0.dll (00007FF9129D0000), size: 12288 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\api-ms-win-crt-environment-l1-1-0.dll', fileVersion: 10.0.14393.2247
Log: Id: 6504: C:\WINDOWS\SYSTEM32\api-ms-win-crt-heap-l1-1-0.dll:api-ms-win-crt-heap-l1-1-0.dll (00007FF9031D0000), size: 12288 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\api-ms-win-crt-heap-l1-1-0.dll', fileVersion: 10.0.14393.2247
Log: Id: 6504: C:\WINDOWS\SYSTEM32\api-ms-win-crt-math-l1-1-0.dll:api-ms-win-crt-math-l1-1-0.dll (00007FF912240000), size: 20480 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\api-ms-win-crt-math-l1-1-0.dll', fileVersion: 10.0.14393.2247
Log: Id: 6504: C:\WINDOWS\SYSTEM32\api-ms-win-crt-stdio-l1-1-0.dll:api-ms-win-crt-stdio-l1-1-0.dll (00007FF9031E0000), size: 16384 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\api-ms-win-crt-stdio-l1-1-0.dll', fileVersion: 10.0.14393.2247
Err: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 14 was sent to QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget
Object tree:
	QtWebEngineCore::RenderWidgetHostViewQtDelegateWidget is child of
	WebGui::WebView is child of
	WebGui::BrowserView is child of
	QMdiSubWindow is child of
	QWidget is child of
	QMdiArea is child of
	Gui::MainWindowLog: Id: 6504: C:\WINDOWS\SYSTEM32\api-ms-win-crt-locale-l1-1-0.dll:api-ms-win-crt-locale-l1-1-0.dll (00007FF910270000), size: 12288 (result: 0), SymType: '-exported-', PDB: 'C:\WINDOWS\SYSTEM32\api-ms-win-crt-locale-l1-1-0.dll', fileVersion: 10.0.14393.2247
Log: Init: Processing command line files
Err: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Err: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: Id: 6504: C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\Qt5Widgets_conda.dll:Qt5Widgets_conda.dll (00007FF8FD190000), size: 5591040 (result: 0), SymType: '-exported-', PDB: 'C:\DIST\FreeCAD_weekly-builds-24894-Win-Conda_vc14.x-x86_64\bin\Qt5Widgets_conda.dll', fileVersion: 5.12.9.0
Err: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 77 was sent to Gui::MainWindow
Object tree:
	Gui::MainWindowErr: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 77 was sent to Gui::MainWindow
Object tree:
	Gui::MainWindowErr: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 77 was sent to Gui::MainWindow
Object tree:
	Gui::MainWindowErr: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 77 was sent to Gui::MainWindow
Object tree:
	Gui::MainWindowErr: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 77 was sent to Gui::MainWindow
...
```

The stable version also does not work but crashes even earlier during splash screen.

luzpaz, 2021-05-18:
please first open a thread on the forum. we don't have enough people looking at this repo and it doesn't serve as a 1st point of contact for bugs regarding installation or crashes. That''s what the forum is for.

Mannshoch, 2022-06-01:
For the record.
I work  for a company that faced such problems on several software. The Problem was the Group policy "RemoteFX" was not activated.

https://social.technet.microsoft.com/Forums/en-US/e139ec92-90d0-4639-a3b5-5a5a4a5b6a79/hardware-support-gpu-on-rdp-connection-to-windows-10-professional?forum=winRDc#9daf555b-5da7-4a1f-be8b-d31b0d6226de

Mannshoch, 2022-06-01:
I for myself use FreeCad 0.19.4  often over RDP and It works great. So if RemotFX is the difference and someon could verify it then https://github.com/FreeCAD/FreeCAD/issues/6253 yes.

adrianinsaval, 2024-07-05:
sounds like this is a problem with windows an RDP rather than a freecad problem, is there any reasonable action we could take here? Maybe someone could write an article on the wiki