FreeCAD/FreeCAD-Bundle #70: shiboken2 not bundeled in Appimage  ?

s-leger, 2021-05-23:
Running daily
FreeCAD_weekly-builds-24894-Linux-Conda_glibc2.12-x86_64.appimage
returns

PySide2/__init__.py: Unable to import shiboken2 from /tmp/.mount_FreeCAYax3Ml/usr/Mod/Web, /tmp/.mount_FreeCAYax3Ml/usr/Mod/Tux, /tmp/.mount_FreeCAYax3Ml/usr/Mod/Test, /tmp/.mount_FreeCAYax3Ml/usr/Mod/TechDraw, /tmp/.mount_FreeCAYax3Ml/usr/Mod/Surface, /tmp/.mount_FreeCAYax3Ml/usr/Mod/Start, /tmp/.mount_FreeCAYax3Ml/usr/Mod/Spreadsheet, /tmp/.mount_FreeCAYax3Ml/usr/Mod/Sketcher, ...

luzpaz, 2021-05-30:
can you try with the latest appimage?

s-leger, 2021-05-30:
Still the same here
PySide2/__init__.py: Unable to import shiboken2 from /tmp/.mount_FreeCAUH4zqc/usr/Mod/Web (...)

s-leger, 2021-05-30:
Looks like it is a conflict with /home/%user%/.local/lib/python3.9/site-packages/   python modules. 
When removed, will load yours as expected.

closed 2021-05-30 by s-leger