FreeCAD/FreeCAD-Bundle #71: where to get the libpack matching FreeCAD_weekly-builds-24910

arthurx70, 2021-05-26:
where to get the libpack matching FreeCAD_weekly-builds-24910,the version of python in libpack is 3.8.6+,and the version of python in FreeCAD_weekly-builds is 3.8.8.

luzpaz, 2021-05-30:
i dont think we create a libpack for the dev version. The windows build in this repo is created from conda (different than how the stable builds are created). Am i right, @looooo ?

looooo, 2021-11-29:
> The windows build in this repo is created from conda (different than how the stable builds are created). Am i right, @looooo ?

stable builds are also from conda. It's the same procedure. But there are alternatives to conda. 

I close this issue. Feel free to reopen if this is still an issue.

closed 2021-11-29 by looooo