FreeCAD/FreeCAD-Bundle #73: This week build does not start up on linux

X-Ryl669, 2021-06-14:


X-Ryl669, 2021-06-14:
I've extracted the appimage and run under GDB, got this:
```
(gdb) r
Starting program: ./squashfs-root/usr/bin/freecad
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
warning: the debug information found in "./squashfs-root/usr/bin/../lib/././libnsl-2.31.so" does not match "I/squashfs-root/usr/bin/../lib/././libnsl.so.1" (CRC mismatch).

FreeCAD 0.20, Libs: 0.20R25065 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Loading Assembly4 WorkBench
[New Thread 0x7ffff0976640 (LWP 8787)]
libGL error: MESA-LOADER: failed to open iris: ./squashfs-root/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: ./squashfs-root/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: ./squashfs-root/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: swrast
Fontconfig error: Cannot load default config file
[Detaching after fork from child process 8788]
[Detaching after fork from child process 8789]
During initialization the error "'utf-8' codec can't encode characters in position 13-14: surrogates not allowed" occurred in freecad.appimage
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 201, in <module>
  File "<string>", line 163, in InitApplications
<class 'SystemError'>: <built-in function PrintError> returned a result with an error set
Error in FreeCADGuiInit.py: <built-in function PrintError> returned a result with an error set
[Thread 0x7ffff0976640 (LWP 8787) exited]
[Inferior 1 (process 8781) exited with code 01]
```

X-Ryl669, 2021-06-14:
The issue comes from the updated libstdc++ on my system and mismatch from what's inside the AppImage.
It seems that when loading Mesa's OpenGL's drivers (which are loaded dynamically based on my system's overlay set up in `/usr/lib/dri/`), the libstdc++ from the AppImage (which is version 6.0.28) is loaded instead of my system's 6.0.29 version. So they fail loading, preventing the GUI to start.

I don't know if it's an issue with AppImage itself or with FreeCAD building of the image that don't use system's libstdc++. 

I've replaced the symbolic link in the AppImage's /usr/lib folder to point to my system's library instead and it worked!

luzpaz, 2021-06-19:
Can confirm the crash on:
```
Operating System: Manjaro Linux
KDE Plasma Version: 5.21.5
KDE Frameworks Version: 5.82.0
Qt Version: 5.15.2
Kernel Version: 5.12.9-1-MANJARO
OS Type: 64-bit
Graphics Platform: X11
Processors: 4 × Intel® Core™ i7-6560U CPU @ 2.20GHz
Memory: 15.5 GiB of RAM
Graphics Processor: Mesa Intel® Iris® Graphics 540
```

Here's the error
```
$ ./FreeCAD_weekly-builds-25065-Linux-Conda_glibc2.12-x86_64_a8491e74a0f1abfe09fd788d0e9f4efd.AppImage 
FreeCAD 0.20, Libs: 0.20R25065 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Assembly4 workbench loaded 
Sheet Metal workbench loaded
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_FreeCAL6BPFg/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_FreeCAL6BPFg/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: swrast
/tmp/.mount_FreeCAL6BPFg/AppRun: line 41:  6926 Aborted                 (core dumped) ${MAIN} "$@"
```

luzpaz, 2021-06-19:
Workaround posted: https://forum.freecadweb.org/viewtopic.php?f=4&t=58351&p=511200#p510504

probonopd, 2021-07-18:
What happens if you don't bundle any `libstdc++` inside the AppImage?

luzpaz, 2021-07-18:
> What happens if you don't bundle any `libstdc++` inside the AppImage?

@looooo have you ever tried this? 

Hi @probonopd! Thanks for weighing in :smile:

looooo, 2021-07-18:
@probonopd @luzpaz I guess then we have issues with older distributions.

looooo, 2021-07-18:
There is for sure a reason why it is included in conda envs. Actually it was added some time ago to make the envs more reliable. But actually it has the disadvantage of having to be updated to work with newer distros. 

I hope someone from conda-forge will work on updating the GCC compiler stuff.

probonopd, 2021-07-18:
Possibly https://github.com/darealshinji/AppImageKit-checkrt/ can be used?

luzpaz, 2021-07-23:
@looooo can we rebuild the appimage now since https://github.com/conda-forge/ctng-compilers-feedstock/issues/55 is fixed ?

looooo, 2021-07-23:
Great! Yes, let's try to rebuild and see if new packages are used.

looooo, 2021-07-23:
looks good: https://github.com/FreeCAD/FreeCAD-Bundle/runs/3143181732?check_suite_focus=true#step:12:160

The build-number is still the same, but the appimage should have the new packages included. Maybe someone can try.

luzpaz, 2021-07-23:
@looooo d/ling now. Will report back soon

luzpaz, 2021-07-23:
@looooo works for me!  
```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.25288 (Git) AppImage
Build type: Release
Branch: master
Hash: bbb557aab6129efd9486d6df19f06c2976fdb7e5
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)
```

luzpaz, 2021-07-23:
Closing ticket. Please re-open if issue persists

closed 2021-07-23 by luzpaz

adrianinsaval, 2021-08-07:
@looooo shouldn't the stable release appimage be rebuilt too? I just tested and it has the same problem in archlinux

ouinouin, 2022-05-03:
confirm rebuilding the weekly image removing the libstdc++ works in ubuntu 22.04.

probonopd, 2022-05-03:
So, https://github.com/darealshinji/AppImageKit-checkrt/ was not looked at?

AndKe, 2022-05-04:
I see the same issue now on Ubuntu 22.04   using:  ./FreeCAD_weekly-builds-28774-Linux-Conda_glibc2.12-x86_64.AppImage 

Is it a regression?
```
$ ./FreeCAD_weekly-builds-28774-Linux-Conda_glibc2.12-x86_64.AppImage 
FreeCAD 0.20, Libs: 0.20R28774 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

libGL error: MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
/tmp/.mount_FreeCA57PZ1n/AppRun: line 42: 23871 Aborted                 (core dumped) ${MAIN} "$@"

```

AndKe, 2022-05-04:
@ouinouin - can you please share the rebuilt weekly?

X-Ryl669, 2022-05-04:
That's a typical case of catch-22 problem here. The main issue is that the app image is using the system's OpenGL library. Since those libraries depends on the system's libstdc++ library, you can only have 2 alternatives here:
1. Include your own libstdc++ library in the app image, but it'll fail on system where the libstdc++ is not the exact same version as the app image (like in your case for Ubuntu above)
2. Depend on the system's libstdc++ for FreeCAD (so the app image does not embed the libstdc++ and rely on the system's version). This does not work since libstdc++'s init code checks the version with the binary requirement that was compiled and bails out if it does not match.

Another possible solution would be to build FreeCAD's AppImage with libstdc++ as a static library. I'm not sure if it's possible and how to do that, but as far as I understand, it's the only viable approach.

looooo, 2022-05-10:
I think relying on the conda-forge approach is the way to continue. I am not an expert on this topic, but I think there are two issues:
1. Old systems that have an incompatible version. This is solved by including the libstdc++ version
2. New systems that have an incompatible Version of this library: this is solved by updating the libstdc++ version of conda-forge.

But I am not sure what's the process of updating this library. So I guess it's best to report this issue to conda-forge.

adrianinsaval, 2022-05-14:
Why not the checkrt thing? The problem with just waiting for Conda update is that this will likely happen again in the future, having a way of deciding to use system or appimage version would solve this permanently

looooo, 2022-05-15:
@adrianinsaval Yes, thats true. If anyone wants to get into this issue, I am happy for a PR. Otherwise, I will try to apply this once I have some free time.