FreeCAD/FreeCAD-Bundle #75: Change the description of this repo

luzpaz, 2021-09-01:
[image]

It is more than just the appimage...

luzpaz, 2021-09-06:
@looooo bump

sgrogan, 2021-09-06:
This is me. We have the (conda-based) AppImage of course, but also the conda-based .dmg and .7z. In addition I use the releases to publish Libpack based assets, although there is nothing in the repo that supports this. As you know we also do some experimental/convenience packaging here like glider and asm3.

We have a better name now. I'll try to come up with something better for the description. As you all know I'm no wordsmith, so we would, of course, entertain pull requests.

luzpaz, 2021-09-07:
you can't submit PRs to change the repository descriptions :sweat_smile:

luzpaz, 2021-09-08:
@sgrogan how about: `Weekly generated FreeCAD development builds for testing the latest features/bug fixes` ?

luzpaz, 2021-09-11:
@sgrogan ping

luzpaz, 2022-09-10:
@looooo can you give me permissions to be able to change stuff like this?

looooo, 2022-09-10:
sry, I don't have the permission to do this. Maybe pinging @yorikvanhavre

luzpaz, 2023-01-14:
@yorikvanhavre we need to update the repo description.

luzpaz, 2023-02-27:
@yorikvanhavre ping. If you need help, I'm happy to update. I need privs though.

yorikvanhavre, 2023-02-27:
Sorry, forgot this... What about this?
[image]

closed 2023-03-03 by luzpaz