FreeCAD/FreeCAD-Bundle #78: weird behavior of 'Loft' interacting with 'Linear Patern' and such since 0.20

discip, 2021-10-05:
Good evening,
first of all thank you for this great software!

Despite the greatly welcomed improvements, I have to report an issue concerning the **`loft`** feature interacting with **`linear pattern`**, **`polar pattern`** & **`multi transform`**.

As you can observe in the attached files, **`linear pattern`** only works for an **`additive  loft`** if it is either
1. directly on top of the **`pad`** (0 distance) or
2. if it is intersecting the **`pad`** it has to intersect the **`linear patterned`** part as well.

However this does not apply to **`extruded`** objects!

Happy to answer questions, if I was not able to make myself clear enough. 😊

thanks in advance
kind regards

[examples.zip](https://github.com/FreeCAD/FreeCAD-Bundle/files/7288807/examples.zip)

3mandd, 2021-10-07:
Hey, 
I think the way it goes with FreeCAD,
you have to discuss here: https://forum.freecadweb.org/viewforum.php?f=8
and once confirmed, make bug report

discip, 2021-10-07:
@3mandd
Thanks, will try.

luzpaz, 2021-10-07:
Indeed:
[image]
Please see https://wiki.freecad.org/Tracker

Closing ticket as offtopic

closed 2021-10-07 by luzpaz