FreeCAD/FreeCAD-Bundle #82: create script to run the scripts locally

looooo, 2021-11-02:


looooo, 2021-11-29:
@stepheneb 
thanks for having a look at this.

>1. Am not intending to create a release so the error looking for the missing env var DEPLOY_RELEASE is expected.

should be fixed with: https://github.com/FreeCAD/FreeCAD-Bundle/commit/40207a0a2c0a233af121c63c76fd6b9d2d398335

>2. Not sure if the PackagesNotFoundError error is important or not?
not a big issue, but we can remove them from the script:
https://github.com/FreeCAD/FreeCAD-Bundle/commit/e84a2b58b3f4f1c141decc6d472a0268c73db3eb

>3. The error at the end trying to remove libb/python3.8/site-packages/mpmath/ctx_mp_python.py from the app seems strange. Not sure why it's looking in the python3.8 directory? Those files are in the python3.9 directory:

https://github.com/FreeCAD/FreeCAD-Bundle/commit/b2f09dba9ef2f7b7d5a93b7b723b0de02f300c56

Maybe you can try again.

stepheneb, 2021-11-29:
@looooo 

Tested this morning. It's much smoother. 

Noticed this at the end of the process. Not sure if it's an issue.

```
sed: 1: "APP/FreeCAD.app/Content ...": invalid command code A
cp: APP/FreeCAD.app/Contents/Resources/bin_tmp/assistant: No such file or directory

```

Running the built-in tests generates a segmentation fault on this test:

    test_get_svg_from_arch_space_with_zero_vector (drafttests.test_svg.DraftSVG)

Not certain if that's a real issue with the code or related to the packaging.

Lots of details below including the segmentation fault report at the end.

Starting in my top-level FreeCAD-Bundle directory:

Update repo:

```
[FreeCAD-Bundle(master)]$ git status
On branch master
Your branch is up to date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	conda/osx/.dmg-SHA256.txt
	conda/osx/.dmg.dmg
	conda/osx/APP/FreeCAD.app/Contents/Info.plist
	conda/osx/APP/FreeCAD.app/Contents/Resources/Ext/
	conda/osx/APP/FreeCAD.app/Contents/Resources/LICENSE
	conda/osx/APP/FreeCAD.app/Contents/Resources/Mod/
	conda/osx/APP/FreeCAD.app/Contents/Resources/bin/
	conda/osx/APP/FreeCAD.app/Contents/Resources/cmake/
	conda/osx/APP/FreeCAD.app/Contents/Resources/conda-meta/
	conda/osx/APP/FreeCAD.app/Contents/Resources/doc/
	conda/osx/APP/FreeCAD.app/Contents/Resources/etc/
	conda/osx/APP/FreeCAD.app/Contents/Resources/fonts/
	conda/osx/APP/FreeCAD.app/Contents/Resources/lib/
	conda/osx/APP/FreeCAD.app/Contents/Resources/libexec/
	conda/osx/APP/FreeCAD.app/Contents/Resources/man/
	conda/osx/APP/FreeCAD.app/Contents/Resources/mkspecs/
	conda/osx/APP/FreeCAD.app/Contents/Resources/phrasebooks/
	conda/osx/APP/FreeCAD.app/Contents/Resources/plugins/
	conda/osx/APP/FreeCAD.app/Contents/Resources/qml/
	conda/osx/APP/FreeCAD.app/Contents/Resources/resources/
	conda/osx/APP/FreeCAD.app/Contents/Resources/sbin/
	conda/osx/APP/FreeCAD.app/Contents/Resources/share/
	conda/osx/APP/FreeCAD.app/Contents/Resources/ssl/
	conda/osx/APP/FreeCAD.app/Contents/Resources/translations/
	conda/osx/APP/FreeCAD.app/Contents/Resources/var/
	conda/osx/APP/FreeCAD.app/Contents/packages.txt

nothing added to commit but untracked files present (use "git add" to track)
[FreeCAD-Bundle(master)]$ git pull
remote: Enumerating objects: 62, done.
remote: Counting objects: 100% (62/62), done.
remote: Compressing objects: 100% (52/52), done.
remote: Total 62 (delta 30), reused 16 (delta 8), pack-reused 0
Unpacking objects: 100% (62/62), 13.67 KiB | 237.00 KiB/s, done.
From github.com:FreeCAD/FreeCAD-Bundle
   444a06d..71f05b6  master         -> origin/master
   0e78972..1dd2ff6  0.19.3         -> origin/0.19.3
 * [new branch]      release_0.19.3 -> origin/release_0.19.3
Updating 444a06d..71f05b6
[... 2168 lines cut ...]
    0x7fff2bdd3000 -     0x7fff2bdf7fff  libJPEG.dylib (2130.7.3) <EE8BD43F-037C-3AA8-AA5F-685BC1E858BE> /System/Library/Frameworks/ImageIO.framework/Versions/A/Resources/libJPEG.dylib
    0x7fff2bdf8000 -     0x7fff2be27fff  com.apple.ATSUI (1.0 - 1) <92428504-C766-3ACE-BC36-89F1A5CAB61C> /System/Library/Frameworks/ApplicationServices.framework/Versions/A/Frameworks/ATSUI.framework/Versions/A/ATSUI
    0x7fff2be28000 -     0x7fff2be2cfff  libGIF.dylib (2130.7.3) <EE211D7D-4994-3BFB-BC49-C63069B800E8> /System/Library/Frameworks/ImageIO.framework/Versions/A/Resources/libGIF.dylib
    0x7fff2be2d000 -     0x7fff2be36fff  com.apple.CMCaptureCore (1.0 - 82.6) <C1960208-BCA3-3D47-9DBE-CF1A81C2053B> /System/Library/PrivateFrameworks/CMCaptureCore.framework/Versions/A/CMCaptureCore
    0x7fff2be37000 -     0x7fff2be7efff  com.apple.print.framework.PrintCore (16.1 - 531.1) <8139D1A2-7037-3B4F-AB8F-2BF38D64D798> /System/Library/Frameworks/ApplicationServices.framework/Versions/A/Frameworks/PrintCore.framework/Versions/A/PrintCore
    0x7fff2be7f000 -     0x7fff2bf4cfff  com.apple.TextureIO (3.10.9 - 3.10.9) <0E29AC8D-246E-3F7F-A8A0-10E09DD64603> /System/Library/PrivateFrameworks/TextureIO.framework/Versions/A/TextureIO
    0x7fff2bf4d000 -     0x7fff2bf55fff  com.apple.InternationalSupport (1.0 - 61.1) <8E0C26AC-7462-34B5-B608-0DB611CCB687> /System/Library/PrivateFrameworks/InternationalSupport.framework/Versions/A/InternationalSupport
    0x7fff2bf56000 -     0x7fff2bfd0fff  com.apple.datadetectorscore (8.0 - 674) <3E182298-355E-337F-A398-B266DEC5E6F3> /System/Library/PrivateFrameworks/DataDetectorsCore.framework/Versions/A/DataDetectorsCore
    0x7fff2bfd1000 -     0x7fff2c02efff  com.apple.UserActivity (439 - 439) <E011EB02-BB18-3154-8010-AD4C6D21792B> /System/Library/PrivateFrameworks/UserActivity.framework/Versions/A/UserActivity
    0x7fff2c02f000 -     0x7fff2c7cbfff  com.apple.MediaToolbox (1.0 - 2780.10.4.1) <A90F330D-ACB5-30FC-9F25-CA12F6D8DE3C> /System/Library/Frameworks/MediaToolbox.framework/Versions/A/MediaToolbox
    0x7fff2cc73000 -     0x7fff2cc9bfff  com.apple.locationsupport (2420.19.23 - 2420.19.23) <0CED90D1-F2D8-34B3-8501-236B0EAA8C7F> /System/Library/PrivateFrameworks/LocationSupport.framework/Versions/A/LocationSupport
    0x7fff2cc9c000 -     0x7fff2cccdfff  libSessionUtility.dylib (76.69) <E6D8E0BB-3C76-343B-B8EA-03C411971A6C> /System/Library/PrivateFrameworks/AudioSession.framework/libSessionUtility.dylib
    0x7fff2ccce000 -     0x7fff2ce02fff  com.apple.audio.toolbox.AudioToolbox (1.14 - 1.14) <7C7D69A5-6686-3040-88F4-87B892B96C0D> /System/Library/Frameworks/AudioToolbox.framework/Versions/A/AudioToolbox
[... cut ...]
    0x7fff2d124000 -     0x7fff2d183fff  libcups.2.dylib (494.3) <353033B4-E6E6-375A-8BC7-0EE2F67FC8C3> /usr/lib/libcups.2.dylib
    0x7fff2d184000 -     0x7fff2d193fff  com.apple.LangAnalysis (1.7.0 - 254) <EE21E044-0B25-3D3E-A892-A276452D8EC8> /System/Library/Frameworks/ApplicationServices.framework/Versions/A/Frameworks/LangAnalysis.framework/Versions/A/LangAnalysis
    0x7fff2d194000 -     0x7fff2d19efff  com.apple.NetAuth (6.2 - 6.2) <08F7BAF1-D834-3FAF-9FCC-7FBC32BAF037> /System/Library/PrivateFrameworks/NetAuth.framework/Versions/A/NetAuth
    0x7

looooo, 2021-11-29:
> Running the built-in tests generates a segmentation fault on this test:

Is this also failing with the current weekly-build?

stepheneb, 2021-11-29:
Yes,

```
$ ~/Downloads/FreeCAD.app/Contents/MacOS/FreeCAD -t 0
...
test_export_svg (drafttests.test_svg.DraftSVG)
Create some figures and export them to an SVG file. ... 
------------------------------------------------------------------------------
  Temporary document 'DraftSVG'
  Test 'importSVG.export'
  file=/Users/stephen/Downloads/FreeCAD.app/Contents/Resources/share/Mod/Draft/drafttest/out_test.svg
  exists=False
  Arguments to placeholder function
  p1=/Users/stephen/Downloads/FreeCAD.app/Contents/Resources/share/Mod/Draft/drafttest/out_test.svg; p2=None
  p3=None; p4=None
  p5=None
  #-----------------------------------------------------#
  #    This test is not implemented currently
  #-----------------------------------------------------#
  Automatic PASS
ok
test_get_svg_from_arch_space_with_zero_vector (drafttests.test_svg.DraftSVG)
Try to get a svg string from an Arch Space with a zero-vector as direction. ... 
------------------------------------------------------------------------------
  Temporary document 'DraftSVG'
/Users/stephen/Downloads/FreeCAD.app/Contents/MacOS/FreeCAD: line 19: 34921 Segmentation fault: 11  "${PREFIX}/bin/freecad" $@
```

stepheneb, 2021-11-29:
Also noticed the app icons are different.
[image]

looooo, 2021-11-29:
@stepheneb ok, so we should report this issue to FreeCAD. 
Not sure about the icon issue.