FreeCAD/FreeCAD-Bundle #89: No new buillds?

luzpaz, 2022-01-26:
Especially for windows?

luzpaz, 2022-01-26:
@donovaly ping

donovaly, 2022-01-26:
I publish from time to time new builds here: https://github.com/donovaly/FreeCADInstProj/releases

I can also put them to FreeCAD's repository if you like.

I only provide new builds when there are substantial new features since I don't have an automated build process yet.

luzpaz, 2022-01-26:
@donovaly probably a good idea to offer them here as well...what do you think @looooo ?

donovaly, 2022-01-27:
I uploaded a recent build:
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2022-01-27:
Do you mind 7zipping it?

donovaly, 2022-01-27:
> Do you mind 7zipping it?

No, because the .exe is already a self-extracting ZIP + it takes care of the file assignments, provides an uninstaller etc.

Note that not everybody can unpack 7zip. For example at work I cannot install anything and have to rely on Windows' included tools and they cannot handle 7Z but ZIP.
(I started to provide the builds for my colleagues.)

donovaly, 2022-01-27:
Also not that my build is a fully functional FreeCAD, including translations, TechDraw templates and the like, while the zipped weekly builds purposely don't include e.g. some templates.

So maybe my builds don't fit in here perfectly.

noahwilliamsson, 2022-01-27:
I've understood these builds are currently done/uploaded manually but I'm curious what's holding automated, regular builds for each platform back?

looooo, 2022-01-27:
The conda Windows builds are currently failing.

luzpaz, 2022-01-27:
Link?  
Is there a bot that can notify us somehow ?

donovaly, 2022-01-27:
> Is there a bot that can notify us somehow ?

You mean when there are new versions available? Then just watching https://github.com/FreeCAD/FreeCAD-Bundle by setting a star should do the job. If not you must write a script, I assume GitHub has some API for this.

noahwilliamsson, 2022-01-27:
@donovaly I assume @luzpaz means when the https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds release is updated.  For that case, I doubt watching the repo will do the trick.  At least nothing shows up in [DevHub](https://devhubapp.com/) when the existing `weekly-builds` release is updated. (btw, I can really recommend the cross-platform DevHub app if you're interested in keeping an eye on GitHub activities, including PRs, issues, comments, commits etc).

looooo, 2022-01-28:
> Link? Is there a bot that can notify us somehow ?

https://forum.freecadweb.org/viewtopic.php?f=42&t=65259

luzpaz, 2022-02-11:
@looooo what about the macOS builds ? They've also stalled

brianredbeard, 2022-02-12:
and Linux

looooo, 2022-02-14:
WIP. Hopefully, new builds will be available today.

donovaly, 2022-02-14:
> which ones should be deleted?

When you have new LP builds then the existing LP builds

donovaly, 2022-02-15:
> @donovaly I am not responsible for the lp builds, only conda :)

Who is responsible for the LP builds? @sgrogan?

luzpaz, 2022-03-26:
I just manually triggered the [FreeCAD Bundeling action](FreeCAD bundeling) waiting to see the conda builds build and uploaded to https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2022-03-26:
macOS build result (no change)  

[image]

luzpaz, 2022-03-26:
@looooo what am i missing ?

looooo, 2022-03-27:
There is the need to build freecad first via the freecad-feedstock. Currently there are a lot of not resolved issues. So please be kind and accept that conda based builds are lacking some stuff. I currently can't resolve these issues, but I am planning to get back on top with qt6/occt 7.6. For now I guess we have to accept the current state. I will continue to trigger the feedstock but not work on the issues instead concentrate on qt6 once it arrives at conda-forge.

luzpaz, 2022-03-27:
I don't want to create more pressure. The problem is that we're on the verge of a release and people want to test, are being told they can and then they can't test latest changes. So, there are 2 issues here:  
1. you're overworked, more people need to be looking at this 
2. communication needs to happen in an optimized way. We need to have a prominent way for the errors in feedstock to be announced and then triaged

looooo, 2022-03-31:
@luzpaz the feedstock should be ok again, for all 3 platforms. Somehow the upload for osx is not working. I created an issue here:
https://github.com/svenstaro/upload-release-action/issues/68

looooo, 2022-04-01:
Now everything should run fully automatically. It's a bit more complex than it needs to be, but here is how it works:
1. six times a day a github action is triggered which graps the freecad-source, inserts the git information and creates a compressed file:: https://github.com/FreeCAD/FreeCAD-Bundle/blob/master/.github/workflows/source_creation.yml
2. on sunday at 6 utc a github action is triggered which increases the build-number of the freecad-feedstock: https://github.com/looooo/freecad-feedstock/blob/master/.github/workflows/increase_build_number.yml
3. on sunday at 20 utc the freecad-bundeling-process is triggered: https://github.com/FreeCAD/FreeCAD-Bundle/blob/master/.github/workflows/actions.yml

closed 2022-04-06 by luzpaz