FreeCAD/FreeCAD-Bundle #90: Supporting Qt6

luzpaz, 2022-01-27:
Thread on the forum [discussing](https://forum.freecadweb.org/viewtopic.php?f=10&t=65089) steps to support Qt6

luzpaz, 2022-01-27:
Heads up to all FC packagers:  
@sgrogan  
@kkremitzki  
@ipatch  
@adrianschroeter  
@hobbes1069  
@greyltc  
@hfiguiere  
@vejmarie  
@donovaly  
@apeltauer  
@waebbl

hobbes1069, 2022-01-27:
This would be great as Qt isn't updating pyside for Qt 5 to be compatible with Python 3.10/3.11.

ipatch, 2022-01-27:
@hobbes1069 

have you taken a look at this?

https://github.com/FreeCAD/homebrew-freecad/issues/267

and more specifically this,

https://github.com/FreeCAD/homebrew-freecad/issues/267#issuecomment-1004308048

so drilling into the bug report i filed with the upstream pyside project,

https://bugreports.qt.io/browse/PYSIDE-1763

it seems certain patches could be back ported for pyside 5.15.2 to work with python ≥ 3.10

however this approach seems it will only get more brittle as dependencies move forward ie. python.

i agree there should be some effort applied to begin supporting qt6 as opposed to applying patches.

in another note, i know python 3.10 is released now, but does freecad use any language feature added in python 3.10 that would make it more beneficial to use python 3.10 over python 3.9. or are we just updating for the sake of updating?

finally, i think this github issue should be place in the freecad src repo, ie. github.com/freecad/freecad (or are we placing here because the upstream repo does not have github issues enabled on the project and is using its own bug tracker?) either several cmakefiles will require updating to support qt6 at the least, and not mention all the header import statements that import/use qt libs that have probably been renamed since qt5. it will definitely require some effort.

hfiguiere, 2022-01-27:
The problem with Python 3.10 is that it will be on most Linux distro and in the next Flatpak runtime (August 2022). So it will definitely be a problem.

hobbes1069, 2022-01-28:
Exactly. Fedora 35 already has Python 3.10 and Fedora 36 (current rawhide) will have 3.11.

waebbl, 2022-01-29:
On Gentoo we have Python 3.10 and 3.11_alpha available. It's planned to make 3.10 the default Python interpreter in July. Qt6 is packaged, but still masked, and AFAICS there's not much pressure to change this.

It seems like the Qt folks are preparing a special PySide version with Python 3.10 compatibility: [5.15.2.1 to be release soon, special version with Python 3.10 support](https://wiki.qt.io/Qt_for_Python_Development_Notes#13._January_2022).

donovaly, 2022-01-29:
> The problem with Python 3.10 is that it will be on most Linux distro and in the next Flatpak runtime (August 2022). So it will definitely be a problem.

Why is this a problem? What code that runs with Python 3.8 cannot be executed in Python 3.10?
Also, what Qt 5.15.2 compatible code won't be executable with Qt 6?

I find it important to know precisely what code is affected and what not.

luzpaz, 2022-01-29:
It's discussed in the forum thread: https://forum.freecadweb.org/viewtopic.php?f=10&t=65089
Issues with coin3d/pivy and pyside

donovaly, 2022-01-29:
> It's discussed in the forum thread: https://forum.freecadweb.org/viewtopic.php?f=10&t=65089 Issues with coin3d/pivy and pyside

I cannot find anything when searching for "coin3d" in this thread. Can you therefore please list the affected code here in a short list to keep the overview?

hobbes1069, 2022-01-29:
Hopefully a new PySide release @waebbl mentioned get's released soon, but to answer part of your question:
https://bugzilla.redhat.com/show_bug.cgi?id=1990768
https://bugzilla.redhat.com/show_bug.cgi?id=2025599