FreeCAD/FreeCAD-Bundle #93: Path mounting errors?

KenwoodFox, 2022-03-01:
This could be a system problem but im not sure, after updating recently using the aur package for the weekly appimage builds, im getting lots of instability, screen shows black and i get this error in the logs. Frequent crashes, i know the weekly builds are buggy but i cant help but imagine this is more than the usual.

```
During initialization the error "expected str, bytes or os.PathLike object, not list" occurred in freecad.freecad_glider
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 238, in InitApplications
  File "/tmp/.mount_freecasZWZJ6/usr/lib/python3.9/posixpath.py", line 76, in join
    a = os.fspath(a)
TypeError: expected str, bytes or os.PathLike object, not list
--------------------------------------------------------------------------------
```

luzpaz, 2022-03-01:
Please post full [about](https://wiki.freecad.org) info. 
This looks like issue in @looooo's Glider workbench

KenwoodFox, 2022-03-02:
... What?

Your about link goes to the freecad wiki homepage,

Is this an OpenGlider bug? is that a plugin? Im just using the stock appimage.

Im not compiling the appimage locally, this is from here.

KenwoodFox, 2022-03-02:
```
FreeCAD 0.20, Libs: 0.20R27809 (Git)
```
[image]

luzpaz, 2022-03-02:
Glider WB is @looooo's baby. It's is only active when built from conda. 

BTW, my bad, .i guess this is the place to report your issue because FreeCAD-appimage repo was renamed to FreeCAD-Bundle.  Forgot about that.

KenwoodFox, 2022-03-02:
No worries! I was just confused for a momen~ So the builds here do include this bug?
I know they're weekly releases and can be buggy but this seems to be pretty nasty, anything we can do?

luzpaz, 2022-03-02:
I guess, wait till @looooo sees this ?? :laughing:

KenwoodFox, 2022-03-02:
My favorite kind of solution :rofl: the one where i just wait for somebody else to fix it~

LMK if i can do anything ill throw a PR together if needed

luzpaz, 2022-03-02:
BTW, these days I'm using the Snap package, the nightly 'edge' version. 
Try it, it updates nightly automagically
`snap install --edge freecad-ppd`

KenwoodFox, 2022-03-02:
> BTW, these days I'm using the Snap package, the nightly 'edge' version. Try it, it updates nightly automagically `snap install --edge freecad-ppd`

My religion is against snap :P so ill wait/make a pr to fix the appimage~

KenwoodFox, 2022-03-02:
Could i remove glider from the bundle? 

https://github.com/FreeCAD/FreeCAD-Bundle/blob/8e28d9ad26b500b5d2ebd1645ac6b00279ef358e/conda/linux/create_bundle.sh#L8

KenwoodFox, 2022-03-02:
Having trouble figuring out how to build locally, is there a guide?

looooo, 2022-03-02:
I guess this is related to a change on the freecad side. But the trace-back does not give a good hint what's wrong.

KenwoodFox, 2022-03-02:
Is there any more information i can get? More verbose logs? @looooo

looooo, 2022-03-02:
> Is there any more information i can get? More verbose logs? @looooo

checking the log file, but there is not a lot of information. I guess the issue is this line:
https://github.com/FreeCAD/FreeCAD/blob/master/src/App/FreeCADInit.py#L238

It was introduces with this commit:
https://github.com/FreeCAD/FreeCAD/commit/02cfa11c8e23427874ea5c910817da91a505afa7

I hope @chennes can explain the changes.

luzpaz, 2022-03-02:
That looks like code chennes touched recently for the Addon Manager

chennes, 2022-03-02:
Yeah, that line is wrong: I've removed it in https://github.com/FreeCAD/FreeCAD/commit/3af20938cd3dec19fe2a30750bce101fec0db18c

KenwoodFox, 2022-03-02:
Thats good news! Next time bunlder runs it should be fixed?

KenwoodFox, 2022-03-02:
Cron says the next build is sunday, any chance we could hit it with a manual run?

chennes, 2022-03-02:
I think that's @looooo if he's around

KenwoodFox, 2022-03-02:
Sorry to bug so much, i would apprecaite it a lot!

looooo, 2022-03-02:
I will trigger a new build. New bundle will be available tomorrow.

KenwoodFox, 2022-03-02:
Sweet~

looooo, 2022-03-03:
@KenwoodFox can you try again. Bundles are updated

KenwoodFox, 2022-03-03:
Yep! Just tried when i woke up, the aur package brought it down.

There is no longer any issues in the log, and it looks a bit better, the start screen still does not work but you clearly fixed the path bug.

[image]
[image]

luzpaz, 2022-03-25:
Start page issue is reported in #97 

Can we close this ticket?

KenwoodFox, 2022-03-28:
Yep! Ill subscrube to #97

closed 2022-03-28 by KenwoodFox