FreeCAD/FreeCAD-Bundle #95: BUG in BRepOffsetAPI_MakeOffset: segfault

xxxajk, 2022-03-10:
While scripting, I've ran into an interesting crash. It is repeatable with the current weekly. 
I noticed it happens with an older weekly release initially, but still happens with the current one.

FreeCAD_weekly-builds-27936-Linux-Conda_glibc2.12-x86_64.AppImage

Here's the way to trigger the bug.

```
circle=Part.Circle()
#any value <=1.0 yields segfault
circle.Radius=1.0
edges=[Part.Edge(circle)]
W = Part.Wire(edges)
S=Part.Shape(W)
NS=S.makeOffset2D(-0.25, join=2)
```

This will report `SIGSEGV signal raised: 11` in the Report view, and reports the following in the python console:
```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
Base.FreeCADError: BRepOffsetAPI_MakeOffset has crashed! (Unknown exception caught)
```

This shouldn't happen, even if radius becomes near zero (a point).
When radius is zero, it properly throws an error, and that's fine, unless it happens to be a complex set of wires, in which case the circle or arc should simply be omitted.

xxxajk, 2022-03-10:
I have found a pythonic work-around.
If I grow, then shrink, then shrink, it works fine.
E.G.
```
circle=Part.Circle()
circle.Radius=1.0
edges=[Part.Edge(circle)]
W = Part.Wire(edges)
S=Part.Shape(W)
NS=S.makeOffset2D(0.25, join=2)
S=NS.makeOffset2D(-0.25, join=2)
NS=S.makeOffset2D(-0.25, join=2)
```

works!

adrianinsaval, 2022-03-10:
why open an issue here? Specially considering there's already one in the main repo. This repo is purely about packaging, the devs are very unlikely to pay attention here. Also take into account that our guidelines ask taht you first post your issues in the help forum not diretly in the bug tracker.

xxxajk, 2022-03-10:
I reported it as a link, as indicated above in the log. Just thought it could be something deeper, which it apparently is. :-)
That said, I will do the initial postings there.

closed 2022-03-10 by donovaly