FreeCAD/FreeCAD-Bundle #101: Weekly releases with date/time.

leoheck, 2022-04-04:
Where is the release date of these files? I never know if there is a new version.

It would be awesome if we could have a new release every time, with the date/time written somewhere to make it easier to understand that a new release is available.

[image]

luzpaz, 2022-04-06:
Is there a way to automate the editing of the release text?

leoheck, 2022-04-06:
I think so.. it would be interesting to also make a new release every time. How are you guys creating releases? Manually? Or this is handled by the Github Actions or something?

leoheck, 2022-04-06:
I see, there is a workflow here.

I found this action that creates a release... 
https://github.com/actions/create-release

leoheck, 2022-04-06:
@luzpaz maybe this helps... https://github.com/FreeCAD/FreeCAD-Bundle/pull/104

luzpaz, 2022-04-06:
Cool, maybe we can use realthunder's approach in his A3 repo 
[image]

He uses collapsed divs to hid previous changelog since his releases are rolling. I could improve a script that I was working on that automatically outputs a oneliner commit message for all commits released from one release revision to next. Each week or whenever we generate new builds the script can fold previous changelog  away and then output the commits that made it in to the latest weekly release. In this [thread](https://forum.freecadweb.org/viewtopic.php?p=583871#p583871) I explore the script

luzpaz, 2022-04-07:
So essentially, to test this PoC:
1. cd in to your FreeCAD master cloned
2. `git pull upstream master`
3. get the latest AppImage revision number `curl -s -H "Accept: application/vnd.github.v3+json" https://api.github.com/repos/FreeCAD/FreeCAD-bundle/releases/tags/weekly-builds | jq '.assets[].name' | grep -e 'AppImage\"$' | sed -r 's/.*-([0-9]{5}).*/\1/'`
4. Get the result and swap it for the '28445' in this oneliner `git rev-list --count HEAD | xargs -I{} echo '{}-28445' | bc | xargs -I{} git log --pretty=format:'%h' --skip={} -n 1 | xargs -I{} git log --oneline --ancestry-path {}..HEAD`

TBD

leoheck, 2022-04-07:
I don't think Realthinder's approach is a good one. Why do you want to use it? Because it will be easy to apply?

luzpaz, 2022-04-07:
Well how else do you propose to record changes between appimage releases if we keep publishing to the same tag each time?

leoheck, 2022-04-07:
Pushing another tag. Could have time to check the proposal on my PR?

luzpaz, 2022-04-07:
52 tags a year (that's not counting bi-weekly builds and one-off builds push hot-fixes etc...). Do you think that's the better solution ? The stable tag builds will get buried in all that clutter

leoheck, 2022-04-07:
A tag is just a label. The number of tags is not a big deal. It is going to be good for the user to be able to test and use another release if there is one that is not working. For me, the latest is not working. I cannot use Freecad from here yet. And it was working last week. Now, talking about the releases, it looks like it has a limit of 2 GB. Then this might be the issue. In this case, we could have N working releases by removing the oldest one after Nth whenever the new one is released. For instance. We could agree that 3 or a good number, then we can work to make this happen. What do you think of this approach?

looooo, 2022-04-07:
I would prefer to keep things simple. There is always the option to use conda directly to install older builds. Conda packages for freecad  are stored for about 3 months. The bundles only gathers the packages from conda-forge.

adrianinsaval, 2022-05-26:
release assets include the date in the filename now and github now shows how long ago they were uploaded, should this be closed now?
[image]

luzpaz, 2022-05-26:
Nice that the fix came from github.  Thanks for pointing this out @adrianinsaval :+1:    
Closing

closed 2022-05-26 by luzpaz