FreeCAD/FreeCAD-Bundle #102: Cannot launch it on (pre-released) Ubuntu 22.04

leoheck, 2022-04-04:
I am still experiencing issues with the AppImage.

Some days ago it was working, not it cannot be launched anymore. 
Not sure if this is a bug here or if it is something that has changed on Ubuntu in the last few days.

This is what happens when I launch it from the command line. 

```bash
➜  Downloads chmod a+x FreeCAD_weekly-builds-28611-Linux-Conda_glibc2.12-x86_64.AppImage

➜  Downloads ./FreeCAD_weekly-builds-28611-Linux-Conda_glibc2.12-x86_64.AppImage    
Extracting usr/share/icons/default/freecad_weekly.png to "/home/lheck/.local/share/icons/default/appimagekit_545935d9f7331ab16d77afefe17637d5_freecad_weekly.png"
Extracting usr/share/mime/packages/org.freecadweb.FreeCAD.xml to "/home/lheck/.local/share/mime/packages/appimagekit_545935d9f7331ab16d77afefe17637d5_org.freecadweb.FreeCAD.xml"
WARNING: Unable to resize the application icon into a 128x128 image: "Unable to load image.". It will be written unchanged.
WARNING: Unable to resize the application icon into a 256x256 image: "Unable to load image.". It will be written unchanged.
FreeCAD 0.20, Libs: 0.20R28611 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Assembly4 workbench (v0.11.10) loaded
libGL error: MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
A3 available
/tmp/.mount_FreeCAyYbTbH/AppRun: line 42: 80416 Aborted                 (core dumped) ${MAIN} "$@"
```

The new image was installed, it is in 
```/home/lheck/.local/share/icons/default/appimagekit_545935d9f7331ab16d77afefe17637d5_freecad_weekly.png```

However, I still cannot see this image in the launcher. I would like to know how is the process to generate the bubble, so I can do it here to test if the process works.

luzpaz, 2022-04-05:
Can you try invoking:  
`MESA_LOADER_DRIVER_OVERRIDE=i965 FreeCAD_weekly-builds-28611-Linux-Conda_glibc2.12-x86_64.AppImage`

leoheck, 2022-04-06:
I found this `swrast_dri.so` in `/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so` in my system.
Then I made a symbolic link for it in the place where Freecad was looking for. 
Later it gave me this issue with the missing `iris_dri.so`, then I did the same.
```bash
sudo mkdir -p /usr/lib/dri/
sudo ln -s /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so /usr/lib/dri/swrast_dri.so 
sudo ln -s /usr/lib/x86_64-linux-gnu/dri/iris_dri.so /usr/lib/dri/iris_dri.so
```

It looks like something is better now, but it is still giving me another issue that I could not workaround yet.

```bash
➜  Applications ./FreeCAD_weekly-builds-28611-Linux-Conda_glibc2.12-x86_64_d6f4bdb223b526d4390282a40e90e65c.AppImage
FreeCAD 0.20, Libs: 0.20R28611 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Assembly4 workbench (v0.11.10) loaded
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_FreeCAGzpMkZ/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libLLVM-13.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_FreeCAGzpMkZ/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libLLVM-13.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_FreeCAGzpMkZ/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libLLVM-13.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
A3 available
/tmp/.mount_FreeCAGzpMkZ/AppRun: line 42: 78158 Aborted                 (core dumped) ${MAIN} "$@"
➜  Applications 
```

leoheck, 2022-04-06:
Maybe it is missing something during the build process. I am not sure how these AppImages have to be but I feel that it should be linking static libraries instead, isn't it? Maybe we have to use this flag to build it `-static-libstdc++`

leoheck, 2022-04-07:
I don't know Artix, but it is messing it's version. Could you add this too?

onekk, 2022-04-08:
Artix has no version, as as an Arch Linux derivative is a rolling release distribution.

If this is what you intend.

FC package versions, are related to the miniconda install (a sort of virtual environment inside the Linux distribution) that is almost similar in working with an AppImage "expanded on disk".

Commands I have posted are usually advised when there are problem with graphics chain as they show the "graphics card info" and the "OpenGL infrastructure".

I think they are "bare minimum" information to start, when troubleshooting a MESA problem.

glxinfo is usually in "mesa-utils". Sorry I don't know how it is called on Debian, but a package search with command name usually will be useful to find the containing package, if I remember well there are also some some apt... commands that will find the right package in a Debian distribution.

Regards

Carlo D.

leoheck, 2022-04-08:
Interesting, so if it is a rolling release, how do you compare or track issues related to the system? How do you know if an upgrade from today broke something that was working yesterday?

onekk, 2022-04-08:
Usually it is not too difficult, when you upgrade you see what is modified, and there the distribution forum, plus "Arch Linux" forum as a source of information.

Plus it is not difficult to downgrade packages, or even use an "package freeze archive" that "crystallize" package status on a precise date.

It is not as stable as Debian, but usually you could "pin" a problematic package, and live with it until problems are fixed, or even use an AUR package (unofficial "recipe" to builds packages in a different way that could solve problems).

Main advantage is that most of the time you use "last official" version of softwares and as Arch it is one of the most used distribution, you have usually a good chance to find information about problema and solutions.

see maybe: 

https://wiki.archlinux.org/title/Intel_graphics

To have an idea of the level of details and completness the official Arch Linux wiki.

I'm using a SystemD free flavour of Arch so maybe something has to be adapted, but it is not so difficult as there is also a dedicated ArtixLinux forum too.

But this discussion is off-topic here, I apologize for the long answer. 

Regards

Carlo D.

yoshinari-nomura, 2022-04-29:
I encounterd the same coredump, even with FreeCAD_0.19.3-Linux-Conda_glibc2.12-x86_64.AppImage on Ubuntu 22.04.

luzpaz, 2022-04-29:
> I would like to know how to recreate the AppImage package from an extracted one. I was not able to do yet. With that, I can fix the AppImage for myself.

https://wiki.freecad.org/AppImage#Repackaging_AppImages

leoheck, 2022-04-29:
I saw this and did not pass through because I thought it was a typo or wrong code...
But now you have shared it I am wondering if that it might be right...
Maybe AppImage has something inside to repack (itself) from an existing extracted folder...
[image]

leoheck, 2022-04-29:
No.. it does not work like that, it is the wrong code, unfortunately.

luzpaz, 2022-04-29:
Are you 100% certain? See https://github.com/AppImage/AppImageKit/#appimagetool-usage

leoheck, 2022-04-29:
No, I am just dumb. this is a tool `./appimagetool-x86_64.AppImage squashfs-root` and not the AppImage I want to use. Sorry, @luzpaz, thanks for helping me on this.

leoheck, 2022-04-29:
It worked flawlessly. Damm! I found this some days ago trying desperately to find this thing. Now I have an Ultimate Cura that works. Let me test with Freecad too.

leoheck, 2022-04-29:
Nice, it is the cure for Freecad too.

```
# Extract the package
 ./FreeCAD_weekly-builds-28765-Linux-Conda_glibc2.12-x86_64.AppImage --appimage-extract 

# Go inside
cd squashfs-root 

# Remove (rename) libstdc out of the sight... 
mv ./usr/lib/libstdc++.so.6 ./usr/lib/libstdc++.so.6.removed

# Recrearte a appImage file
./appimagetool-x86_64.AppImage squashfs-root 

# Rename to keep the same original name.
mv FreeCAD_Weekly-x86_64.AppImage FreeCAD_weekly-builds-28765-Linux-Conda_glibc2.12-x86_64.AppImage
```

yoshinari-nomura, 2022-04-30:
Thanks, it worked for me with both weekly-build and stable one!

As a side note, Ubuntu 22.04 also needs libfuse2.
```sh
sudo apt-get install libfuse2
```

universam1, 2022-05-04:
Thanks @leoheck for the workaround, want to confirm this solves my issue on Ubuntu 22.04.

Is there a permanent solution in view?

adrianinsaval, 2022-05-09:
ultimately this is a duplicate of #67 so it should be closed IMO

leoheck, 2022-05-17:
Preloading was not working on Ubuntu 22.04 and this thread here has a workaround.

leoheck, 2022-05-17:
Preloading was not working. And here I put a valid workaround.

adrianinsaval, 2022-05-17:
It appears you were given the wrong command here, did you try using the LD_PRELOAD command from the other issues and the known os specific issues post in the forum? The workaround you posted here was also posted before on the forum and one of the duplicates. Either way, workaround or not, this is a duplicate issue of #67 and #73

paulcobbaut, 2022-07-11:
I have the same issue since today on Debian stable. I have been using FreeCAD like this almost every day for six weeks without problems.

```
paul@PC 11-07-2022 13:22 ~/Downloads $ ./FreeCAD_0.20-29177-2022-06-15-conda-Linux-x86_64-py39.AppImage 
FreeCAD 0.20, Libs: 0.20R29177 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Interlocking laser cut workbench loaded
/tmp/.mount_FreeCAtqwkHQ/AppRun: line 42: 1686355 Aborted                 ${MAIN} "$@"

```
I tried using the solution (renaming ./usr/lib/libstdc++.so.6 in the AppImage) but the error persists.

```
paul@PC 11-07-2022 13:22 ~/Downloads $ ./FreeCAD_Weekly-x86_64.AppImage 
FreeCAD 0.20, Libs: 0.20R29177 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Interlocking laser cut workbench loaded
/tmp/.mount_FreeCAoeYKGw/AppRun: line 42: 1686430 Aborted                 ${MAIN} "$@"
```

probonopd, 2024-08-05:
It may be beneficial to look into what is going on here.

* I suspect that `AppRun` or some other script sets  `LD_LIBRARY_PATH` to point to the libraries inside the AppImage
* Now, when a binary or library from _outside_ the AppImage gets loaded, this  `LD_LIBRARY_PATH` is also used; it is "leaking"
* If the local operating system has a newer libstdc++ than what is bundled inside the AppImage, then libraries and applications outside of the AppImage will fail if they rely on that newer libstdc++
* Hence, the AppImage should either not rely on using `LD_LIBRARY_PATH` (e.g., by using `rpath` instead), or it should use something like https://invent.kde.org/brauch/appimage-exec-wrapper in oder to prevent `LD_LIBRARY_PATH` from "leaking" to the outside of the AppImage

adrianinsaval, 2024-09-09:
how is one supposed to use the appimage-exec-wrapper? Is there any documentation?

probonopd, 2024-09-14:
Please see https://github.com/AppImage/AppImageKit/wiki/Creating-AppImages/cc2441518975caca23e9ce2dba6f08a22c678d1e#environment-variables.

leoheck, 2024-09-14:
This is an old issue. Is this still a problem? 
For me this problem is in the past. I can tell that weeklies are working on Ubuntu 24.04 (released 2 years after 22.04 was released).

adrianinsaval, 2024-09-18:
> This is an old issue. Is this still a problem? For me this problem is in the past. I can tell that weeklies are working on Ubuntu 24.04 (released 2 years after 22.04 was released).

it might not be happening right now for you but it could happen again when the system and bundled libraries are not compatible. And the root cause might also be causing trouble when launching external applications

anyways, I checked our appimage and the binaries inside it have relative rpaths so they shouldn't actually need LD_LIBRARY_PATH to function so I removed that from our AppRun script, hopefully that will fix problems like this. If get reports that it doesn't I'll work on adding the exec wrapper

closed 2024-10-27 by adrianinsaval