FreeCAD/FreeCAD-Bundle #103: Add FreeCAD libs to PYTHONPATH

reox, 2022-04-05:
I noticed it is possible to run the internal python interpreter from the appimage (due to #30). This is useful if standalone freecad scripts shall be started (see the discussion in the forum: https://forum.freecadweb.org/viewtopic.php?f=22&t=67780)

Unfortunately, the python interpreter is not able to find freecad related libraries, because they do not reside in the PYTHONPATH.
A possible solution would be to simply add /usr/lib to the PYTHONPATH, by adding this line into AppRun:
```
export PYTHONPATH=${HERE}/usr/lib:${PYTHONPATH}
```
Or maybe there is another better way to achieve that.

luzpaz, 2022-04-06:
Are we taking in to account different package paradigms like snap and flatpack etc..?

reox, 2022-04-06:
I was close to creating a PR but I'm not sure if this is the best solution, that also works for other packaging formats, and it might introduce other issues... A more general solution would probably be to install the FC libs into the system-wide site-packages folder directly - but that would change how FC would have to search for libs. Hopefully, someone with more knowledge of the packaging process can figure something out :)

adrianinsaval, 2022-07-04:
For other packaging formats there is the option to use the INSTALL_TO_SITEPACKAGES flag to allow importing freecad from python, not sure what would be the `site packages` path for the appimage

adrianinsaval, 2022-07-04:
IMO it would be better to set `PATH_TO_FREECAD_LIBDIR=$APPDIR/usr/lib/` then to use freecad first run `import freecad` then whatever else you want

closed 2022-08-01 by looooo

Majestychen, 2025-04-08:
Just like operating Word with pywin32, is it possible to operate FreeCAD with pywin32?
eg:
import win32com
word = win32com.client.Dispatch('Word.Application')
Regarding FreeCAD：
fc = win32com.client.Dispatch('FreeCAD.Application')

commit f8eec47a (2022-08-01): [fixes #103] set PATH_TO_FREECAD_LIBDIR env var

Allows to import freecad when running python console from appimage