FreeCAD/FreeCAD-Bundle #105: Crash in `FreeCAD.AppImage -t 0` (r28647 and rr28671)

luzpaz, 2022-04-11:
@looooo any idea what causes this crash?

Tested on r28647
```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28647 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1c1743b)
Hash: 1c1743b67d209f2b6501de4f75645a33b69d1c31
Python 3.9.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: English/United States (en_US)
Installed mods: 
  * Behave-Dark-Colors 0.0.1
  * A2plus 0.4.55a
```

```
Module to read results not found.
Module pyNastran not found. Writing Mystran solver input will not be work.
test_femobjects_make (femtest.app.test_object.TestObjectCreate) ... doc objects count: 56, method: test_femobjects_make 
Save FreeCAD all objects file to /tmp/FEM_unittests/objects_create_all_531c2c143e6a/all_objects.FCStd ... 
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x42560) [0x7f332df0a560]
#1  0x7f32ad82b8b3 in vtkXMLWriter::WriteArrayHeader(vtkAbstractArray*, vtkIndent, char const*, int, int) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkIOXML-9.0.so.1+0xa3
#2  0x7f32ad82d515 in vtkXMLWriter::WriteArrayInline(vtkAbstractArray*, vtkIndent, char const*, int) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkIOXML-9.0.so.1+0x35
#3  0x7f32ad8139dc in vtkXMLUnstructuredDataWriter::WriteCellsInlineWorker(char const*, vtkDataArray*, vtkIndent) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkIOXML-9.0.so.1+0x12c
#4  0x7f32ad8180f0 in vtkXMLUnstructuredGridWriter::WriteInlinePiece(vtkIndent) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkIOXML-9.0.so.1+0x100
#5  0x7f32ad81274b in vtkXMLUnstructuredDataWriter::WriteInlineMode(vtkIndent) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkIOXML-9.0.so.1+0x10b
#6  0x7f32ad811b43 in vtkXMLUnstructuredDataWriter::WriteAPiece() from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkIOXML-9.0.so.1+0x43
#7  0x7f32ad815cad in vtkXMLUnstructuredDataWriter::ProcessRequest(vtkInformation*, vtkInformationVector**, vtkInformationVector*) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkIOXML-9.0.so.1+0x40d
#8  0x7f32acd5fc22 in vtkExecutive::CallAlgorithm(vtkInformation*, int, vtkInformationVector**, vtkInformationVector*) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkCommonExecutionModel-9.0.so.1+0x52
#9  0x7f32acd59649 in vtkDemandDrivenPipeline::ExecuteData(vtkInformation*, vtkInformationVector**, vtkInformationVector*) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkCommonExecutionModel-9.0.so.1+0x39
#10  0x7f32acd56c32 in vtkCompositeDataPipeline::ExecuteData(vtkInformation*, vtkInformationVector**, vtkInformationVector*) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkCommonExecutionModel-9.0.so.1+0x82
#11  0x7f32acd5c485 in vtkDemandDrivenPipeline::ProcessRequest(vtkInformation*, vtkInformationVector**, vtkInformationVector*) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkCommonExecutionModel-9.0.so.1+0x555
#12  0x7f32acd8b228 in vtkStreamingDemandDrivenPipeline::ProcessRequest(vtkInformation*, vtkInformationVector**, vtkInformationVector*) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkCommonExecutionModel-9.0.so.1+0x3a8
#13  0x7f32acd5ae53 in vtkDemandDrivenPipeline::UpdateData(int) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkCommonExecutionModel-9.0.so.1+0xa3
#14  0x7f32acd8c692 in vtkStreamingDemandDrivenPipeline::Update(int, vtkInformationVector*) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkCommonExecutionModel-9.0.so.1+0x122
#15  0x7f32ad822b10 in vtkXMLWriter::Write() from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkIOXML-9.0.so.1+0x40
#16  0x7f32ad7c93db in vtkXMLDataObjectWriter::WriteInternal() from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkIOXML-9.0.so.1+0x45b
#17  0x7f32ad82352a in vtkXMLWriter::RequestData(vtkInformation*, vtkInformationVector**, vtkInformationVector*) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkIOXML-9.0.so.1+0x34a
#18  0x7f32acd5fc22 in vtkExecutive::CallAlgorithm(vtkInformation*, int, vtkInformationVector**, vtkInformationVector*) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkCommonExecutionModel-9.0.so.1+0x52
#19  0x7f32acd59649 in vtkDemandDrivenPipeline::ExecuteData(vtkInformation*, vtkInformationVector**, vtkInformationVector*) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkCommonExecutionModel-9.0.so.1+0x39
#20  0x7f32acd56c32 in vtkCompositeDataPipeline::ExecuteData(vtkInformation*, vtkInformationVector**, vtkInformationVector*) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkCommonExecutionModel-9.0.so.1+0x82
#21  0x7f32acd5c485 in vtkDemandDrivenPipeline::ProcessRequest(vtkInformation*, vtkInformationVector**, vtkInformationVector*) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkCommonExecutionModel-9.0.so.1+0x555
#22  0x7f32acd8b228 in vtkStreamingDemandDrivenPipeline::ProcessRequest(vtkInformation*, vtkInformationVector**, vtkInformationVector*) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkCommonExecutionModel-9.0.so.1+0x3a8
#23  0x7f32acd5ae53 in vtkDemandDrivenPipeline::UpdateData(int) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkCommonExecutionModel-9.0.so.1+0xa3
#24  0x7f32acd8c692 in vtkStreamingDemandDrivenPipeline::Update(int, vtkInformationVector*) from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkCommonExecutionModel-9.0.so.1+0x122
#25  0x7f32ad822b10 in vtkXMLWriter::Write() from /tmp/.mount_FreeCAlU1bym/usr/lib/./libvtkIOXML-9.0.so.1+0x40
#26  0x7f32adf1afc1 in Fem::PropertyPostDataObject::SaveDocFile(Base::Writer&) const from /tmp/.mount_FreeCAlU1bym/usr/lib/Fem.so+0x2f1
#27  0x7f332f1cfd91 in Base::ZipWriter::writeFiles() from /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libFreeCADBase.so+0x81
#28  0x7f332f48fbc2 in App::Document::saveToFile(char const*) const from /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libFreeCADApp.so+0x3b2
#29  0x7f332f490319 in App::Document::save() from /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libFreeCADApp.so+0x289
#30  0x7f332f49044c in App::Document::saveAs(char const*) from /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libFreeCADApp.so+0xac
#31  0x7f332f4e2e3b in App::DocumentPy::saveAs(_object*) from /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libFreeCADApp.so+0xbb
#32  0x7f332f4e2f29 in App::DocumentPy::staticCallback_saveAs(_object*, _object*) from /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libFreeCADApp.so+0x29
#33  /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libpython3.9.so.1.0(+0x146faa) [0x7f332f8f7faa]
#34  /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_MakeTpCall+0x286) [0x7f332f8e6dc6]
#35  /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x5123) [0x7f332f8e3e43]
#36  /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libpython3.9.so.1.0(+0x13b9bb) [0x7f332f8ec9bb]
#37  /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libpython3.9.so.1.0(+0x14575e) [0x7f332f8f675e]
#38  /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x3d7) [0x7f332f8df0f7]
#39  /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libpython3.9.so.1.0(+0x13b9bb) [0x7f332f8ec9bb]
#40  /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x684) [0x7f332f8df3a4]
#41  /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libpython3.9.so.1.0(+0x12cf99) [0x7f332f8ddf99]
#42  /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libpython3.9.so.1.0(_PyFunction_Vectorcall+0xd5) [0x7f332f8ec6a5]
#43  /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libpython3.9.so.1.0(+0x1458d0) [0x7f332f8f68d0]
#44  /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x33fa) [0x7f332f8e211a]
#45  /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libpython3.9.so.1.0(+0x12cf99) [0x7f332f8ddf99]
#46  /tmp/.mount_FreeCAlU1bym/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_FastCallDictTstate+0x13e) [0x7f332f8e63de]
[... 186 lines cut ...]
#92  0x7f9c47ec94ec in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCANgOcid/usr/bin/../lib/libFreeCADBase.so+0x6c
#93  0x7f9c49172d26 in Gui::MainWindow::delayedStartup() from /tmp/.mount_FreeCANgOcid/usr/bin/../lib/libFreeCADGui.so+0x226
#94  /tmp/.mount_FreeCANgOcid/usr/bin/../lib/libFreeCADGui.so(+0x8d6883) [0x7f9c49178883]
#95  0x7f9c471f7594 in QObject::event(QEvent*) from /tmp/.mount_FreeCANgOcid/usr/bin/../lib/libQt5Core.so.5+0xbc
#96  0x7f9c47a3607f in QWidget::event(QEvent*) from /tmp/.mount_FreeCANgOcid/usr/bin/../lib/libQt5Widgets.so.5+0xcc5
#97  0x7f9c47ae034f in QMainWindow::event(QEvent*) from /tmp/.mount_FreeCANgOcid/usr/bin/../lib/libQt5Widgets.so.5+0xdd
#98  0x7f9c49178fac in Gui::MainWindow::event(QEvent*) from /tmp/.mount_FreeCANgOcid/usr/bin/../lib/libFreeCADGui.so+0x7c
[... cut ...]
#110  0x7f9c471e2e8f in QCoreApplication::exec() from /tmp/.mount_FreeCANgOcid/usr/bin/../lib/libQt5Core.so.5+0xfd
#111  0x7f9c48ce3e7d in Gui::Application::runApplication() from /tmp/.mount_FreeCANgOcid/usr/bin/../lib/libFreeCADGui.so+0x18bd
#112  /tmp/.mount_FreeCANgOcid/usr/bin/freecad(+0x655f) [0x564bf8d8e55f]
#113  /usr/lib/libc.so.6(+0x2d310) [0x7f9c46c52310]
#114  /usr/lib/libc.so.6(__libc_start_main+0x81) [0x7f9c46c523c1]
#115  /tmp/.mount_FreeCANgOcid/usr/bin/freecad(+0x6889) [0x564bf8d8e889]
```

looooo, 2022-04-11:
I can reproduce with a conda-environment. Looks like vtk leads to this crash.

luzpaz, 2022-04-12:
Opened a [forum thread](https://forum.freecadweb.org/viewtopic.php?f=4&t=67965)

looooo, 2022-05-17:
Is this issue still valid. recently I updated to vtk9.1.0.

luzpaz, 2022-05-17:
@looooo forgot about this. WIll test with the latest weekly once it finishes d/ling

luzpaz, 2022-05-17:
@looooo still waiitng for the latest appimage (r28884) to d/l but in the meantime i ran it on r28810 and it crashed

<details>
<summary>r28810 crash output</summary>

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x42560) [0x7fcdeb62f560]
#1  /usr/lib/libc.so.6(+0x184f9e) [0x7fcdeb771f9e]
#2  0x7fcdbb2159dc in cast_internal(_object*, _object*, char const*, int) from /tmp/.mount_FreeCAF4ZpS9/usr/lib/python3.9/site-packages/pivy/_coin.so+0x3c
#3  0x7fcdbb215b41 in cast(_object*, _object*) from /tmp/.mount_FreeCAF4ZpS9/usr/lib/python3.9/site-packages/pivy/_coin.so+0x61
#4  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x146faa) [0x7fcded020faa]
#5  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_MakeTpCall+0x286) [0x7fcded00fdc6]
#6  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x5123) [0x7fcded00ce43]
#7  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x13b9bb) [0x7fcded0159bb]
#8  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x14575e) [0x7fcded01f75e]
#9  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x3d7) [0x7fcded0080f7]
#10  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x13b9bb) [0x7fcded0159bb]
#11  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x684) [0x7fcded0083a4]
#12  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x12cf99) [0x7fcded006f99]
#13  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyFunction_Vectorcall+0xd5) [0x7fcded0156a5]
#14  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x1458d0) [0x7fcded01f8d0]
#15  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x33fa) [0x7fcded00b11a]
#16  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x12cf99) [0x7fcded006f99]
#17  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_FastCallDictTstate+0x13e) [0x7fcded00f3de]
#18  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_Call_Prepend+0x6a) [0x7fcded01dd5a]
#19  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x1f815b) [0x7fcded0d215b]
#20  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_MakeTpCall+0x286) [0x7fcded00fdc6]
#21  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x48eb) [0x7fcded00c60b]
#22  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x12cf99) [0x7fcded006f99]
#23  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyFunction_Vectorcall+0xd5) [0x7fcded0156a5]
#24  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x1458d0) [0x7fcded01f8d0]
#25  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x33fa) [0x7fcded00b11a]
#26  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x12cf99) [0x7fcded006f99]
#27  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_FastCallDictTstate+0x13e) [0x7fcded00f3de]
#28  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_Call_Prepend+0x6a) [0x7fcded01dd5a]
#29  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x1f815b) [0x7fcded0d215b]
#30  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_MakeTpCall+0x286) [0x7fcded00fdc6]
#31  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x48eb) [0x7fcded00c60b]
#32  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x12cf99) [0x7fcded006f99]
#33  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyFunction_Vectorcall+0xd5) [0x7fcded0156a5]
#34  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x1458d0) [0x7fcded01f8d0]
#35  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x33fa) [0x7fcded00b11a]
#36  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x12cf99) [0x7fcded006f99]
#37  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_FastCallDictTstate+0x13e) [0x7fcded00f3de]
#38  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_Call_Prepend+0x6a) [0x7fcded01dd5a]
#39  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x1f815b) [0x7fcded0d215b]
#40  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_MakeTpCall+0x286) [0x7fcded00fdc6]
#41  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x48eb) [0x7fcded00c60b]
#42  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x12cf99) [0x7fcded006f99]
#43  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyFunction_Vectorcall+0xd5) [0x7fcded0156a5]
#44  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x1458d0) [0x7fcded01f8d0]
#45  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x33fa) [0x7fcded00b11a]
#46  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x12cf99) [0x7fcded006f99]
#47  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_FastCallDictTstate+0x13e) [0x7fcded00f3de]
#48  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_Call_Prepend+0x6a) [0x7fcded01dd5a]
#49  /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libpython3.9.so.1.0(+0x1f815b) [0x7fcded0d215b]
[... cut ...]
#81  0x7fcdebbaae8f in QCoreApplication::exec() from /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libQt5Core.so.5+0xfd
#82  0x7fcded6b15bd in Gui::Application::runApplication() from /tmp/.mount_FreeCAF4ZpS9/usr/bin/../lib/libFreeCADGui.so+0x18bd
#83  /tmp/.mount_FreeCAF4ZpS9/usr/bin/freecad(+0x655f) [0x5631a91c055f]
#84  /usr/lib/libc.so.6(+0x2d310) [0x7fcdeb61a310]
#85  /usr/lib/libc.so.6(__libc_start_main+0x81) [0x7fcdeb61a3c1]
#86  /tmp/.mount_FreeCAF4ZpS9/usr/bin/freecad(+0x6889) [0x5631a91c0889]
testRayPick (MeshTestsApp.PivyTestCases) ... 
```

</details>

looooo, 2022-05-17:
seems to be a different issue.

luzpaz, 2022-05-17:
OK, just ran r28884.  
Good news: The crash is gone! :+1: 
Bad news: I'm see test errors  

<details>
<summary>Error log for r28884 `.AppImage -t 0` test</summary>

```
testHtmlCase (TestWebGui.WebGuiTestCases) ... ok

======================================================================
ERROR: testRayPick (MeshTestsApp.PivyTestCases)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/Mesh/MeshTestsApp.py", line 447, in testRayPick
    det=coin.cast(det,str(det.getTypeId().getName()))
SystemError: <built-in function cast> returned NULL without setting an error

======================================================================
ERROR: test_import_circle (OpenSCADTest.app.test_importCSG.TestImportCSG)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/OpenSCAD/OpenSCADTest/app/test_importCSG.py", line 109, in test_import_circle
    doc = self.utility_create_scad("circle(10.0);","circle")
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/OpenSCAD/OpenSCADTest/app/test_importCSG.py", line 82, in utility_create_scad
    return importCSG.open(filename)
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/OpenSCAD/importCSG.py", line 124, in open
    tmpfile = callopenscad(filename)
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/OpenSCAD/OpenSCADUtils.py", line 212, in callopenscad
    check_output2([osfilename, '-o', outputfilename, inputfilename])
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/OpenSCAD/OpenSCADUtils.py", line 182, in check_output2
    raise OpenSCADError('%s %s\n' % (stdoutd.strip(),stderrd.strip()))
OpenSCADUtils.OpenSCADError: " Can't open input file '/tmp/tmpi8fpbli3/circle.scad'!\n"

======================================================================
ERROR: test_import_cube (OpenSCADTest.app.test_importCSG.TestImportCSG)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/OpenSCAD/OpenSCADTest/app/test_importCSG.py", line 100, in test_import_cube
    doc = self.utility_create_scad("cube([1.0,2.0,3.0]);","cube")
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/OpenSCAD/OpenSCADTest/app/test_importCSG.py", line 82, in utility_create_scad
    return importCSG.open(filename)
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/OpenSCAD/importCSG.py", line 124, in open
    tmpfile = callopenscad(filename)
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/OpenSCAD/OpenSCADUtils.py", line 212, in callopenscad
    check_output2([osfilename, '-o', outputfilename, inputfilename])
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/OpenSCAD/OpenSCADUtils.py", line 182, in check_output2
    raise OpenSCADError('%s %s\n' % (stdoutd.strip(),stderrd.strip()))
OpenSCADUtils.OpenSCADError: " Can't open input file '/tmp/tmpi8fpbli3/cube.scad'!\n"

======================================================================
ERROR: test_import_cylinder (OpenSCADTest.app.test_importCSG.TestImportCSG)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/OpenSCAD/OpenSCADTest/app/test_importCSG.py", line 92, in test_import_cylinder
    doc = self.utility_create_scad("cylinder(50.0,d=10.0);","cylinder")
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/OpenSCAD/OpenSCADTest/app/test_importCSG.py", line 82, in utility_create_scad
    return importCSG.open(filename)
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/OpenSCAD/importCSG.py", line 124, in open
    tmpfile = callopenscad(filename)
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/OpenSCAD/OpenSCADUtils.py", line 212, in callopenscad
    check_output2([osfilename, '-o', outputfilename, inputfilename])
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/OpenSCAD/OpenSCADUtils.py", line 182, in check_output2
    raise OpenSCADError('%s %s\n' % (stdoutd.strip(),stderrd.strip()))
OpenSCADUtils.OpenSCADError: " Can't open input file '/tmp/tmpi8fpbli3/cylinder.scad'!\n"

======================================================================
ERROR: test_import_difference (OpenSCADTest.app.test_importCSG.TestImportCSG)
----------------------------------------------------------------------
Traceback (most recent call last):
[... 394 lines cut ...]

======================================================================
ERROR: test_femobjects_open_head (femtest.gui.test_open.TestObjectOpen)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/Fem/femtest/gui/test_open.py", line 111, in test_femobjects_open_head
    self.document.saveAs(file_path)
Base.FreeCADError: boost::filesystem::current_path: No such file or directory

======================================================================
ERROR: testMultiTransformCase (TestPartDesignGui.PartDesignTransformed)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAY1gXBq/usr/Mod/PartDesign/TestPartDesignGui.py", line 220, in setUp
    App.ActiveDocument.saveAs(self.TempDoc)
Base.FreeCADError: boost::filesystem::current_path: No such file or directory

----------------------------------------------------------------------
Ran 792 tests in 262.161s

FAILED (errors=40)
System exit
```

</details>

luzpaz, 2022-05-17:
It may be dependant because of some of the file path related issues?

adrianinsaval, 2022-07-04:
is this still an issue? was a forum discussion ever created?

closed 2022-07-04 by luzpaz