FreeCAD/FreeCAD-Bundle #108: macOS arm build

tcurdt, 2022-04-18:
Is there any chance for the 0.20 release to also include an official macOS arm build?

luzpaz, 2022-04-18:
CC @ipatch

ipatch, 2022-04-18:
i was recently granted access to a m1 machine, but my time has been limited. and the time that i do have, i've been spending it working on getting a freecad build built on a big sur box and being able to distribute a `FreeCAD.app` bundle that will run on macos 10.13 to the latest version. this has proven more difficult than i had originally planned, but have made some recent headway.

presently i'm using a macos vm to build and distribute the macos `FreeCAD.app` mac app bundles hosted on the homebrew-freecad tap. since the builds are built on macos mojave they will build on any version of macos mojave and newer. if you have an m1 mac then you should 🤞 be able to run the `FreeCAD.app` bundle that i'm distributing via the homebrew-freecad tap via Rosetta 2. that is something i haven't tested myself yet. as i'm more interested (when i have available time) to build on big sur and make certain the release will run on macos high sierra and newer.

i'm choosing to build on big sur because once i get all build on big sur and run on macos high sierra and newer issues ironed out i'll attempt to circle back around and see if i can build a native arm build of `FreeCAD.app`.

the major limitations at this time i believe are that freecad doesn't support qt6 (that i'm aware of) i think there's a forum thread about that. and i don't qt5 supports apple aarch64 ie m1 procs. so using features like the start workbench won't work because they rely on qtwebengine which is dependent on qt5 which qt5 never fully supported m1 processors, maybe that has changed but i haven't fully looked in to. so to get all the bells and whistles with freecad one would need to either get qtwebengine v5 working with m1 or update the freecad src to work with qt v6.2 as i believe that is the version that supports all the features of m1 procs. or you can selectively disable certain features of freecad. if you look at the freecad formula in the homebrew freecad tap you'll where there is an option to disable the qtwebengine during the build so you maybe be able to build a native m1 `FreeCAD.app` bundle. (not something i've tested).

i hope all this helps, and i wish i had unlimited amount of time to make this stuff happen unfortunately i don't so i try to get things when i can.

tcurdt, 2022-04-18:
> This has proven more difficult than i had originally planned, but have made some recent headway.

Are the difficulties shared in a ticked somewhere?

> If you have an m1 mac then you should 🤞 be able to run the `FreeCAD.app` bundle that i'm distributing via the homebrew-freecad tap via Rosetta 2.

I can confirm the weekly build at least working on Monterey via Rosetta 2.

> the major limitations at this time i believe are that freecad doesn't support qt6 (that i'm aware of) i think there's a forum thread about that. and i don't qt5 supports apple aarch64 ie m1 procs.

I think I read there is a qt5 backport by now. Not sure to what extend this works though.

> i hope all this helps, and i wish i had unlimited amount of time to make this stuff happen unfortunately i don't so i try to get things when i can.

Much appreciate sharing all the details.

luzpaz, 2022-04-18:
> the major limitations at this time i believe are that freecad doesn't support qt6 (that i'm aware of) i think there's a forum thread about that.

Indeed. We're not there yet. Most likely we will go full speed ahead on Qt6 after v0.20 is released. Here's the forum thread: https://forum.freecadweb.org/viewtopic.php?f=10&t=65089

looooo, 2022-05-10:
Conda-folge is also slowly getting closer to osx_arm support. qt is already available for osx-arm64. I am currently working on pyside2 to be updated to qt5.15 + osxarm64. Once we have pyside2 available for osx-arm64 I think it should be possible to update pivy too. After pivy is done it should be possible to compile freecad for osx_arm64. This is at least the roadmap for conda/conda-forge. Once this is done it should be possible to create a bundle for osx_arm64.

looooo, 2022-07-04:
pyside2-osx-arm64: https://anaconda.org/freecad/pyside2/files?version=5.15.5
pivy-osx-arm64: https://anaconda.org/freecad/pivy/files

what's next?

looooo, 2022-07-04:
Not sure, I will try. Maybe we need to use occt 7.6 as I am not sure if there are osx-arm packages for Occt 7.5. If so we need to use the internal smesh and netgen x.y. Not sure what is a good combination... @donovaly

luzpaz, 2022-07-04:
@ipatch have you heard of any 3rd party deps needed (beside what looooo is postulating above) for ARM support ?

ipatch, 2022-07-04:
unfortunately i would not know. i don't have access to a m1 (ARM) based mac. i had limited access to a box @chennes setup but the connection to it was spotty at best and i messaged him several times on gitter without a response.

JohnOCFII, 2022-07-05:
> @ipatch have you heard of any 3rd party deps needed (beside what looooo is postulating above) for ARM support ?

Forum user @doia (who appears to be Github user @j-wiedemann) was able to successfully build FreeCAD 0.20 development versions for M1 Mac.  I had used his info to successfully build on my M1 MacBook Pro as well, but for some reason it stopped working a couple of months ago.  He might be able to offer advice.

j-wiedemann, 2022-07-05:
> Forum user @Doia (who appears to be Github user @j-wiedemann) 

No, it is not true. I'm rockn on the forum.

JohnOCFII, 2022-07-05:
> No, it is not true. I'm rockn on the forum.

Oops.  Sorry about that.

chennes, 2022-07-05:
@ipatch Where did we leave off? (sorry, I don't check gitter very often)

looooo, 2022-07-05:
I was able to compile freecad for osx-arm64 via conda-packages. To install and test with conda:
```
conda create -n freecad_test freecad -c conda-forge -c freecad/label/dev
conda activate freecad_test
freecad
```

OCCT is now 7.6
smesh is not working as expected. Result in non working fem workbench.

lots of work to do :)

dirkolbrich, 2022-07-06:
@DOIA here ;-)
my arm64 builds broke as well. Using Python 3.9 gives me errors during cmake setup on Pivy and Pyside2, using Python 3.10 collides with the homebrew installed boost library. So, starting from 0.

chennes, 2022-07-06:
@dirkolbrich For the record, I'm getting that same error on my current Intel builds as well.

dirkolbrich, 2022-07-06:
linking @ageeye homebrew repo here as he managed to create a working arm64 build. Haven't digested the brew formulae yet.

https://github.com/ageeye/homebrew-freecad/releases/tag/M1_2022_07

looooo, 2022-07-12:
osx-arm bundles are available :) please test. I still have to do the builds manually, but I am working on cross-compiling so we have a fully automated weekly build for osx-arm64
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

also the osx-bundle (and the AppImage) was updated to occt7.6 / qt5.15 (finally). There is no need for the software-gl switch to make the start-wb work. So feel free to test :)

luzpaz, 2022-07-12:
> also the osx-bundle (and the AppImage) was updated to occt7.6
7.6 or 7.6.2  ?

looooo, 2022-07-13:
> > also the osx-bundle (and the AppImage) was updated to occt7.6
> 
> 7.6 or 7.6.2 ?

7.6.2

luzpaz, 2023-01-14:
@looooo can we close this ticket?

looooo, 2023-01-19:
hmm conda osx-arm builds are still somewhat beta I guess. But I am not going to add a osx-arm64 bundle for freecad 0.20. I will try to do this for 0.21 or 0.1 or whatever the next version is named.

Enver-Yilmaz, 2023-01-22:
Hi,

I tried FreeCAD_weekly-builds-31483-2022-12-31-conda-macOS-arm-py311.dmg on M1 mac without Rosetta 2 installed and it failed to execute. Budle itself looks like native but execution failed with the 'You need to install rosetta' message.

Enver-Yilmaz, 2023-01-22:
I also tried FreeCAD_weekly-builds-31641-2023-01-22-conda-macOS-x86_64-py310.dmg. Despite its name, it is probably universal and failed to execute with no rosetta message too.

fpunkt, 2023-07-16:
This is still the case with the 2032-07-xx arm files and I still refuse to install Rosetta ..

What works for me is to start the FreeCAD binary from the terminal (for convenience with symbolic link..)

`bin/freecad -> /Applications/FreeCAD-2023-07-10.app/Contents/Resources/bin/freecad`

When starting the first time it takes a while and then MacOS refuses with "unknown developer" message. Open System settings (? guessing the name here since I don't use the English language setting)  and allow the binary to be executed (somewhere in the security section, scroll down and there is the message opening FreeCAD was refused and it offers the "allow anyway" button)

Then next start from terminal works fine ... (well, not the nice FreeCAD icon but otherwise seems to work flawlessly for me)

chennes, 2023-07-17:
We will definitely have a native 0.21 binary for ARM (right, @adrianinsaval ?), but we won't being going back and producing a 0.20 version. And it will probably be codesigned (by me, at first, due to it taking forever to get the appropriate legal accounts set up for the FPA). So hopefully a lot of these headaches will go away soon.

adrianinsaval, 2023-07-17:
> We will definitely have a native 0.21 binary for ARM (right, @adrianinsaval ?)

Yep, we already have weekly builds for it.

JohnOCFII, 2023-07-17:
> > We will definitely have a native 0.21 binary for ARM (right, @adrianinsaval ?)
> 
> Yep, we already have weekly builds for it.

The weekly Arm builds are working fine for me (We've had a few bumps, but last week or two have been fine, other than having just a generic icon, and that we can't rename the App).  Not sure what the OP issue is.  When I look in `Activity Monitor` FreeCAD shows up as an Arm App.

```
    - OS: macOS 13.4.1
    - Word size of FreeCAD: 64-bit
    - Version: 0.21.0.33576 (Git)
    - Build type: Release
    - Branch: master
    - Hash: 8e6fbd2886d3ad51b75232bd0019d92d496e80cb
    - Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
    - Locale: C/Default (C)
```

adrianinsaval, 2023-07-23:
Are you using the latest arm bundle? The application is natively compiled and so is the bundle so I don't understand why it's asking you this

fpunkt, 2023-07-23:
Yes, I'm using the latest weekly. There was one weekly ARM binary from about Jan 2022 (?) that could be started from the Finder without Rosetta installed.
After a some pause updated ARM builds appear but cannot be started from Finder, although all binaries and libraries appear to be for ARM(*). Starting from terminal by typing `Downloads/FreeCAD.app/Contents/MacOS/FreeCAD` works after the binary has been "cleared/blessed" in System Settings.

(*) well, not all binaries - there are some Windows .exe files in the package that might be candidates for future cleanup.

```
mmf:~ frank$ cd /Volumes/FreeCAD_weekly-builds-33576-2023-07-13-conda-macOS-arm64-py310/
mmf:FreeCAD_weekly-builds-33576-2023-07-13-conda-macOS-arm64-py310 frank$ find . -type f -perm +111 | xargs file|grep -v 'shell script\|ASCII text\|Perl\|Python' | fgrep -v arm64
./FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/pip/_vendor/distlib/w64-arm.exe:                                                                                                          PE32+ executable (GUI) Aarch64, for MS Windows
./FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/pip/_vendor/distlib/w32.exe:                                                                                                              PE32 executable (GUI) Intel 80386, for MS Windows
./FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/pip/_vendor/distlib/t32.exe:                                                                                                              PE32 executable (console) Intel 80386, for MS Windows
./FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/pip/_vendor/distlib/w64.exe:                                                                                                              PE32+ executable (GUI) x86-64, for MS Windows
./FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/pip/_vendor/distlib/t64-arm.exe:                                                                                                          PE32+ executable (console) Aarch64, for MS Windows
./FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/pip/_vendor/distlib/t64.exe:        
```

JohnOCFII, 2023-07-24:
> Yes, I'm using the latest weekly. There was one weekly ARM binary from about Jan 2022 (?) that could be started from the Finder without Rosetta installed.

I'm curious.  I DO have Rosetta installed, but when I launch FreeCAD from the icon, it launches successfully, and Activity Viewer shows than a native Arm App is running.  Curious if there is something else in the launch process that either tries to launch an Intel process, or looks for Rosetta -- something that succeeds on my machine, but fails on yours.
```
    - OS: macOS 13.4.1
    - Word size of FreeCAD: 64-bit
    - Version: 0.21.0.33576 (Git)
    - Build type: Release
    - Branch: master
    - Hash: 8e6fbd2886d3ad51b75232bd0019d92d496e80cb
    - Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
    - Locale: C/Default (C)
  ```

adrianinsaval, 2023-08-22:
Would using zsh instead help?

chennes, 2023-08-22:
Probably -- which script are we talking about here? zsh has been the default shell for quite a long time now.

tcurdt, 2023-08-23:
The stackexchange post (at best) guesses it could be related to the shell script and then presents a work around.
It sure cannot hurt to try zsh - but I find this angle rather suspect. Bash is as much a universal binary as zsh is.
I'd be surprised if it is that - but never know 🤷‍♂️

On the first glance this also looks like it could be converted into posix style.
```
#!/bin/sh

HERE=$(dirname "$(dirname "$0")")
cat "$HERE/packages.txt"
export PREFIX="$HERE/Resources"
export LD_LIBRARY_PATH="$PREFIX/lib"
export PYTHONHOME="$PREFIX"
export FONTCONFIG_FILE="/etc/fonts/fonts.conf"
export FONTCONFIG_PATH="/etc/fonts"
export LANG="UTF-8"
export SSL_CERT_FILE="$PREFIX/ssl/cacert.pem"
export GIT_SSL_CAINFO="$PREFIX/ssl/cacert.pem"

product_version=$(sw_vers -productVersion)
if [ "$(echo "$product_version > 10.99" | bc)" -eq 1 ]; then
  export QT_MAC_WANTS_LAYER=1
fi

"$PREFIX/bin/freecad" "$@"
```
That should drop the bash vs zsh angle - but I am also just guessing here.

adrianinsaval, 2024-08-13:
can I get someone to test the latest bundle? I added a field to the Info.plist file that should tell app that this is indeed an arm64 executable. The solution posted about doesn't really work because mac doesn't do any variable substitution in the plist file so those env variables are invalid. unfortunately we need the bash script

adrianinsaval, 2024-08-13:
I'm gonna close this issue in favor of the more current and relevant #283

closed 2024-08-13 by adrianinsaval

fpunkt, 2024-08-15:
Hi, 

I've tried with FreeCAD_weekly-builds-38459-conda-macOS-arm64-py311.dmg but unfortunately the behaviour is still the same: asks to install Rosetta but works fine when started from the terminal (after going through system settings / security etc)

regards, F.

fpunkt, 2024-08-15:
Hi again,

now it works!

After starting via the command line I've noticed that the Dock now shows the FreeCad Icon - previous versions, when started via command line, had the exec icon.

Clicking on the FreeCad Icon in the right part of the Dock (recently closed apps) works fine. Starting from FInder/Applications also works fine.
I've rebooted and it still works fine.

Thanks!

F.

(still the hiccup that the very first start required the command line and allowing to execute in system settings, but I'm certainly not complaining...)