FreeCAD/FreeCAD-Bundle #110: Path workbench Post Process not starting spindle in git build 27422

bhowiebkr, 2022-04-22:
Missing commands like M3, M4, S##### and possibly others

bhowiebkr, 2022-04-22:
This also happens if "Split Output" is enabled or not, including options by Operation and Tool in the dropdown.

bhowiebkr, 2022-04-22:
If there is an easy fix let me know or the minimum gcodes I need to manually add to get this running. I'm still noobish with gcode, but I guess M03 (start) SXXXXX (rpm) and M05 (stop at the end of the file?). I don't have an ATC or a tool height sensor working so I gotta split out files for each tool or op to manually set the Z-height.

lagnat, 2022-05-06:
Possibly unrelated but I had a similar problem because I left the spindle speed at zero for the tool

luzpaz, 2022-05-07:
Wrong place to report this. What version of FC are you using btw?