FreeCAD/FreeCAD-Bundle #120: Crashing when used over SSH

remidemers, 2022-06-11:
Issue 0004755 is still present. It appears when started from a remote terminal over SSH. When started from a local terminal the issue do not appear. I experienced it with two images:
-   FreeCAD_weekly-builds-29085-2022-06-10-conda-Linux-x86_64-py39.AppImage
-   FreeCAD_0.19.3-Linux-Conda_glibc2.12-x86_64.AppImage

How to reproduce:
Install Debian 11.3 with only SSH and base utilities on computer #1
login...
`su`
`apt install i3`
`apt install firefox`
`exit`
`hostname -I`

Note the computer #1 local IP address.

`mkdir bin`
`cd bin`
`firefox`

With firefox, download the FreeCAD appimage and move it to ~/bin
`ln -s FreeCAD_weekly-builds-29085-2022-06-10-conda-Linux-x86_64-py39.AppImage freecad`
`./freecad`

You should see the FreeCAD working properly.
Close FreeCAD and go to computer #2
Computer #2 is a linux Mint 20.3 with i3 as user interface. It it able to run FreeCAD without problem but some slowness. On computer #2, open a gnome-terminal and enter the commands:

`ssh userpc1@computer1_IP -Y`    // userpc1 and computer1 to be the replaced with actual names
`export LIBGL_ALWAYS_INDIRECT=1`
`freecad`

Result:
I can see the FreeCAD splash screen with the nice drawing picture and modules loading text appears as expected normally for a few seconds and the it crashes. This is the output on the terminal:

userpc1@computer1:~/bin$ ./freecad
FreeCAD 0.20, Libs: 0.20R29085 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

/tmp/.mount_freecaW7LatP/AppRun: line 42:  8427 Aborted                 ${MAIN} "$@"

Please ask me if I can help with more information or testing. Thanks a lot.

luzpaz, 2022-06-11:
Any relation to https://github.com/FreeCAD/FreeCAD/issues/5592 ?

remidemers, 2022-06-11:
> Any relation to [FreeCAD/FreeCAD#5592](https://github.com/FreeCAD/FreeCAD/issues/5592) ?

I re-tested by calling the appimage directly. Same result.

luzpaz, 2022-07-01:
@remidemers please help test possible fix in https://github.com/FreeCAD/FreeCAD/pull/7034

adrianinsaval, 2022-07-02:
how have you setup your X server in the client machine?

adrianinsaval, 2022-07-02:
> Issue 0004755 is still present

this is confusing, what are you referring to? https://tracker.freecadweb.org/view.php?id=0004755 this doesn't seem related at all and on github 4755 is a PR removing py2 code

remidemers, 2022-07-13:
> how have you setup your X server in the client machine?

I cannot remember in details. I started with a basic Arch and added xorg and many other packages. Can I drop some config files to help?

remidemers, 2022-07-13:
Sorry for being inactive a long time. I want to clarify the bug and mention some findings: The computer runs FreeCad perfectly when called from related keyboard and screen. The problem exist only when FreeCad is called from remote computer through SSH. I read somewhere that latest OpenGL API cannot go through X forwarding. It is possible that the problem is there and unsolvable on the FreeCad side.
  https://stackoverflow.com/questions/53494718/how-can-i-xforwarding-opengl-applications-from-server

adrianinsaval, 2022-08-01:
> Yes, it shows by the error message that it fails at the same code line:
> /tmp/.mount_FreeCAMuoQyh/AppRun: line 41: 354130 Segmentation fault MAIN"@"

This error is as generic as it gets, whatever fails in FreeCAD this error would be shown because it's from the script that calls FreeCAD within the appimage
https://github.com/FreeCAD/FreeCAD-Bundle/blob/b75c4f81324d75300024017de878e33bc61cf9f4/conda/linux/AppDir/AppRun#L42

I understand why you got confused but these problems are unrelated. My bet is on the problem you posted so it probably can't be fixed on our side.

adrianinsaval, 2024-08-27:
no response for 2 years and likely not solvable on our side so I'm closing this issue

closed 2024-08-27 by adrianinsaval