FreeCAD/FreeCAD-Bundle #124: Part Design Pad tool do not work in linux AppImage 24212 works in Windows 0.20 24489 (git)

FreeCAD-Bug-Importer, 2022-07-06:
Issue imported from https://tracker.freecad.org/view.php?id=4606

* **Reporter:** jhelmine
* **Date submitted:** 3/25/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** #pending-forum, appimage

# Original report text

Hello,

Example were padding do not work in prebuild 24212 Linux version but works on Windows, compiled from git 24489.
Test file in http://jhelmine.kapsi.fi/3D/LedKehysVaunuun_test.FCStd (same included to report also).

# Additional information

Padding tested in:
FreeCAD_0.19-24212-Linux-Conda_glibc2.12-x86_64.AppImage
FreeCAD_0.19-23578-Linux-Conda_glibc2.12-x86_64.AppImage

# Steps to reproduce

To repeat error open file LedKehysVaunuun_test.FCStd.
First use "Part Design" then activate Body001Yläpallo.
Click Sketch006RuuvinRunko and click Pad this sketch.
Padding to normal direction do not show anything else than outlines in already made body. Reverse direction padding adds structure.

Windows version (0.20 from master) works as expected and I can continue. Before error situation parameters in spreadsheet ("data") was changed and paddings and pockets removed. Changing data.PallonD (text "Pallon halkaisija") to small value like 50 or less makes error situation, multiple bodies. This was done before error but correcting to back do not help.

# FreeCAD Info

```
OS: Ubuntu 20.10 (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24212 (Git) AppImage
Build type: Release
Branch: master
Hash: 3d227cf7179c00e57f2733f947b4b1b0368d219a
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: Finnish/Finland (fi_FI)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 3/27/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2021-03-27 10:25
Forum thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=57132

---

### Comment by Kunda1 2021-03-27 11:47
Per looo this looks like a discrepancy between OCCT 7.4.0 and 7.5.0

adrianinsaval, 2022-07-06:
I don't believe this is a bundle issue, most likely an OCC issue, I can now reproduce in all of these:
```
OS: Linux 5.18.9-arch1-1 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.29393 (Git) AppImage
Build type: Release
Branch: master
Hash: 6820e0a9ec85203a6f342ca72a2ff8fd417beaf1
Python 3.10.5, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
```
```
OS: Linux 5.18.9-arch1-1 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.29446 +1 (Git)
Build type: None
Branch: patch-2
Hash: 11c280bc675aa498a2fb6e29ea61cb9c065d8bc9
Python 3.10.5, Qt 5.15.5, Coin 4.0.1, Vtk 9.1.0, OCC 7.5.3
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
```
```
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.9, Qt 5.15.4, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.2
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
```
Note that in my previous report I did not even test the appimage

adrianinsaval, 2022-07-06:
aditionally the base shape is displayed differently depending on the pad length, very weird:
[image]