FreeCAD/FreeCAD-Bundle #125: Text of weekly-builds download page is innacurate

DuncanLithgow, 2022-07-24:
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

The current text for windows download is:

> There are different versions for windows
> 3.1 Bundles created using conda (need to be unzipped)
> 3.2 LP (libpack) bundles using FreeCAD's own libpack.
> 3.2.1 exe: installer
> 3.2.2 zip: Bundles that need to be unzipped.

But that is not what is actually available, what's available is only a .7z download (which is not a zip file, it's an archive). 

I'm happy to send a pull request with a more accurate text.

luzpaz, 2022-07-29:
Thanks @DuncanLithgow !
Please add revised text to this ticket?

DuncanLithgow, 2022-07-29:
For Linux the Appimages (conda-based) run on most distributions
For OSX download the Dmg's (conda-based)
For Windows download and unpack the 7zip archive

DuncanLithgow, 2022-08-20:
@luzpaz text suggestion is in previous comment

luzpaz, 2022-08-21:
@DuncanLithgow sorry, updated! Thanks! 
Closing

closed 2022-08-21 by luzpaz