FreeCAD/FreeCAD-Bundle #126: weekly builds stuck on version 29485

adrianinsaval, 2022-07-26:
The bundles are apparently being regenerated but keep using the same version, despite newer versions apparently available on conda-forge, perhaps the bundles are asking for an older version of python that has been disabled in conda-forge?

adrianinsaval, 2022-07-26:
people are confused about the situation: https://forum.freecadweb.org/viewtopic.php?p=611921#p611921

looooo, 2022-07-30:
I am confused too :). There are still issues with the freecad-feedstock windows builds:
```
CMake Error at D:/bld/freecad_1658989635143/_build_env/Library/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Coin3D (missing: COIN3D_LIBRARIES)
```
I guess something must have changed on the freecad side as there were no changes in the dependencies. 

Additional the smesh linking wasn't working.

looooo, 2022-07-30:
should be fixed by updating to occt7.6:
https://github.com/FreeCAD/FreeCAD-Bundle/commit/68913c432fb15a224d15f6f6b7a8723ee95a19f9

adrianinsaval, 2022-07-30:
I think there was a change with the coin3D cmake files but the builds were stuck before that

adrianinsaval, 2022-07-30:
Did you manage to build ifcopenshell or is it going to be skipped for the 7.6 builds?

looooo, 2022-07-30:
no I wasn't able to finish the IfcOpenShell builds.

looooo, 2022-07-30:
@adrianinsaval I added the FindCoin3D.cmake file back via a patch, but still I have the same issue with finding coin3d. Do you know what else has changed? @wwmayer

adrianinsaval, 2022-07-30:
Are these builds clean or are they caching something? I had coin errors in my local build after that change too but once I cleaned my build directory and reran cmake it worked fine

looooo, 2022-07-31:
No caching involved.

wwmayer, 2022-07-31:
> Do you know what else has changed?

No. I have only removed the FindCoin3D.cmake file to use that of CMake.

looooo, 2022-08-01:
@wwmayer What I don't understand is this:

in FreeCAD we are trying to find Coin3D. But for the pivy package I am using this command:
https://github.com/coin3d/pivy/blob/master/CMakeLists.txt#L10

I am not sure how `find_package(Coin3D REQUIRED)` should work without the additional cmake file. Are you using a newer version of coin? But actually I think there is no newer release available...

looooo, 2022-08-01:
Ok, I found this:
https://cmake.org/cmake/help/latest/module/FindCoin3D.html

but actually this is not working for conda + windows.

this line seems to be the culprint:
```
find_library(COIN3D_LIBRARY_RELEASE coin2 "[HKEY_LOCAL_MACHINE\\SOFTWARE\\SIM\\Coin3D\\2;Installation Path]/lib")
```

Why it is looking for coin2?

looooo, 2022-08-01:
this should solve this issue:
https://gitlab.kitware.com/cmake/cmake/-/merge_requests/7532

but we need to find a temporary solution. I already tried to patch the FindCoin3D file back, but this wasn't sucessful. Maybe someone can review the diff here:
https://github.com/looooo/freecad-feedstock/commit/e2faa8568e9e5f4acaa65b740eab164ac9ca4a2e#diff-b67c39fd3c8f77594d173eb9493b3ea220aac1b83134476705f1580fb7c71ebb

seems like the FINDCoin3D.cmake file from cmake is still preferred.

looooo, 2022-08-01:
@wwmayer would it be possible to revert the commit and keep the FindCoin3D.cmake file until my PR for cmake is merged and a new cmake version is available?

looooo, 2022-08-02:
I decided to try a different approach for the coin3d issue. Patching the cmake-feedstock to have the necessary diff for FindCoin3D.cmake included + uploading a cmake package to freecad/label/dev to continue working on the freecad build as soon as possible.

looooo, 2022-08-02:
ok, now only the smesh issue is blocking the windows weekly builds:

```

FemMeshShapeNetgenObject.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: int __cdecl SMDS_MeshInfo::NbVolumes(enum SMDSAbs_ElementOrder)const " (__imp_?NbVolumes@SMDS_MeshInfo@@QEBAHW4SMDSAbs_ElementOrder@@@Z) referenced in function "public: virtual class App::DocumentObjectExecReturn * __cdecl Fem::FemMeshShapeNetgenObject::execute(void)" (?execute@FemMeshShapeNetgenObject@Fem@@UEAAPEAVDocumentObjectExecReturn@App@@XZ)
FemMesh.cpp.obj : error LNK2001: unresolved external symbol "__declspec(dllimport) public: int __cdecl SMDS_MeshInfo::NbVolumes(enum SMDSAbs_ElementOrder)const " (__imp_?NbVolumes@SMDS_MeshInfo@@QEBAHW4SMDSAbs_ElementOrder@@@Z)
FemMesh.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: int __cdecl SMDS_MeshInfo::NbElements(enum SMDSAbs_ElementType)const " (__imp_?NbElements@SMDS_MeshInfo@@QEBAHW4SMDSAbs_ElementType@@@Z) referenced in function "private: void __cdecl Fem::FemMesh::copyMeshData(class Fem::FemMesh const &)" (?copyMeshData@FemMesh@Fem@@AEAAXAEBV12@@Z)
FemMesh.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: class SMDS_MeshNode const * __cdecl SMDS_Mesh::FindNode(int)const " (__imp_?FindNode@SMDS_Mesh@@QEBAPEBVSMDS_MeshNode@@H@Z) referenced in function "public: virtual void __cdecl `anonymous namespace'::CTETRAElement::addToMesh(class SMESHDS_Mesh *)" (?addToMesh@CTETRAElement@?A0x78421bce@@UEAAXPEAVSMESHDS_Mesh@@@Z)
FemMeshPyImp.cpp.obj : error LNK2001: unresolved external symbol "__declspec(dllimport) public: class SMDS_MeshNode const * __cdecl SMDS_Mesh::FindNode(int)const " (__imp_?FindNode@SMDS_Mesh@@QEBAPEBVSMDS_MeshNode@@H@Z)
FemMeshPyImp.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: int __cdecl SMESH_Mesh::NbNodes(void)const " (__imp_?NbNodes@SMESH_Mesh@@QEBAHXZ) referenced in function "public: class Py::Long __cdecl Fem::FemMeshPy::getNodeCount(void)const " (?getNodeCount@FemMeshPy@Fem@@QEBA?AVLong@Py@@XZ)
FemMeshPyImp.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: int __cdecl SMESH_Mesh::NbEdges(enum SMDSAbs_ElementOrder)const " (__imp_?NbEdges@SMESH_Mesh@@QEBAHW4SMDSAbs_ElementOrder@@@Z) referenced in function "public: class Py::Long __cdecl Fem::FemMeshPy::getEdgeCount(void)const " (?getEdgeCount@FemMeshPy@Fem@@QEBA?AVLong@Py@@XZ)
FemMeshPyImp.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: int __cdecl SMESH_Mesh::NbFaces(enum SMDSAbs_ElementOrder)const " (__imp_?NbFaces@SMESH_Mesh@@QEBAHW4SMDSAbs_ElementOrder@@@Z) referenced in function "public: class Py::Long __cdecl Fem::FemMeshPy::getFaceCount(void)const " (?getFaceCount@FemMeshPy@Fem@@QEBA?AVLong@Py@@XZ)
FemMeshPyImp.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: int __cdecl SMESH_Mesh::NbTriangles(enum SMDSAbs_ElementOrder)const " (__imp_?NbTriangles@SMESH_Mesh@@QEBAHW4SMDSAbs_ElementOrder@@@Z) referenced in function "public: class Py::Long __cdecl Fem::FemMeshPy::getTriangleCount(void)const " (?getTriangleCount@FemMeshPy@Fem@@QEBA?AVLong@Py@@XZ)
FemMeshPyImp.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: int __cdecl SMESH_Mesh::NbQuadrangles(enum SMDSAbs_ElementOrder)const " (__imp_?NbQuadrangles@SMESH_Mesh@@QEBAHW4SMDSAbs_ElementOrder@@@Z) referenced in function "public: class Py::Long __cdecl Fem::FemMeshPy::getQuadrangleCount(void)const " (?getQuadrangleCount@FemMeshPy@Fem@@QEBA?AVLong@Py@@XZ)
FemMeshPyImp.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: int __cdecl SMESH_Mesh::NbPolygons(enum SMDSAbs_ElementOrder)const " (__imp_?NbPolygons@SMESH_Mesh@@QEBAHW4SMDSAbs_ElementOrder@@@Z) referenced in function "public: class Py::Long __cdecl Fem::FemMeshPy::getPolygonCount(void)const " (?getPolygonCount@FemMeshPy@Fem@@QEBA?AVLong@Py@@XZ)
FemMeshPyImp.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: int __cdecl SMESH_Mesh::NbVolumes(enum SMDSAbs_ElementOrder)const " (__imp_?NbVolumes@SMESH_Mesh@@QEBAHW4SMDSAbs_ElementOrder@@@Z) referenced in function "public: class Py::Long __cdecl Fem::FemMeshPy::getVolumeCount(void)const " (?getVolumeCount@FemMeshPy@Fem@@QEBA?AVLong@Py@@XZ)
FemMeshPyImp.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: int __cdecl SMESH_Mesh::NbTetras(enum SMDSAbs_ElementOrder)const " (__imp_?NbTetras@SMESH_Mesh@@QEBAHW4SMDSAbs_ElementOrder@@@Z) referenced in function "public: class Py::Long __cdecl Fem::FemMeshPy::getTetraCount(void)const " (?getTetraCount@FemMeshPy@Fem@@QEBA?AVLong@Py@@XZ)
FemMeshPyImp.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: int __cdecl SMESH_Mesh::NbHexas(enum SMDSAbs_ElementOrder)const " (__imp_?NbHexas@SMESH_Mesh@@QEBAHW4SMDSAbs_ElementOrder@@@Z) referenced in function "public: class Py::Long __cdecl Fem::FemMeshPy::getHexaCount(void)const " (?getHexaCount@FemMeshPy@Fem@@QEBA?AVLong@Py@@XZ)
FemMeshPyImp.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: int __cdecl SMESH_Mesh::NbPyramids(enum SMDSAbs_ElementOrder)const " (__imp_?NbPyramids@SMESH_Mesh@@QEBAHW4SMDSAbs_ElementOrder@@@Z) referenced in function "public: class Py::Long __cdecl Fem::FemMeshPy::getPyramidCount(void)const " (?getPyramidCount@FemMeshPy@Fem@@QEBA?AVLong@Py@@XZ)
[... cut ...] struct std::less<class SMESH_subMesh *>,class std::allocator<struct std::pair<class SMESH_subMesh * const,class std::vector<int,class std::allocator<int> > > > > &)" (?Evaluate@StdMeshers_UseExisting_2D@@UEAA_NAEAVSMESH_Mesh@@AEBVTopoDS_Shape@@AEAV?$map@PEAVSMESH_subMesh@@V?$vector@HV?$allocator@H@std@@@std@@U?$less@PEAVSMESH_subMesh@@@3@V?$allocator@U?$pair@QEAVSMESH_subMesh@@V?$vector@HV?$allocator@H@std@@@std@@@std@@@3@@std@@@Z)
```

@wwmayer is this again connected to the int32 /int64 issue?

wwmayer, 2022-08-02:
> in FreeCAD we are trying to find Coin3D. But for the pivy package I am using this command:

Good catch! In CMake there are two modes to find packages (https://cmake.org/cmake/help/latest/command/find_package.html):
1. the module mode (that is used by FreeCAD)
2. The config mode (that is used by pivy)

In FreeCAD we use `find_package(Coin3D REQUIRED)` that uses the file FindCoin3D.cmake provided by CMake and already exists since 2008.

In pivy you use `find_package(Coin CONFIG REQUIRED)` and according to the documentation it expects either a file CoinConfig.cmake or coin-config.cmake.
On my system there is indeed a file /usr/lib/x86_64-linux-gnu/cmake/Coin-4.0.0/coin-config.cmake. Unfortunately, it's not yet part of all Coin3D versions.

The official FindCoin3D.cmake works on Linux and Windows (MSVC) but it doesn't work out of the box with MINGW. Therefore I needed this fix: 86906304957
In case you cannot find a way to determine Coin3D for Conda we can also extend SetupCoin3D.cmake to achieve that (instead of restoring the old FindCoin3D.cmake file).

> I decided to try a different approach for the coin3d issue.

Apparently you get it working now. That's good!

wwmayer, 2022-08-02:
> ok, now only the smesh issue is blocking the windows weekly builds:

Do you build from current master (i.e. ede18d20)? Do you apply a patch for the FreeCAD build? And which smesh version do you currently use, i.e. trelau's or FreeCAD's?

> is this again connected to the int32 /int64 issue or is this a general linking issue?

It's most likely the same problem.

looooo, 2022-08-03:
> Do you build from current master (i.e. ede18d20)? Do you apply a patch for the FreeCAD build? And which smesh version do you currently use, i.e. trelau's or FreeCAD's?

I build freecad with the external smesh repository from @trelau

> It's most likely the same problem.

yes, I guess I found the problem. I didn't use the branch (of the external smesh) which fixed this issue for the windows package. I fixed this and waiting now for a new smesh package. Hopefully the windows bundles are back soon.

wwmayer, 2022-08-03:
Great!

adrianinsaval, 2022-08-04:
thanks for the efforts, there's new builds available for all platforms now

closed 2022-08-04 by adrianinsaval

Ognian, 2022-10-14:
Somehow it looks for me that this is happening again:
v. 30492 is build (a least) on the 9.10.2022 and 13.10.2022
the last commit in the build (30492) is from 1.10.2022

looooo, 2022-10-14:
There is an issue with compiling freecad: https://github.com/FreeCAD/FreeCAD/issues/7585

There was a refactoring that breaks freecad with external smesh.