FreeCAD/FreeCAD-Bundle #127: FreeCAD_weekly-builds-29485-2022-07-24-conda-Linux-x86_64-py310: libQt5Core.so.5: version `Qt_5.15' not found

gneiss15, 2022-07-27:
As indicated by the Title, the lib is missing inside the appimage.
The version of libQt5Core that is includes is 5.12.

Steps to reproduce:
- Start FreeCad
- Select Workbanch "OpenScad". Report View will show:
It looks like you may be using a sandboxed version of FreeCAD. If OpenSCAD execution fails to load the temporary file, use FreeCAD's OpenSCAD Workbench Preferences to change the transfer mechanism.
- Open a scad fle. Report View will show:
...  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/tmp/.mount_FreeCArLATO5/usr/Mod/OpenSCAD/importCSG.py", line 124, in open
    tmpfile = callopenscad(filename)
  File "/tmp/.mount_FreeCArLATO5/usr/Mod/OpenSCAD/OpenSCADUtils.py", line 212, in callopenscad
    check_output2([osfilename, '-o', outputfilename, inputfilename])
  File "/tmp/.mount_FreeCArLATO5/usr/Mod/OpenSCAD/OpenSCADUtils.py", line 182, in check_output2
    raise OpenSCADError('%s %s\n' % (stdoutd.strip(),stderrd.strip()))
<class 'OpenSCADUtils.OpenSCADError'>: " /tmp/.mount_OpenSCfp96hc/AppRun.wrapped: /tmp/.mount_FreeCArLATO5/usr/lib/libQt5Core.so.5: version `Qt_5.15' not found (required by /tmp/.mount_OpenSCfp96hc/AppRun.wrapped)\n"

adrianinsaval, 2022-07-27:
the library is most likely not missing but it is not the one openscad was compiled against, openscad needs your system libraries but it can't find when launched from the appimage, have you looked at the preference page suggested in the report view?

gneiss15, 2022-07-29:
It's missing, because it is not in the path of the AppImage. Inside my system (Ubuntu 18.04) there is although only the Version 5.12 and I am able to launch and work with OpenScad (direct not from FreeCad). So my installed version of OpenScad didn't need version Qt_5.15, it must be one of the python scripts that request this special version.

gneiss15, 2022-07-30:
After some more investigations I found:
I tried using the portable version of OpenScad, that produce the shown error.
If I use the local installed one it works (even that the solid imported is not exactly the same as the one generated by the standalone version).
So the problem seams to be, that when OpenScad (the AppImage version) was called via OpenSCADUtils.py the lib-path wasn't set to the one of the AppImage!
Currently I have no idea how to do this.

adrianinsaval, 2022-08-01:
> It's missing, because it is not in the path of the AppImage

yes but it has no reason to be there, it most definitively has some version of qtcore or FreeCAD wouldn't work

> So the problem seams to be, that when OpenScad (the AppImage version) was called via OpenSCADUtils.py the lib-path wasn't set to the one of the AppImage!
> Currently I have no idea how to do this.

me neither, what is your setup to launch the appimage version of openscad? Does it work if you use FreeCAD from a regular package or flatpak/snap? If not then this is probably something that can't be fixed in the appimage but needs special code to handle this case in the openscad workbench itself.

gneiss15, 2022-08-01:
I (currently) don't have a regular package or flatpak/snap version of Freecad. The OpenScad Appimage is working if used from outside of FreeCad.
I believe the problem is similar to the one described in Issue 107 [https://github.com/FreeCAD/FreeCAD-Bundle/issues/107](url).

May the AppImage of OpenScad didn't set the LD_LIBRARY_PATH if it was already set. So maybe the patch provided there will fix this issue too.

adrianinsaval, 2022-08-01:
> I (currently) don't have a regular package or flatpak/snap version of Freecad.

Could you install one and test? I'm trying to understand if this needs fixing here in the bundle or in FreeCAD itself

gneiss15, 2022-08-02:
OK, I have installed the "normal" version, which is (due to Ubuntu 18.04) Version 0.16.
In this version there is no problem using either version of OpenScad.
So the problem seams to be the AppImage version of FreeCad.
It may be although a problem of the different FreeCad-versions, but I don't know how to install a newer version.

gneiss15, 2022-08-02:
More findings:
I wrote a small (.sh-)wrapper for the OpenScad-AppImage (see atached file [FreeCadOpenSCAD.zip](https://github.com/FreeCAD/FreeCAD-Bundle/files/9245613/FreeCadOpenSCAD.zip)).
When using this wrapper there is no more error when opening a scad file in FreeCad!

In principal the wrapper simply does a "unset LD_LIBRARY_PATH" before exec'ing the AppImage.
So my above believe (same problem as issue 107) seams to be thru.

On the other hand the imported geometry (of the file CR2032.scad) produces some errors, the importer of FreeCad can handle the output of this version of OpenScad not correct.

The AppImage of OpenScad I use is: version 2022.02.25.ai11146 (git fb10c5dbf).

gneiss15, 2022-08-03:
More infos see: [https://forum.freecadweb.org/viewtopic.php?f=10&t=70761&p=614795#p614795](url)

gneiss15, 2022-08-04:
Because this seams to be a more general problem I had opened a discussion here: [https://forum.freecadweb.org/viewtopic.php?f=10&t=70790](url)

adrianinsaval, 2024-09-18:
could someone please test if this issue still exist with the latest weekly? I mean the one that just got uploaded, if you have downloaded earlier today or yesterday I expect those to fail, even if they have the same revision number as the one today.

Appimages now shouldn't have LD_LIBRARY_PATH set anymore which is likely what was causing these problems with external applications

closed 2024-10-27 by adrianinsaval